espressif/esp-idf #1934: [Question]: Internal RAM 520, allowed for allocation 301KiB ? missing 219?

farizadp, 2018-05-07:
Hello,

Can anyone explain where the rest of 219KiB of internal RAM is used? Below is the early print of the firmware.

esp-idf sdk: e676676541ff31b72818246d267b0ccc529cb0d7

## Debug Logs

```
I (1252) heap_init: Initializing. RAM available for dynamic allocation:
I (1258) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1265) heap_init: At 3FFC2EB0 len 0001D150 (116 KiB): DRAM
I (1271) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1277) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1284) heap_init: At 400924CC len 0000DB34 (54 KiB): IRAM

```

negativekelvin, 2018-05-07:
Run `make size` to see your apps static memory use

Then add 64kb for cache, 8kb for rom reserved, and 64kb for bluetooth reserved

igrr, 2018-05-07:
Hi @farizadp, in addition to @negativekelvin's comment, please have a look at the similar question on the forum: https://esp32.com/viewtopic.php?f=2&t=3802.

projectgus, 2018-05-14:
Closing as this question appears to be answered.

closed 2018-05-14 by projectgus

negativekelvin, 2018-05-15:
Your debug shows 301K heap available which seems correct if you aren't using bluetooth

projectgus, 2018-05-15:
At the moment we don't allow all of the memory to be statically allocated, only 192KB of DRAM can be allocated this way. So the XX.X% used doesn't account for this memory.

This is why the amount of free RAM at runtime (when the extra DRAM is available) is higher than the amount "available" in the `make size` output.

We plan to fix this static allocation limit.

farizadp, 2018-05-15:
@projectgus: Please help my summarize it. Please correct my statement in case it isn't correct
- ESP32 physical memory is 520KB
- According to the output of ```make size```, I still have ~119 available for static allocation. Correct?
- Does it mean also that ~119KB is available for allocation during runtime (i.e. calling malloc)?
- According to the my debug output, available RAM at runtime is ~301KB. This doesn't represent available RAM at runtime due to the static allocation limit in the SDK I'm currently using. Correct?
- Although the debug output says ~301KB available for allocation, I still miss 219KB to get into complete physical memory of 520KB. Could you explain where this 219KB are?

projectgus, 2018-05-15:
* ESP32 total physical memory (all types) is 520KB

Can be broken down as:
* Static allocation (`make size)` allows you to allocate up to 128KB IRAM and 176.5KB DRAM [*].
* An additional 139.5KB of DRAM is added to the heap at runtime only (will appear in free RAM at runtime but not at compile time, on top of any memory not allocated statically).
* Just under 4KB of DRAM is reserved for ROM code use (via static accesses)
* 8KB of fast RTC RAM (contents kept on reset) 
* 64KB of RAM is reserved for caching accesses to flash cache mapped regions
= 520KB

If you enable Bluetooth the available static DRAM goes down by 64KB, as this memory is used directly by the Bluetooth controller.

This information isn't kept in a single obvious place (I think the next datasheet release will correct some of this), but some references:
* ESP32 Datasheet sections 3.1.2 & 3.1.4
* https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/soc_memory_layout.c#L69
* https://github.com/espressif/esp-idf/blob/master/components/esp32/ld/esp32.ld#L19

[*] I originally said this was 192KB but I had this wrong.

projectgus, 2018-05-15:
> Although the debug output says ~301KB available for allocation, I still miss 219KB to get into complete physical memory of 520KB. Could you explain where this 219KB are?

If your program is using 60KB+70KB (static allocated), plus 64KB for caches, plus 8KB RTC fast memory and 4KB used by ROM, then this gets us to 206. Plus 301KB made available as heap at runtime is 507KB. So the shortcoming is 13KB.

I can't explain this 13KB based on the output that's been posted in this thread, some of it may be padding or there's something we've missed..

negativekelvin, 2018-05-15:
RTC fast ram doesn't count towards the 520K.  This is from an app where everything adds up.  Probably the outputs shown above are from two different apps.

```
3ffae000-3ffae6e0 - rom reserved
3ffae6e0-3ffb0000 - heap dram
3ffb0000-3ffb21f4 - app dram .data - esp_image: segment 1: paddr=0x00015974 vaddr=0x3ffb0000 size=0x021f4 (8692) load
3ffb21f4-3ffb2a18 - app dram .bss (2084)
3ffb2a18-3ffe0000 - heap dram
3ffe0000-3ffe0440 - reserve rom pro
3ffe0440-3ffe4000 - heap dram/iram
3ffe4000-3ffe4350 - reserve rom app
3ffe4350-40000000 - heap dram/iram
40070000-40078000 - pro cache
40078000-40080000 - app cache
40080000-40080400 - app iram - esp_image: segment 2: paddr=0x00017b70 vaddr=0x40080000 size=0x00400 (1024) load
40080400-40088498 - app iram - esp_image: segment 3: paddr=0x00017f78 vaddr=0x40080400 size=0x08098 (32920) load
40088498-40088bd0 - app iram - esp_image: segment 5: paddr=0x00034a78 vaddr=0x40088498 size=0x00738 (1848) load
40088bd0-400A0000 - heap

Total sizes:
 DRAM .data size:    8692 bytes
 DRAM .bss  size:    2080 bytes
Used static DRAM:   10772 bytes ( 169964 available, 6.0% used)
Used static IRAM:   35792 bytes (  95280 available, 27.3% used)
```

projectgus, 2018-05-15:
> RTC fast ram doesn't count towards the 520K.

Hmm, you're right. I've never counted it all before. Will confirm and ask the docs team to update the datasheet.

farizadp, 2018-05-15:
I'll try to analyze my app based on the information above. FYI, I'm currently running out of RAM and I want to establish second TLS socket. To me, 1 TLS socket apparently consumes ~50KB (correct me if I'm wrong, using AES128-GCM-SHA256). I've got now ~90KB left (using ```esp_get_free_heap_size()``` ) and already have 1 socket opened.