espressif/esp-idf #1936: installing the ESP-IDF

Buffalchill, 2018-05-08:
I tried to install the ESP-IDF on two different windows pc's but it didn't work. I had to make a new windows user and called it "esp" because the ESP-IDF does not support spaces in directorys.

i just tried to "make menuconfig" but that didn't work, also I can't copy the "hello_world" example.
Here is the output of the mysys32.

[image]

chegewara, 2018-05-08:
Did you run this file:
windows_install_prerequisites.sh, from ...\msys\home\USER\esp\esp-idf\tools\windows folder?
You have also remember that in msys you cant use normal windows notations. You have to use commands like this:
`cd /drive_letter/folder/subfolder/etc`

There is no need to create new windows user. You can use any folder/subfolder without space that is not your user subfolder. If you will follow those instructions you would not have any issues:
http://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html

projectgus, 2018-05-08:
Hi @Buffalchill,

Sorry for the unpleasant experience you've had. We're working on some changes to make Windows setup smoother in the future.

I think there is typo in your IDF path, it is set to `C:/mysys32` rather than `C:/msys32`.

Angus

Buffalchill, 2018-05-08:
Thank you guys. That realy helped me i got it running now.
but i dont understand now can i also install the IDF into another folder than the "home/user-name/esp" ?

chegewara, 2018-05-08:
You can, but you have to change IDF_PATH. That should be enough i think.

Buffalchill, 2018-05-08:
hi yes it was possible. thank you.
i just had to use `mkdir -p esp`  instead of  `mkdir -p ~/esp`