espressif/esp-idf #1940: Code for Host interface via SDIO or SPI

ckinnera, 2018-05-09:
Hi,
I am sorry to post here, but I tried posting in the ESP32 forum no answer.
Basically I am looking for a reference code or any documentation related to host interface protocol.
I wanted to use ESP32 module as a network processor for my STM32FXXX application processor.
Any pointers or a contact details whom I can contact will be of great help.

Thanks in advance.

Best Regards,
Chiranjeevi

negativekelvin, 2018-05-10:
https://github.com/espressif/esp32-at/tree/master/main/interface/hspi

ckinnera, 2018-05-10:
Thanks for the link.
But I am not looking for AT interface. I want around 20 Mbps of speed, so looking for Either SDIO or SPI host interface.

Best Regards,
Chiranjeevi

ckinnera, 2018-05-10:
I am sorry, please ignore my email.
Thanks a lot for the link I will check and ask you in case of questions.

Best Regards,
Chiranjeevi

closed 2018-05-14 by FayeY

ckinnera, 2018-05-14:
Hi,

Is there a firmware for wroom32 which supports socket interface over SPI ?

Thanks,
Chiranjeevi

sathishgkg, 2019-01-12:
Hi im working in esp32 using at commands, with uart i can able to send at commands and get it response from esp32, i want to do this at communction using sdio from my master microcontroller to esp32 plz guid me to complete this stuck for long time.

Thankyou