espressif/esp-idf #1945: Looped noise on A2DP sink example

jeanthom, 2018-05-11:
Hi !

I noticed that the vanilla A2DP example has an annoying bug when used with an I2S DAC (I use a PCM5102). After disconnecting the A2DP source, the ESP32 continues to push data to the DAC (which leads to unbearable noise). I guess that's because the I2S DMA buffer isn't flushed after being disconnected from the A2DP source.

How to reproduce :

1. Flash the [a2dp_sink example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_sink)
2. Connect to the ESP32 using any A2DP source (I tested using a Fedora laptop, but also on various iOS devices)
3. Play some music (to fill the buffers with non-zero data)
3. Disconnect from the ESP32 or disable audio profile
4. You should still get some sound on your I2S DAC, some looped noise

Issue #1789 has a fix that works for me.

me-no-dev, 2018-05-11:
This is caused by the fact that the driver continues to run through the same samples, because it does not clear the DMA buffers after it is done with them. I advise you to have a static buffer of X samples that are zero, which you can push to I2S after the last packet you send. the number of consecutive pushes that you need is equal to the number of DMA buffers selected when instantiating the I2S hardware. There is also a zero_dma_buffers function but I am not 100% positive how good of idea it is to be used (since it might clear some samples that you are currently playing).

@igrr I have a driver that I am working on for Arduino that does not have this problem. Maybe we can find a middle ground and implement the "fix" in IDF instead.

jeanthom, 2018-05-11:
I tried to push an empty buffer after receiving "Disconnect" event but that didn't improve much the situation...

Where can I find the zero_dma_buffers() function ? I don't mind clearing samples that are currently being played since the I2S DAC shouldn't be playing music once the A2DP source is disconnected.

me-no-dev, 2018-05-11:
https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/i2s.h#L431

jeanthom, 2018-05-12:
Thank's.

I did this patch :

```diff
diff --git a/examples/bluetooth/a2dp_sink/main/bt_app_av.c b/examples/bluetooth/a2dp_sink/main/bt_app_av.c
index b1c432c5..a515cab5 100644
--- a/examples/bluetooth/a2dp_sink/main/bt_app_av.c
+++ b/examples/bluetooth/a2dp_sink/main/bt_app_av.c
@@ -105,6 +105,11 @@ static void bt_av_hdl_a2d_evt(uint16_t event, void *p_param)
         m_audio_state = a2d->audio_stat.state;
         if (ESP_A2D_AUDIO_STATE_STARTED == a2d->audio_stat.state) {
             m_pkt_cnt = 0;
+        } else {
+            ESP_LOGI(BT_AV_TAG, "Flushing I2S buffer");
+            if (i2s_zero_dma_buffer(I2S_NUM_0) != ESP_OK) {
+                ESP_LOGE(BT_AV_TAG, "Flushing I2S buffer failed");
+            }
         }
         break;
     }
```

But it did not solve my issue, event though I see "Flushing I2S buffer" just after disconnecting from the ESP32. Maybe I'm not calling i2s_zero_dma_buffer() at the right time ?

For the moment being I'm sticking with #1789 solution.

blueMoodBHD, 2018-09-25:
Hi @jeanthom , you can clear the DMA buffers after doing some delay, or clear the DMA buffers after getting `ESP_A2D_CONNECTION_STATE_EVT` with `ESP_A2D_CONNECTION_STATE_DISCONNECTED`.

And about the solution in  #1789, we will do some discussion.

closed 2018-11-28 by igrr