espressif/esp-idf #1948: Feature / Implementation Request

paulreimer, 2018-05-12:
The following bootloader snippet (commented out in `esp-idf`) is helpful for setting up factory reset behaviour:
https://github.com/espressif/esp-idf/blob/master/components/bootloader/subproject/main/bootloader_start.c#L86-L90

I think an even better approach would be possible with a weakly defined symbol (or a function pointer that could be set) so that it is possible to extend behaviour at that point, without having to patch `esp-idf`.

Is this something that seems easy? It would save a lot of housekeeping overhead from maintaining a bootloader fork. Maybe it's already possible, and I just can't see how to configure it.

paulreimer, 2018-05-12:
Something like this by default:
```
__attribute__((weak)) int override_selected_boot_partition(int boot_index) {
  return boot_index;
}
```

then we could write e.g. (although not sure where to put it locally within an app, to patch the bootloader subproject):
```
int override_selected_boot_partition(int boot_index) {
  bool factory = ?;
  return factory? -1 : boot_index;
}
```

projectgus, 2018-05-14:
Hi @paulreimer ,

We had some discussion about this approach (hooks vs fully custom logic) a while back, and in the end we settled on the approach you've linked above - the idea is that the basic bootloader is so skeletal (less than 100 LoC) that modifying it simple, and this allows you to hook any stage of the process.

> without having to patch esp-idf.

The recommended approach (maybe not documented clearly enough) is to copy the bootloader component in your project, ie `<project dir>/components/bootloader/...`. Most of the IDF-specific logic lives in `bootloader_support`, and with luck you will not need to copy this component. The bootloader_support API should remain compatible when you upgrade IDF.

Does that seem like a viable approach? If it's not working for you, please let us know and we'll work through it with you some more.

FayeY, 2018-05-23:
Hi @paulreimer , is there any update about this issue?

paulreimer, 2018-05-25:
I had done the solution that @projectgus mentioned for my prototype (vendor'ing the bootloader component, patching/overriding it).

I just noticed this merge request from @projectgus which is an excellent solution. With these features (gpio selection available in `menuconfig`, nice!) I can use this new bootloader without modifications.
https://github.com/espressif/esp-idf/commit/c7bedc80c11d24f667864ea77b25d75853791c33

At a glance, I am assuming that the new config `CONFIG_BOOTLOADER_HOLD_TIME_GPIO` delays the boot by this much, regardless of GPIO state? Or does it boot instantly if the button is NOT pressed or is released before the target amount of time has elapsed?

paulreimer, 2018-05-26:
Another question, to confirm whether `BOOTLOADER_OTA_DATA_ERASE` (which says "Clear OTA data on factory reset (select factory partition)" in the help string), is actually required to boot into the factory partition? In my case, I want to boot the factory partition but not (necessarily) erase an OTA partition (so I think that only `BOOTLOADER_FACTORY_RESET` would be required. But I'm not sure what the default behaviour would be if only that and a GPIO pin are enabled.

https://github.com/espressif/esp-idf/blob/c7bedc80c11d24f667864ea77b25d75853791c33/components/bootloader/Kconfig.projbuild#L85

paulreimer, 2018-05-30:
I confirmed that the bootloader will delay for `CONFIG_BOOTLOADER_HOLD_TIME_GPIO` no matter whether the GPIO pin is pressed or not.

As I see it, it should boot immediately as soon as the pin is detected in a released state. (For example, if I wanted a 30 second factory reset timeout, right now this feature would delay ALL boots by 30 seconds.)

projectgus, 2018-05-31:
Hi @paulreimer ,

Sorry for the delay in getting back to you.

> I confirmed that the bootloader will delay for CONFIG_BOOTLOADER_HOLD_TIME_GPIO no matter whether the GPIO pin is pressed or not.

Can you elaborate on what you're seeing here? The behaviour of the check function is such that it should be returning with GPIO_SHORT_HOLD (-1) the moment that the pin is no longer low:
https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/bootloader_common.c#L50

> Another question, to confirm whether BOOTLOADER_OTA_DATA_ERASE (which says "Clear OTA data on factory reset (select factory partition)" in the help string), is actually required to boot into the [...]

I think this may be a bug. If BOOTLOADER_FACTORY_RESET is enabled but BOOTLOADER_OTA_DATA_ERASE is cleared then it should be possible to boot to the factory image, but it looks like this may cause the factory reset mode to be ignored (if the button is held down then it will still erase any custom data partitions which are set to be erased.)

I'll check into this and get back to you.

projectgus, 2019-03-01:
Closing due to a lack of response. @paulreimer if there are still shortcomings in the current bootloader, please open new issues and we'll look into what can be improved.

closed 2019-03-01 by projectgus