espressif/esp-idf #1949: make monitor shows up in new terminal but code is unreadable

ahmedawais, 2018-05-12:
Hi,
  i am trying to use the bluetooth of esp32 and running the code in eclipse IDE whihc is installed on virtual machine ubuntu. the code is running fine but when i build target of make monitor in eclipse it runs but after that is shows an unreadable code. please help

ahmedawais, 2018-05-12:
yes in menuconfig FLASH SPI is 26 Mhz

negativekelvin, 2018-05-12:
Not spi speed, Main XTAL frequency

ahmedawais, 2018-05-12:
where is the option in menuconfig to set this XTAL frequency ?

negativekelvin, 2018-05-12:
https://esp-idf.readthedocs.io/en/latest/get-started/index.html#monitor

ahmedawais, 2018-05-12:
thanks @negativekelvin  code is working now. but the ESP32 bluetooth is not appearing in the andriod phone i am using nRF UART app. do you have any idea ?

nkolban, 2018-05-12:
@ahmedawais You are likely going to want to create a new issue for a completely different topic or else post a question to the ESP32 forum found here:

https://esp32.com/index.php

projectgus, 2018-05-14:
@ahmedawais , yes please ask this question on the forum. Provide as many details there as you can.

closed 2018-05-14 by projectgus