espressif/esp-idf #1953: Continuous interrupt on GPIO36 when wifi sleep mode WIFI_PS_MODEM is used

abhimanyuv1, 2018-05-14:
Hi, 

When wifi sleep mode WIFI_PS_MODEM is use GPIO36 have continuous interrupt. I have seen the errata where it says some RTC peripherals can cause this. Question is it possible to fix it or it will be always a side effect of it. I can't change GPIO36 as i have use all gpio for something else. Any other fixes HW or SW is possible?

Thanks

kewalmshah, 2018-05-15:
Hi @abhimanyuv1, could you please help modify this example to reproduce the above issue, https://github.com/espressif/esp-idf/blob/master/examples/wifi/power_save/main/power_save.c
This will help us to debug faster.

abhimanyuv1, 2018-05-16:
Hi Please find the source code with changes.
[power_save.tar.gz](https://github.com/espressif/esp-idf/files/2008431/power_save.tar.gz)

In my project i am using v3.0 branch where i can reproduce this issue easily. I also tried on master and found that issue is not present anymore. May be something changed which fixes it.

If you could please share the fix i can use it on v3.0 branch.

Thanks Abhimanyu V

kewalmshah, 2018-05-17:
Hi abhimanyuv1,

We have a fix for this in libphy, and we will  update it in V3.0 as well.

Thanks

abhimanyuv1, 2018-05-21:
Thanks, do you know when it is expected to be update? after merging please free to close this issue. Let me know i will be happy to test it as well on my setup.

lrodenasSistel, 2019-09-25:
I am facing a very similar issue, described in #4117

abhimanyuv1, 2019-12-17:
Hi, did the new libphy was never merged to 3.0 branch? i have project which was on v3.0 and i wanted to loook at reducing power using modem sleep.

abhimanyuv1, 2019-12-17:
FYI, just for curiosity i tested this on v3.3.1 and it has same issue.

jack0c, 2022-01-05:
version since release/v3.3 fixed this bug,
for 3.3 : 5490dbcaf7daba44aeb053fab1e7d76d2cbd985b

closed 2022-01-05 by jack0c