espressif/esp-idf #1954: CMAKE ESP32: missing dependancy on esp32 component

nicola-lunghi, 2018-05-14:
My component is including esp_wifi.h

```
#include "esp_wifi.h"
```

```
In file included from /home/nlunghiadm/src/esp32/esp-idf/components/esp32/include/esp_wifi.h:69:0,
                 from ../components/esp32-homie/homie.c:5:
/home/nlunghiadm/src/esp32/esp-idf/components/esp32/include/esp_event.h:23:27: fatal error: tcpip_adapter.h: No such file or directory
compilation terminated.
```

in esp-idf/components/esp32/include/esp_event.h there is a 

```#include "tcpip_adapter.h"```

but in CMakeLists.txt for esp32 component (esp-idf/components/esp32/CMakeLists.txt)

```
if(BOOTLOADER_BUILD)
    # For bootloader, all we need from esp32 is headers
    set(COMPONENT_ADD_INCLUDEDIRS include)
    set(COMPONENT_REQUIRES ${COMPONENTS})
    set(COMPONENT_SRCDIRS "")
    register_component(esp32)

...

else()
    # Regular app build

    set(COMPONENT_SRCDIRS ". hwcrypto")
    set(COMPONENT_ADD_INCLUDEDIRS "include")

    set(COMPONENT_REQUIRES driver)  # required because esp_sleep.h uses gpio_num_t & touch_pad_t
    set(COMPONENT_PRIV_REQUIRES
        app_trace bootloader_support ethernet log mbedtls nvs_flash
        pthread spi_flash tcpip_adapter vfs wpa_supplicant xtensa-debug-module)

    register_component()

```
I think something is missing from the BOOTLOADER part

    set(COMPONENT_PRIV_REQUIRES tcpip_adapter)

projectgus, 2018-05-17:
Thanks @nicola-lunghi for reporting this. I'm going to see if we can build a CI tool which checks header dependencies like this, to make sure each component's public header only requires the components it publically depends on.

I think the best fix is to add `tcpip_adapter` to `COMPONENT_REQUIRES` for the non-bootloader case. For the bootloader case, the use of esp32 component is very minimal so nothing should be including `esp_event.h`.

In time we'll refactor some of these dependencies so they are less awkward, as well.

projectgus, 2018-06-12:
Hi @nicola-lunghi ,

This should be taken care of in feature/cmake branch now. Please report any other missing dependencies that you find.

Angus

closed 2018-06-12 by projectgus