espressif/esp-idf #1955: xRingbufferSendFromISR causing wptr jumps over rptr

amuzyka-grinn, 2018-05-14:
## Environment

- Development Kit:      none
- Kit version 
- Core (if using chip or module): ESP32
- IDF version release/v3.0
- Development Env:		Eclipse
- Operating System:		Ubuntu
- Power Supply:         USB and Battery

## Problem Description
When testing RMT driver, I discovered bug in xRingbufferSendFromISR with NOSPLIT buffer which causes different types of errors (assertion crashes, item data corruptions).
The problem appears when ring buffer is almost full and rptr points to something from the beginning >0,
freeptr into the beginning of buffer (0) and wptr < buffer size. 
Below log prints debug info after receiving  in separate task an item from  xRingbufferReceive:
```
Rb size 3072 free 3 rptr 3060 freeptr 3048 wptr 3044
Rb size 3072 free 3 rptr 0 freeptr 3060 wptr 3056
Rb size 3072 free 2963 rptr 12 freeptr 0 wptr 108
Rb size 3072 free 1675 rptr 24 freeptr 12 wptr 1408
````

As we can notice buffer free size unexpectedly jumping from 3 to 2963B where rptr moved over 12B (item size). That happens because when adding an item from ISR and freeptr ==0 and where is enough space left at the end of buffer but after item insertion space left is less than sizeof(buf_entry_hdr_t),
the wptr jumps to the beginning, what means wptr == freeptr. In that case during next insertion from ISR,
api 'thinks' buffer is almost empty and is adding item jumping over rptr:

### Expected Behavior
The wptr shouldn't jump over rptr

The root cause is that fragment of code:
https://github.com/espressif/esp-idf/blob/release/v3.0/components/freertos/ringbuf.c#L142
which moves wptr to the beginning of the buffer. My fast workaround is to keep wptr far from
the end of buffer (>sizeof(buf_entry_hdr_t))
in https://github.com/espressif/esp-idf/blob/release/v3.0/components/freertos/ringbuf.c#L102
change to:
````
    //See if we have enough contiguous space to write the buffer.
    if (rem_len < rbuffer_size + 2*sizeof(buf_entry_hdr_t)) {
````
which disables entering buggy if statement at the end of this function. Although the item size has to be larger than 0.

It seems that whole xRingbuffer driver should be redesigned a bit to fix that problem more gently. 

## Debug Logs

```
Rb size 3072 free 3 rptr 3060 freeptr 3048 wptr 3044
Rb size 3072 free 3 rptr 0 freeptr 3060 wptr 3056
Rb size 3072 free 2963 rptr 12 freeptr 0 wptr 108
Rb size 3072 free 1675 rptr 24 freeptr 12 wptr 1408
```

Dazza0, 2018-05-23:
Ringbuffers will be reworked as part of the upcoming release 3.1

WayneKeenan, 2018-05-23:
A few issues regarding the RingBuffer are stating 'Ringbuffers will be reworked as part of the upcoming release 3.1' .

@Dazza0 Are you able to provide any heads-up guidance, such as:  the public facing API will stay the same as it's only the internals that are being fixed?

Dazza0, 2018-05-24:
Please see 21e085d23592d426aab23efbc017891744833ba5 for the rework. Most of the public API remains unchanged with the following exceptions

**Public API Changes**
- `xRingbufferIsNextItemWrapped()` has been **deprecated**
- `xRingbufferAddToQueueSetWrite()` and `xRingbufferRemoveFromQueueSetWrite()`  **deprecated**. Ring buffer write operations are non-deterministic hence violates correct usage of queue sets.
- `xRingbufferReceiveSplit()` and `xRingbufferReceiveSplitFromISR()` **added** as thread safe way to read from allow-split ringbuffers.
- `vRingbufferGetInfo()` **added**

**Behavioral Changes**
- No-Split and Allow-Split buffers will internally be rounded up to a the nearest 32-bit aligned size.

For full details, please see the [ringbuffer docs](https://esp-idf.readthedocs.io/en/latest/api-reference/system/freertos_additions.html#ring-buffers)

closed 2018-06-19 by igrr