espressif/esp-idf #1957: Auto-reconnection of A2DP Sink device bypasses the outside API and state machine control

adherent88, 2018-05-14:
## Environment
All general environment settings, latest framework etc. 

Development board: ESP32 Wrover Dev Kit, **used as A2DP source.** 

## Problem Description
My application contains one ESP32 used as A2DP source, several headsets as A2DP sinks. I can select a certain sink to send audio to let it play. 

With using the A2DP source example code I am able to register the headsets' addresses and do the connection. 
My problem is the **implementation works only for the first time**. After the headsets connected to Source once, next time it will try to automatically reconnect to ESP32 after powered on. And the re-connection also succeeds. The outside state machine control on higher level is all bypassed. I cannot make use of the connections. 

And what is even worse it that, my idea to realize my application is to connect to one device and send audio to it, if I want to switch to another device, I will first disconnect the current device then connect to the next one. With above described problem it is not possible at all. As after the disconnection the headset will automatically connect to source again. And both are connected, from my application point of view I cannot differentiate the two devices!

Is there a way to solve this issue? Or what I have done is the wrong direction?

Thanks for helping me out.

blueMoodBHD, 2018-05-24:
Hi @adherent88  What your wanted is that, the  headsets will not automatically connect to source after disconnecting, is't it? 
After disconnecting, you can call `esp_bt_gap_remove_bond_device()` to remove the bond device, Then the  headsets will failed to  automatically connect to source.
Thanks.

adherent88, 2018-05-28:
Hi @blueMoodBHD 

Thanks for the reply. Yes, I can use that function to make the reconnection fail. But my intention here is to be able to reconnect to a headset(previously disconnected) as fast as possible. If esp_bt_gap_remove_bond_device is used, I need to then do the discovery again, that will take too long.

After several days struggle, I dived into bluedroid code and commented out certain code which reacts the auto-reconnect request.

blueMoodBHD, 2018-05-28:
Hi @adherent88 
Thanks for your effort， Is it works OK now?

adherent88, 2018-05-28:
Hi @blueMoodBHD

if there is no other elegant solution, I will treat it as solved. :)

closed 2018-05-28 by adherent88