espressif/esp-idf #1958: how the leaf node could send the data to other node around the whole mesh network

NOA-Vernon, 2018-05-15:
Hello,

I have run the demo  https://github.com/espressif/esp-idf/tree/master/examples/mesh/internal_transceiver on my esp32 board. I am trying to use 3 esp32 board attach to the mesh network, the topology is root node, intermediate node, leaf node. However, only the root node can send the data to all the nodes around the mesh network, the intermediate node can send data to the leaf node and itself, but the leaf node only can send data to itself.
I called the esp_mesh_get_routing_table() to get the router table and size, but when the leaf node called this API, the router table only saved the  mesh address of itself, and could communicate with its parent node through calling esp_mesh_get_parent_bssid() to get the parent's address and send the data to.  how the leaf node could send the data to other node around the whole mesh network?
any help is appreciated.
thank you in advance.

nkolban, 2018-05-15:
Are you wanting to broadcast data to all nodes or just send data to a single other node?

yudongchen95, 2018-06-04:
https://www.esp32.com/viewtopic.php?f=21&t=5714