espressif/esp-idf #1959: ESP32-Wrover SPI SRAM fail

onkarnaik511, 2018-05-15:
###############  Environment ##################
- Development Kit: ESP32-Wrover
- Development Env: Make
- Operating System: MacOS
- Power Supply:   Battery

############### Problem Description ##################
We are using the ESP32 along with M95 for a GSM application. Currently testing the device. One of the devices, which seemed to be working fine for over 2 weeks, suddenly started restarting. Looks like an issue with the SPI SRAM. Attached debug logs below.

############### Debug Logs ##################
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5544
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13716
entry 0x40078fb4
I (30) boot: ESP-IDF v3.0-rc1-161-g8856cc0 2nd stage bootloader
I (30) boot: compile time 23:42:45
I (39) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (67) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  3 ota_0            OTA app          00 10 00010000 00160000
I (89) boot:  4 ota_1            OTA app          00 11 00170000 00160000
I (97) boot: End of partition table
I (101) boot: No factory image, trying OTA 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x4479c (280476) map
I (212) esp_image: segment 1: paddr=0x000547c4 vaddr=0x3ffc0000 size=0x06a84 ( 27268) load
I (223) esp_image: segment 2: paddr=0x0005b250 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /Users/numadichw/esp/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (224) esp_image: segment 3: paddr=0x0005b658 vaddr=0x40080400 size=0x049b8 ( 18872) load
I (238) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xcb2ec (832236) map
0x400d0018: _stext at ??:?

I (527) esp_image: segment 5: paddr=0x0012b30c vaddr=0x40084db8 size=0x16a98 ( 92824) load
0x40084db8: heap_caps_realloc at /Users/numadichw/esp/esp-idf/components/heap/./heap_caps.c:314

I (566) esp_image: segment 6: paddr=0x00141dac vaddr=0x400c0000 size=0x00000 (     0) load
I (566) esp_image: segment 7: paddr=0x00141db4 vaddr=0x50000000 size=0x00004 (     4) load
I (589) boot: Loaded app from partition at offset 0x10000
I (589) boot: Disabling RNG early entropy source...
I (592) spiram: SPI RAM mode: flash 40m sram 40m
I (595) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (602) cpu_start: Pro cpu up.
I (606) cpu_start: Starting app cpu, entry point is 0x400813f0
0x400813f0: call_start_cpu1 at /Users/numadichw/esp/esp-idf/components/esp32/./cpu_start.c:215

I (597) cpu_start: App cpu up.
E (1505) spiram: SPI SRAM memory test fail. 131072/131072 writes failed, first @ 3F800000

E (1505) cpu_start: External RAM failed memory test!
abort() was called at PC 0x400815bd on core 0
0x400815bd: call_start_cpu0 at /Users/numadichw/esp/esp-idf/components/esp32/./cpu_start.c:188 (discriminator 1)

Backtrace: 0x4008e904:0x3ffe3be0 0x4008ea2b:0x3ffe3c00 0x400815bd:0x3ffe3c20 0x40078c6a:0x3ffe3c40 0x40078d1d:0x3ffe3c70 0x40078f8e:0x3ffe3cb0 0x400790d6:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x4008e904: __srefill_r at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/refill.c:76

0x4008ea2b: _setlocale_r at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/locale/../../../.././newlib/libc/locale/locale.c:287

0x400815bd: call_start_cpu0 at /Users/numadichw/esp/esp-idf/components/esp32/./cpu_start.c:188 (discriminator 1)

Rebooting...

mahavirj, 2018-10-25:
@onkarnaik511 Do you still observe this issue? Could you please try with latest ESP-IDF codebase once?

mahavirj, 2018-11-06:
@onkarnaik511 Please feel free to re-open in case you observe this with latest code-base or stable release.

closed 2018-11-06 by mahavirj