espressif/esp-idf #1961: ESP32 I2C read issue

KrrishnaShah, 2018-05-15:
I'm using ESP32-IDF for interfacing sht30 temperature and humidity sensor.
The ESP32 master writes successfully on the I2C-bus and receives ack.
But the read operation does not return ack after the writing read address byte on the I2C-bus.

platform details ( platformIO ): 
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf

----------------------------------------------------------------------------------------------------
code details:
----------------------------------------------------------------------------------------------------
	#include <stdio.h>
	#include "sdkconfig.h"
	#include "esp_system.h"
	#include "nvs_flash.h"
	#include "driver/i2c.h"
	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "driver/gpio.h"
	
	static float set_temperature_c (uint16_t _temp)
	{
		return (175*((float)_temp/65535.0f) - 45); // (2^16 -1) = 65535
	}

	static float set_temperature_f (uint16_t _temp)
	{
		return (315*((float)_temp/65535.0f) - 49);  // (2^16 -1) = 65535
	}

	static float set_relative_humidity (uint16_t _humidity)
	{ 
		return (100*((float)_humidity/65535.0f)); // (2^16 -1) = 65535
	}

	uint8_t _sht3x_buf[32] = {0};

	void _sht3x_process(void *pv)
	{
		while(1)
		{
			_i2c_write();

			vTaskDelay(300 / portTICK_PERIOD_MS);
			_i2c_read(6);

			__temperature_c = set_temperature_c((_sht3x_buf[0] << 8) | _sht3x_buf[1]);
			__temperature_f = set_temperature_f((_sht3x_buf[0] << 8) | _sht3x_buf[1]);
			__relative_humidity = set_relative_humidity((_sht3x_buf[3] << 8) | _sht3x_buf[4]);

			TRACE("\r\ntemperature: %f C, %f F\r\nhumidity: %f %c\r\n", __temperature_c , 
											__temperature_f , __relative_humidity , 0x25);

			vTaskDelay(5000 / portTICK_PERIOD_MS);
		}
	}

	uint32_t _i2c_read(uint32_t rx_len)
	{
		i2c_cmd_handle_t cmd = i2c_cmd_link_create();

		i2c_master_start(cmd);
		i2c_master_write_byte(cmd, ( _sht->address << 1 ) | I2C_MASTER_READ, 1);

		if (rx_len > 1)
		{
			i2c_master_read(cmd, _sht3x_buf, rx_len - 1, 0x01);
		}

		i2c_master_read_byte(cmd, _sht3x_buf + rx_len - 1, 0x00);
		i2c_master_stop(cmd);

		esp_err_t esp_i2c_err = i2c_master_cmd_begin(_sht->i2c_num, cmd, 1000);
		if (esp_i2c_err == ESP_OK)
		{
			ret = rx_len;
		}

		TRACE("i2c master read error: %d", esp_i2c_err);
		i2c_cmd_link_delete(cmd);

		return ret;
	}

	uint32_t _i2c_write(void)
	{
		uint32_t _len = 0;

		_sht3x_buf[_len] = (0x44 << 1) | I2C_MASTER_WRITE;
		_sht3x_buf[_len++] = 0xE0;
		_sht3x_buf[_len++] = 0x00;

		_sht3x_buf
		i2c_cmd_handle_t cmd = i2c_cmd_link_create();

		i2c_master_start(cmd);
		i2c_master_write(cmd, _sht3x_buf, _len, 1);
		i2c_master_stop(cmd);

		esp_err_t esp_err = i2c_master_cmd_begin(_sht->i2c_num, cmd, 1000);
		if (esp_err == ESP_OK)
		{
			ret = 1;
		}

		printf("i2c_send_error: %d", esp_err);
		i2c_cmd_link_delete(cmd);

		return ret;
	}
	
	void app_main()
	{
		nvs_flash_init();
		
		i2c_config_t i2c_master_config;
		i2c_master_config.mode = I2C_MODE_MASTER;
		i2c_master_config.sda_io_num = 22;
		i2c_master_config.sda_pullup_en = GPIO_PULLUP_ENABLE;
		i2c_master_config.scl_io_num = 23;
		i2c_master_config.scl_pullup_en = GPIO_PULLUP_ENABLE;
		i2c_master_config.master.clk_speed = 100000;

		i2c_param_config(0, &i2c_master_config);
		i2c_driver_install(0, i2c_master_config.mode, 0, 0, 0);

		xTaskCreate(_sht3x_process, "sht process", 2048, NULL, 2, NULL);
	}

kewalmshah, 2018-05-15:
HI @KrrishnaShah ! In i2c_read esp32 sends the ack/nack value. The ACK value for i2c protocol is 0 and NACK value is 1.
In your i2c_read implementation I see NACK immediately  in i2c_master_read:
```
if (rx_len > 1)
{
    i2c_master_read(cmd, _sht3x_buf, rx_len - 1, 0x01);
}
i2c_master_read_byte(cmd, _sht3x_buf + rx_len - 1, 0x00);
```
It should actually be:
```
if (rx_len > 1)
{
    i2c_master_read(cmd, _sht3x_buf, rx_len - 1, 0x00 /*ACK Value*/);
}
i2c_master_read_byte(cmd, _sht3x_buf + rx_len - 1, 0x01 /*NACK Value*/);
```

KrrishnaShah, 2018-05-16:
dear @kewalmshah,

Currently I'm receiving problem at this position:

i2c_master_write_byte(cmd, ( _sht->address << 1 ) | I2C_MASTER_READ, 1);

-> i'm sending i2c read address but in return i'm getting NACK form
[image]

but I'm getting ack from the 
i2c_master_write_byte(cmd, ( _sht->address << 1 ) | I2C_MASTER_WRITE, 1);
[image]

kewalmshah, 2018-05-16:
Hi @KrrishnaShah This looks more of slave side issue. Since esp32 is in master and writing it expects ack from the the slave. Can you check the slave side ?

KrrishnaShah, 2018-05-16:
dear sir,

sht3x slave is running fine with efm32gg
not any problems at 400KHz i2c clock frequency

negativekelvin, 2018-05-16:
Have you tried to do it in one transaction like https://github.com/espressif/esp-iot-solution/blob/413acc39f45c239cb5e16b34f9093b406343a651/components/i2c_devices/sensor/bh1750/bh1750.c#L136-L140

negativekelvin, 2018-05-16:
I think you actually circled the r/w bit but doesn't change the results.  You seem to have a lot of cross coupling of data and clock lines did you use external pullups?

KrrishnaShah, 2018-05-16:
Initially i used internal pullups but it did not work.
now i'm using external pullups with 10Kohms

Then it may the reason that my write command is not accepted by the slave and hence the salve respond with the NACK.

I've sent the command to the slave and then trying to receive the data.

negativekelvin, 2018-05-16:
You only sent fetch command (E0 00) not measurement command

KrrishnaShah, 2018-05-16:
Thank you sir
the problem has been solved..

i had commented all the previous command sequence and was only trying to fetch the data.
but when i un-commented the previous command sequence as per your suggestions it ran well.

thank you again sir for your guidance ...

negativekelvin, 2018-05-16:
You're welcome please mark this issue closed

closed 2018-05-16 by KrrishnaShah