espressif/esp-idf #1963: espcoredump Error

rsnbusa, 2018-05-15:
Hi everybody.

The **espcoredump** is extremely valuable tool _specially_ when you want to save the crash and analyze later the reason for it, when it is a randmom error.

My esp32 app has some random crash so I configured it to save the crash to Flash so I could can pick it up and analyze later.  

Having done so many things to try to get the core dump from flash and not being able to, I tried the Blink app to see if it is something in my app. 

To test it I have an int *p and after a delay of 1000ms *p=0; crash, dumps to the monitor an stops (with the halt option set in menuconfig).

Blink
Guru Meditation Error: Core  0 panic'ed (StoreProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d1f70  PS      : 0x00060530  A0      : 0x00000000  A1      : 0x3ffb66c0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb66a0  
A10     : 0x000000c8  A11     : 0x00000000  A12     : 0x3ffb1068  A13     : 0x3ffb6748  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x400d1f70:0x3ffb66c0 0x7ffffffd:0x3ffb66e0

CPU halted.

When I use the the espcoredump I get (invariably, be it Blink or my app or any app)

```
rsnDesktop:~ rsn$ /Users/rsn/esp/esp-idf-v3.0/components/espcoredump/espcoredump.py -p /dev/tty.SLAB_USBtoUART -b 921600 info_corefile /Users/rsn/esp/esp-idf-v3.0/examples/get-started/blink/build/blink.elf 
espcoredump.py v0.1-dev
esptool.py v2.1
Connecting........_
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
16 (100 %)
16 (100 %)
Read 16 bytes at 0x110000 in 0.0 seconds (13.7 kbit/s)...
Hard resetting...

A fatal error occurred: Invalid start magic number!
```

When dumping to the monitor and running the espcoredump It works for Blink and partially for my app.

For my app it goes crazy. Attached (GarageDump)

So what now? Have been reading this for months and havent found a solution. Is it the esp32 model? The Partition? Espcoredump?

I would kindly like to insist in the importance to be able to save the crash to flash. It is the only way to debug a random crash.

Thanks in advance.

GarageDump.zip is my app dump to uart.
blink.c.zip is the modified blink app with the induced Crash
cored.txt is the Blink uart coredump
[GarageDump.zip](https://github.com/espressif/esp-idf/files/2006842/GarageDump.zip)
[blink.c.zip](https://github.com/espressif/esp-idf/files/2006843/blink.c.zip)
[cored.txt](https://github.com/espressif/esp-idf/files/2006844/cored.txt)

gerekon, 2018-05-22:
@rsnbusa
I checked your blink app with the latest code base from github master branch (commit 92c469b599). It works at my side. Could you check with that commit? Also could you provide me with your sdkconfig file?
What type of ESP32 board do you use? What partition table config file do you use?

Thanks,
Alexey (Espressif)

rsnbusa, 2018-05-22:
Hi Alexey.

Max Osx 10.13.4 (17E202), 32GB Ram, 1TB SSD, Intel i-7, GT 710
Chip is ESP32D0WDQ6 (revision 0) see espcoredump message

For this test I created a new dir and made a new esp32 environment by git cloning esp-idf latest version that is on the github expressif (today May 21-2018)

xtensa version 3.0

**Build process**
make menuconfig
make erase_flash
make clean
make flash

**Attached files**
mipartition.csv.zip [mipartition.csv.zip](https://github.com/espressif/esp-idf/files/2028441/mipartition.csv.zip)
sdkconfig.zip [sdkconfig.zip](https://github.com/espressif/esp-idf/files/2028453/sdkconfig.zip)
Blink.c.zip [blink.c.zip](https://github.com/espressif/esp-idf/files/2028442/blink.c.zip)
Screen 1= Coolterm of Blink where it shows booting, Start Message, Ready to dump message  and the backtrace. [Screen 1.zip](https://github.com/espressif/esp-idf/files/2028447/Screen.1.zip)
Screen 2= Terminal screen of the espcoredump try [Screen 2.zip](https://github.com/espressif/esp-idf/files/2028446/Screen.2.zip)
Screen 3= Backtrace of Blink using gdb. [Screen 3.zip](https://github.com/espressif/esp-idf/files/2028444/Screen.3.zip)
git log.zip [commitlog.txt.zip](https://github.com/espressif/esp-idf/files/2028471/commitlog.txt.zip)

**Result**
Same result, _**Invalid start magic number**_. 
Tried with two other esp32, same result.
While the gdb backtrace is ok on all esp32 I tried.

Would appreciate your help or guidance.

RSN

rsnbusa, 2018-05-22:
Sorry forgot to say I did change the baud rate to as low as 115200. And yes, the commit is the same 92c469b

Regards.

RSN

gerekon, 2018-05-23:
@rsnbusa 
You placed core dump partition to the custom location in the flash. In this case you need to specify that custom offset in the arguments to `espcoredump` script. Your command line should look like 
`espcoredump.py -p /dev/tty.SLAB_USBtoUART info_corefile --off 2162688 /Users/rsn/esp/esp-idf-v3.0/examples/get-started/blink/build/blink.elf`

Note that core dump partition offset should be in decimal.

rsnbusa, 2018-05-23:
Great! Thanks Alexey.

What is the "default" offset for coredump in a partition table? Or should one always give the offset when reading the Flash using espcoredumping?

Did try it, it worked but complain about the GDB not being able to continue. (Attached).

Did use the pc (program counter) and manually started the GDB and it was spot on. 

```
0x400d1c57 is in blink_task (/Users/rsn/esp/esp-idf-v3.0/examples/get-started/blink/main/blink.c:37).
32	    while(1) {
33	        /* Blink off (output low) */
34	        gpio_set_level(BLINK_GPIO, 0);
35	        printf("Ready to dump\n");
36	        vTaskDelay(2000 / portTICK_PERIOD_MS);
37	        *p=0;
38	        printf("Should not see this\n");
39	        /* Blink on (output high) */
40	        gpio_set_level(BLINK_GPIO, 1);
41	        vTaskDelay(1000 / portTICK_PERIOD_MS);
(gdb) 
```
Also confirmed it in my app where I force a dump (*p=0). Same GDB problem but again the pc gives the perfect location.

Anything I can do with the GDB problem?

Regards.

RSN
[espcoredumpBlink.txt.zip](https://github.com/espressif/esp-idf/files/2031793/espcoredumpBlink.txt.zip)

gerekon, 2018-05-23:
> What is the "default" offset for coredump in a partition table? 

The default is 0x110000, which is used when you use partition table files supplied with IDF.
I think in the next version of the script we should add automatic detection of the core dump partition offset in flash.

> Anything I can do with the GDB problem?

Unfortunately no. It is known problem of GDB. We have a fix for it. I think it will be solved in the next toolchain update.

Regards,
Alexey

gerekon, 2018-05-29:
@FayeY I suppose this issue should be closed, since the main problem has been solved.

closed 2018-05-29 by FayeY

rsnbusa, 2018-05-29:
Yes thank you.