espressif/esp-idf #1964: Services in iOS/Android not updating after Service Change by ESP

jim-ber, 2018-05-16:
### Expected Behavior

The project I am currently working on requires the ESP32 to act as a BLE server and it has to be able to add/remove services as we expose or hide capabilities of the device. The device acts as a hub that connects to and controls other devices (using a separate radio). As these devices are connected to this hub, it would expose or remove services for those connected devices.

The hub has to connect to a phone (iOS or Android) and indicate to the client when services have changed on the BLE server. The client device, after receiving the indication, should clear the cache and repopulate its services.

### Actual Behavior

When using the service changed example code provided by @Weijian-Espressif in Issue ([#1777](https://github.com/espressif/esp-idf/issues/1777)), iOS and Android (using LightBlue and Nordic nRF Connect apps) do not clear the cache and repopulate the servers services. If I disconnect and reconnect the services show up as expected but that defeats the purpose for this application.

I have a theory that the this is caused by the lack of the Client Characteristic Configuration Descriptor (0x2902) associated with the Service Changed Characteristic (0x2A05) created by the ESP-IDF.

### Code to reproduce this issue

[service_change_demo.zip](https://github.com/espressif/esp-idf/files/1859163/service_change_demo.zip)

Weijian-Espressif, 2018-05-16:
@jim-ber what is the commit ID of IDF?

jim-ber, 2018-05-16:
#1777

jim-ber, 2018-05-21:
Do you have an example iOS/Android app code that I can use to test this?

Weijian-Espressif, 2018-05-22:
[BabyBluetooth-master.zip](https://github.com/espressif/esp-idf/files/2024789/BabyBluetooth-master.zip)
this is the app log:
2018-05-22 10:46:23.417975+0800 BabyBluetoothAppDemo[439:324479] 搜索到了设备:HH_GATTS_DEMO
2018-05-22 10:46:23.424277+0800 BabyBluetoothAppDemo[439:324479] 搜索到了设备:HH_GATTS_DEMO
2018-05-22 10:46:23.510453+0800 BabyBluetoothAppDemo[439:324479] setBlockOnCancelScanBlock
2018-05-22 10:46:23.567958+0800 BabyBluetoothAppDemo[439:324479] viewWillDisappear
2018-05-22 10:46:25.558383+0800 BabyBluetoothAppDemo[439:324479] >>>已切换到peripheralView
2018-05-22 10:46:26.116250+0800 BabyBluetoothAppDemo[439:324479] 搜索到服务:00FF
2018-05-22 10:46:26.125927+0800 BabyBluetoothAppDemo[439:324479] >>>beats at :Tue May 22 10:46:26 2018
2018-05-22 10:46:26.239091+0800 BabyBluetoothAppDemo[439:324479] ===service name:00FF
2018-05-22 10:46:26.239311+0800 BabyBluetoothAppDemo[439:324479] add indexpath in row:0, sect:0
2018-05-22 10:46:26.239403+0800 BabyBluetoothAppDemo[439:324479] add indexpath in row:1, sect:0
2018-05-22 10:46:26.239489+0800 BabyBluetoothAppDemo[439:324479] add indexpath in row:2, sect:0
2018-05-22 10:46:26.240610+0800 BabyBluetoothAppDemo[439:324479] (
    "<CBCharacteristic: 0x14fe9f4d0, UUID = FF01, properties = 0x1A, value = (null), notifying = NO>",
    "<CBCharacteristic: 0x151954580, UUID = FF02, properties = 0x2, value = (null), notifying = NO>",
    "<CBCharacteristic: 0x151b1a220, UUID = FF03, properties = 0x8, value = (null), notifying = NO>"
)
2018-05-22 10:46:26.324869+0800 BabyBluetoothAppDemo[439:324479] characteristic name:FF01 value is:<11223344>
2018-05-22 10:46:26.384287+0800 BabyBluetoothAppDemo[439:324479] characteristic name:FF02 value is:<11223344>
2018-05-22 10:46:26.445779+0800 BabyBluetoothAppDemo[439:324479] characteristic name:FF03 value is:<11223344>
2018-05-22 10:46:26.505738+0800 BabyBluetoothAppDemo[439:324479] ===characteristic name:00FF
2018-05-22 10:46:26.506027+0800 BabyBluetoothAppDemo[439:324479] CBDescriptor name is :Client Characteristic Configuration
2018-05-22 10:46:26.508695+0800 BabyBluetoothAppDemo[439:324479] ===characteristic name:00FF
2018-05-22 10:46:26.564847+0800 BabyBluetoothAppDemo[439:324479] ===characteristic name:00FF
2018-05-22 10:46:26.624655+0800 BabyBluetoothAppDemo[439:324479] Descriptor name:FF01 value is:0
2018-05-22 10:46:29.066579+0800 BabyBluetoothAppDemo[439:324479] >>>已切换到CharacteristicView
2018-05-22 10:46:29.147576+0800 BabyBluetoothAppDemo[439:324479] >>>beatsBreak :Tue May 22 10:46:29 2018
2018-05-22 10:46:29.147693+0800 BabyBluetoothAppDemo[439:324479] setBlockOnBeatsBreak call
2018-05-22 10:46:29.203993+0800 BabyBluetoothAppDemo[439:324479] CharacteristicViewController Descriptor name:FF01 value is:0
2018-05-22 10:46:29.263877+0800 BabyBluetoothAppDemo[439:324479] CharacteristicViewController Descriptor name:FF01 value is:0
2018-05-22 10:46:29.650742+0800 BabyBluetoothAppDemo[439:324479] viewWillDisappear
2018-05-22 10:46:31.184662+0800 BabyBluetoothAppDemo[439:324479] >>>didUpdateNotificationStateForCharacteristic
2018-05-22 10:46:31.184983+0800 BabyBluetoothAppDemo[439:324479] >>>uuid:FF01,isNotifying:isNotifying
2018-05-22 10:46:31.185369+0800 BabyBluetoothAppDemo[439:324479] uid:FF01,isNotifying:on
2018-05-22 10:46:31.185664+0800 BabyBluetoothAppDemo[439:324479] =======receive service change event!!!!!!!!!!!!!!!!!!!!!!!

chegewara, 2018-05-26:
Some questions about logs.
```
I (80679) BT_GATT: **GATTS_CreateService**

D (80683) BT_GATT: gatt_add_pending_new_srv_start
D (80688) BT_GATT: enqueue a new pending new srv start
D (80693) BT_GATT: Add a new srv chg item

D (80697) BT_GATT: allocate_svc_db_buf allocating extra buffer
D (80703) BT_GATT: gatts_init_service_db

D (80707) BT_GATT: s_hdl = 70 num_handle = 15

D (80712) BT_GATT: add_service_declaration
D (80716) BT_GATT: allocate attr 20 bytes

D (80720) BT_GATT: =====> handle = [0x0046] uuid16 = [0x2800] perm=0x01

D (80727) BT_BTC: btc_transfer_context msg 1 2 7 0x3ffd5770

D (80733) BT_BTC: btc_gatts_arg_deep_free Unhandled deep free 2

D (80739) BT_BTC: btc_task msg 0 2 5 0x3ffd88ac

D (80744) BT_APPL: BTA got event 0x2009

D (80748) BT_APPL: bta_gatts_find_srvc_cb_by_srvc_id  service_id=70
D (80754) BT_APPL: bta_gatts_find_srvc_cb_by_srvc_id  found service cb index =2
I (80762) BT_GATT: GATTS_StartService
D (80766) BT_GATT: total buffer in db [1]
D (80770) BT_GATT: allocated i_sreg=4

D (80774) BT_GATT: s_hdl=70 e_hdl=84 type=0x2800 svc_inst=2 sdp_hdl=0x0

D (80781) BT_GATT: **gatt_sr_is_new_srv_chg: Yes**
D (80786) BT_GATT: gatt_proc_srv_chg
D (80789) BT_GATT: gatt_set_srv_chg
D (80793) BT_GATT: gatt_find_the_connected_bda start_idx=0
D (80798) BT_GATT: gatt_find_the_connected_bda bda :77-04-61-82-d1-72
D (80805) BT_GATT: **gatt_find_the_connected_bda found=1** found_idx=0
D (80811) BT_GATT: gatt_is_srv_chg_ind_pending is_queue_empty=1
D (80817) BT_GATT: srv_chg_ind_pending = 0
D (80821) BT_GATT: gatt_send_srv_chg_ind
D (80825) BT_GATT: gatt_get_ch_state: ch_state=4
I (80830) BT_GATT: GATT_GetConnIdIfConnected status=1

I (80836) BT_GATT: GATTS_HandleValueIndication
```
below is log continuation, its part of code where we are receiving ESP_GATTS_CREATE_EVT event and we can complete service creation by assigning attribute handle. Is this correct order? What will happen if client application will query services before "my" esp32 server application complete attribute handle assign?
```
I (80841) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 77046182d172
D (80849) BT_L2CAP: l2c_link_check_send_pkts
D (80853) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (80860) BT_L2CAP: l2c_link_send_to_lower
D (80864) BT_L2CAP: TotalWin=9,Hndl=0x0,Quota=10,Unack=1,RRQuota=0,RRUnack=0
D (80871) BT_OSI: alarm_cbs_lookfor_available 1 0x3ffc53f0

D (80877) BT_OSI: osi_alarm_cancel failed to stop timer, err 0x103

D (80883) BT_GATT: gatt_find_the_connected_bda start_idx=1
D (80889) BT_GATT: **gatt_find_the_connected_bda found=0** found_idx=4
D (80895) BT_APPL: bta_gatts_start_service service_id= 70
D (80901) BT_BTC: btc_transfer_context msg 1 2 12 0x3ffd5780

D (80907) BT_BTC: btc_gatts_arg_deep_free Unhandled deep free 5

D (80913) BT_BTC: btc_task msg 1 2 7 0x3ffd87fc

D (80918) BLEDevice: gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_CREATE_EVT
D (80925) BLEUtils: GATT ServerEvent: ESP_GATTS_CREATE_EVT
D (80931) BLEUtils: [status: ESP_GATT_OK, service_handle: 70 0x46, service_id: [uuid: 03b80e5a-ede8-4b33-a751-6ce34ec4c700, inst_id: 2]]
D (80943) BLEServer: >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
E (80950) BLEService: 70
```
We have two lines `gatt_find_the_connected_bda found=` one with found devices 0 and one with 1. I know why is that, but is this right? Should that one `gatt_find_the_connected_bda start_idx=1` start with idx=1 and not 0? 

`srv_chg_ind_pending = 0`
Does that means that client app (nRF connect) does not register for service change indications?

PS i cant use bold with code markdown, so few lines is with extra **

Weijian-Espressif, 2018-05-29:
@chegewara sorry,  I don't have android app code for this.  iOS APP and ESP32 client  work normally.

closed 2018-05-30 by FayeY

jim-ber, 2018-05-30:
@Weijian-Espressif Thanks for the code. I changed it a bit and made it refresh automatically when receiving a service changed request. When I am done fine tuning I'll post it here.

@chegewara The product I am working on will have to have an android app to test this. When I have a working prototype I'll post it here as well.

michaelboeding, 2024-05-25:
Did you ever get this working?

weiyuhannnn, 2024-07-02:
@michaelboeding  Do you still have this issue? If so, could you please open a new GitHub issue?  Thx!

michaelboeding, 2024-07-02:
Yes the issue still exists. I just included all the services I needed at the start and wrote my own provisioning protocol. So I removed the esp provisioning since those services stayed cashed and were never removed when using an iPhone.