espressif/esp-idf #1967: CMake: menuconfig on windows fails with COMPONENT_KCONFIGS_PROJBUILD being empty

hak8or, 2018-05-17:
- [x] I have read the documentation [ESP-IDF Programming Guide](http://esp-idf.readthedocs.io/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Core (if using chip or module): [ESP32-Wrover-I|ESP32-Wrover|ESP32-PICO-D4|ESP32-SOLO-I|ESP32-WROOM-32U|ESP-WROOM-32D|ESP-WROOM32|ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 829e6c4
- Development Env:		[CLion/CLI]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description

Running ```idf.py menuconfig``` seems to just hang. For windows I got mconf.exe from [here](http://distortos.org/downloads/tools/kconfig-frontends-windows/) since I couldn't seem to find it elsewhere. I am **NOT** using the standard msys environment espressif offers because that would clutter my windows machine far too much.

Instead I manually got the toolchain and put it into my path, installed cmake and python 2.7 as native windows applications, and am able to build with both ```idf.py build flash monitor --port COM4``` with no issues. I am even able to use CLion successfully. Menuconfig for whatever reason does not seem to work though.

### Expected Behavior

Menuconfig opens

### Actual Behavior

Interface hangs.

### Steps to repropduce

```bash
# Create a folder for this
mkdir issue
cd issue

# Copy an example from IDF
 cp -r ..\idf\examples\get-started\blink\* .

# This gives: Python 2.7.15
python -v

# This gives: cmake version 3.11.1
cmake --version

# This gives: 1.8.2
ninja --version

# This gives: IDF_PATH=C:\code\BookKeeper\idf
c:\code\BookKeeper>set IDF_PATH

# Attempt to run idf.py menuconfig, seems to fail.
idf.py menuconfig
```

## Debug Logs

```
c:\code\BookKeeper\issue>idf.py menuconfig
Running cmake in directory c:\code\BookKeeper\issue\build
Executing "cmake -G Ninja --warn-uninitialized c:\code\BookKeeper\issue"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.10.1.windows.1")
-- Building empty aws_iot component due to configuration
-- Component names: app_trace xtensa-debug-module app_update bootloader_support spi_flash mbedtls lwip ethernet tcpip_adapter micro-ecc aws_iot jsmn bootloader bt coap nvs_flash console cxx driver esp-tls nghttp esp32 log pthread vfs wpa_supplicant esp_adc_cal esptool_py expat fatfs wear_levelling sdmmc freertos heap idf_test json libsodium mdns newlib openssl partition_table smartconfig soc spiffs ulp
-- Component paths: C:/code/BookKeeper/idf/components/app_trace;C:/code/BookKeeper/idf/components/xtensa-debug-module;C:/code/BookKeeper/idf/components/app_update;C:/code/BookKeeper/idf/components/bootloader_support;C:/code/BookKeeper/idf/components/spi_flash;C:/code/BookKeeper/idf/components/mbedtls;C:/code/BookKeeper/idf/components/lwip;C:/code/BookKeeper/idf/components/ethernet;C:/code/BookKeeper/idf/components/tcpip_adapter;C:/code/BookKeeper/idf/components/micro-ecc;C:/code/BookKeeper/idf/components/aws_iot;C:/code/BookKeeper/idf/components/jsmn;C:/code/BookKeeper/idf/components/bootloader;C:/code/BookKeeper/idf/components/bt;C:/code/BookKeeper/idf/components/coap;C:/code/BookKeeper/idf/components/nvs_flash;C:/code/BookKeeper/idf/components/console;C:/code/BookKeeper/idf/components/cxx;C:/code/BookKeeper/idf/components/driver;C:/code/BookKeeper/idf/components/esp-tls;C:/code/BookKeeper/idf/components/nghttp;C:/code/BookKeeper/idf/components/esp32;C:/code/BookKeeper/idf/components/log;C:/code/BookKeeper/idf/components/pthread;C:/code/BookKeeper/idf/components/vfs;C:/code/BookKeeper/idf/components/wpa_supplicant;C:/code/BookKeeper/idf/components/esp_adc_cal;C:/code/BookKeeper/idf/components/esptool_py;C:/code/BookKeeper/idf/components/expat;C:/code/BookKeeper/idf/components/fatfs;C:/code/BookKeeper/idf/components/wear_levelling;C:/code/BookKeeper/idf/components/sdmmc;C:/code/BookKeeper/idf/components/freertos;C:/code/BookKeeper/idf/components/heap;C:/code/BookKeeper/idf/components/idf_test;C:/code/BookKeeper/idf/components/json;C:/code/BookKeeper/idf/components/libsodium;C:/code/BookKeeper/idf/components/mdns;C:/code/BookKeeper/idf/components/newlib;C:/code/BookKeeper/idf/components/openssl;C:/code/BookKeeper/idf/components/partition_table;C:/code/BookKeeper/idf/components/smartconfig;C:/code/BookKeeper/idf/components/soc;C:/code/BookKeeper/idf/components/spiffs;C:/code/BookKeeper/idf/components/ulp
Creating config file C:/code/BookKeeper/issue/sdkconfig...
-- Executing toolchain file!
-- Executing toolchain file!
-- The ASM compiler identification is GNU
-- Found assembler: C:/code/BookKeeper/toolchain/bin/xtensa-esp32-elf-gcc.exe
[... 10 lines cut ...]
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building empty aws_iot component due to configuration
-- Adding linker script C:/code/BookKeeper/issue/build/esp32/esp32_out.ld
-- Adding linker script C:/code/BookKeeper/idf/components/esp32/ld/esp32.common.ld
-- Adding linker script C:/code/BookKeeper/idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script C:/code/BookKeeper/idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/code/BookKeeper/idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Component libraries: ;app_trace;xtensa-debug-module;app_update;bootloader_support;spi_flash;mbedtls;lwip;ethernet;tcpip_adapter;micro-ecc;aws_iot;jsmn;bt;coap;nvs_flash;console;cxx;driver;esp-tls;nghttp;esp32;log;pthread;vfs;wpa_supplicant;esp_adc_cal;expat;fatfs;wear_levelling;sdmmc;freertos;heap;idf_test;json;libsodium;mdns;newlib;openssl;smartconfig;soc;spiffs;ulp
-- Configuring done
-- Generating done
-- Build files have been written to: C:/code/BookKeeper/issue/build
Running ninja in directory c:\code\BookKeeper\issue\build
Executing "ninja menuconfig"...
[0/1] cmd.exe /C "cd /D C:\code\BookKeeper\issue\build && python ...kconfig C:/code/kconfig/mconf.exe C:/code/BookKeeper/idf/Kconfig"
C:/code/BookKeeper/idf/Kconfig:39: can't open file ""
FAILED: CMakeFiles/menuconfig
cmd.exe /C "cd /D C:\code\BookKeeper\issue\build && python C:/code/BookKeeper/idf/tools/kconfig_new/confgen.py --kconfig C:/code/BookKeeper/idf/Kconfig --config C:/code/BookKeeper/issue/sdkconfig --defaults C:/code/BookKeeper/issue/sdkconfig.defaults --create-config-if-missing --env "COMPONENT_KCONFIGS= C:/code/BookKeeper/idf/components/app_trace/Kconfig C:/code/BookKeeper/idf/components/spi_flash/Kconfig C:/code/BookKeeper/idf/components/mbedtls/Kconfig C:/code/BookKeeper/idf/components/lwip/Kconfig C:/code/BookKeeper/idf/components/ethernet/Kconfig C:/code/BookKeeper/idf/components/tcpip_adapter/Kconfig C:/code/BookKeeper/idf/components/aws_iot/Kconfig C:/code/BookKeeper/idf/components/bt/Kconfig C:/code/BookKeeper/idf/components/driver/Kconfig C:/code/BookKeeper/idf/components/esp32/Kconfig C:/code/BookKeeper/idf/components/log/Kconfig C:/code/BookKeeper/idf/components/pthread/Kconfig C:/code/BookKeeper/idf/components/esp_adc_cal/Kconfig C:/code/BookKeeper/idf/components/fatfs/Kconfig C:/code/BookKeeper/idf/components/wear_levelling/Kconfig C:/code/BookKeeper/idf/components/freertos/Kconfig C:/code/BookKeeper/idf/components/heap/Kconfig C:/code/BookKeeper/idf/components/libsodium/Kconfig C:/code/BookKeeper/idf/components/openssl/Kconfig C:/code/BookKeeper/idf/components/spiffs/Kconfig" --env "COMPONENT_KCONFIGS_PROJBUILD= C:/code/BookKeeper/idf/components/bootloader/Kconfig.projbuild C:/code/BookKeeper/idf/components/esptool_py/Kconfig.projbuild C:/code/BookKeeper/idf/components/partition_table/Kconfig.projbuild C:/code/BookKeeper/issue/main/Kconfig.projbuild" --output config C:/code/BookKeeper/issue/sdkconfig && "C:\Program Files\CMake\bin\cmake.exe" -E env "COMPONENT_KCONFIGS= C:/code/BookKeeper/idf/components/app_trace/Kconfig C:/code/BookKeeper/idf/components/spi_flash/Kconfig C:/code/BookKeeper/idf/components/mbedtls/Kconfig C:/code/BookKeeper/idf/components/lwip/Kconfig C:/code/BookKeeper/idf/components/ethernet/Kconfig C:/code/BookKeeper/idf/components/tcpip_adapter/Kconfig C:/code/BookKeeper/idf/components/aws_iot/Kconfig C:/code/BookKeeper/idf/components/bt/Kconfig C:/code/BookKee [... cut ...] _py/Kconfig.projbuild C:/code/BookKeeper/idf/components/partition_table/Kconfig.projbuild C:/code/BookKeeper/issue/main/Kconfig.projbuild" KCONFIG_CONFIG=C:/code/BookKeeper/issue/sdkconfig C:/code/kconfig/mconf.exe C:/code/BookKeeper/idf/Kconfig"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Opening that line (39) in Kconfig is ```source "$COMPONENT_KCONFIGS_PROJBUILD"```. I would be happy to debug more but I am not familiar with how the build system for this works.

projectgus, 2018-05-17:
Hi @hak8or,

Thanks for the very comprehensive bug report. The root cause here is that we use a patched mconf to support the syntax shown on Kconfig line 39 (as per the error). I realise now that it's probably time we renamed "IDF mconf" to avoid confusion.

>  For windows I got mconf.exe from here since I couldn't seem to find it elsewhere. I am NOT using the standard msys environment espressif offers because that would clutter my windows machine far too much.

As you're using the CMake branch then you don't need MSYS at all. But you will need our version of mconf. You can get this one of two ways.

* Run the ESP-IDF Tools Installer (this is not MSYS):
https://docs.espressif.com/projects/esp-idf/en/feature-cmake/get-started/windows-setup.html#esp-idf-tools-installer
(It sounds like you have everything else from the Tools Installer already installed, so this is probably not worth doing at this stage.)

* Download the mconf binary by itself. This is supposed to be [linked from this docs page](https://docs.espressif.com/projects/esp-idf/en/feature-cmake/get-started/windows-setup-scratch.html#mconf), but I just realised this link is broken. It should point to https://github.com/espressif/kconfig-frontends/releases/ . Very sorry about that.

We will fix the link and also look into renaming this binary to avoid other people running into this problem in the future.

hak8or, 2018-05-17:
Ah, thank you so much for the prompt reply! I copied your mconf binary (the entire folder contents) from your kconfig-frontends repo (v4.6.0.0-idf-20180319) and typing ```idf.py menuconfig``` now works!

Also, the installer you provide in the cmake branch's docs looks fantastic! If I didn't (incorrectly) assume that the windows section of the docs was the same as in the non cmake branch I would have used it, ah well.

For some weird reason though, arrow keys don't seem to work in your menuconfig. I am able to use the letter keys, for example pressing "B" to enter "Bootloader Config", and the Tab key to go through "Select", "Exit", "Load", etc also works. Pressing the escape key and enter key also works. It seems to just be arrow keys which aren't working. Would you like me to create a new issue for this?

projectgus, 2018-05-17:
> For some weird reason though, arrow keys don't seem to work in your menuconfig. I am able to use the letter keys, for example pressing "B" to enter "Bootloader Config", and the Tab key to go through [...]

Please do, you can link back to this issue for context/environment/etc.

Are you using Windows Command Prompt or something else?