espressif/esp-idf #1970: NFC PN532 with esp32 dev borad I2C

KrrishnaShah, 2018-05-17:
This is a code block I've written for the pn532 NFC module to communicate with ESP32 through I2C port.

Problem: I'm getting NACK form the SLA-W. I'm unable to move froward from this point. It took me around 3 days but still failed.

SHT3x i2c salve is working fine with esp32 upto 400KHz.

The PN532 NFC module run quite good with arduino at 100KHz. Device ran ok with efm32gg but only at 10KHz and with LPC1768 at 5KHz. So I came to try NFC with ESP32 but failed multiple time.

I really appreciate any kind of help form any one here.

CODE:
--------------------------------------------------------------------------
	#include <stdio.h>
	#include "sdkconfig.h"
	#include "esp_system.h"
	#include "nvs_flash.h"
	#include "driver/i2c.h"
	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "driver/gpio.h"

	static uint32_t __blinky_interval = 500;

	void __app_set_blink_interval(uint32_t interval)
	{
		__blinky_interval = interval;
	}

	static void __blinky(void *pv)
	{
		static uint32_t led_gpio_crrent_state = 0;

		gpio_config_t led_gpio =    {
										GPIO_SEL_2,
										GPIO_MODE_OUTPUT,
										GPIO_PULLUP_DISABLE,
										GPIO_PULLDOWN_DISABLE,
										GPIO_INTR_DISABLE
									};

		if (gpio_config(&led_gpio) == ESP_ERR_INVALID_ARG)
		{
			printf("failed to configure GPIO_2\n");
		}

		printf("configured GPIO_2\n");

		while(1)
		{
			printf("free heap: %d\n", xPortGetFreeHeapSize());
			led_gpio_crrent_state ^= 1;
			gpio_set_level(GPIO_NUM_2, led_gpio_crrent_state);
			vTaskDelay(__blinky_interval / portTICK_PERIOD_MS);

			fflush(stdout);
		}
	}

    #include <string.h>

	xSemaphoreHandle pn532_p70_intr_semphr = NULL;

	uint32_t pn532_p70_irq 	= GPIO_NUM_23;
	uint32_t pn532_rst		= GPIO_NUM_22;

	uint32_t i2c_num 		= 0;
	uint32_t i2c_clock 		= 100000;
	uint32_t i2c_sda		= GPIO_NUM_18;
	uint32_t i2c_scl		= GPIO_NUM_5;

	uint8_t  pn532_packet_buff[128] = {0};
	uint32_t pn532_address	= (0x48 >> 1);

	static const uint8_t 	ack_frame[] 	= {0x00, 0x00,0xFF,0x00,0xFF,0x00};

	void __pn532_reset(uint32_t _pulse_ms)
	{
		gpio_set_level(pn532_rst, 0);
		vTaskDelay(_pulse_ms);
		gpio_set_level(pn532_rst, 1);
		printf("Reset Complete\n");
	}

	BaseType_t __wait_for_p70_irq(uint32_t wait_ms)
	{
		return xSemaphoreTakeFromISR(pn532_p70_intr_semphr, wait_ms);
	}

	void IRAM_ATTR __pn532_irq_func(void *arg)
	{
		if (gpio_get_level(pn532_p70_irq) == 0)
		{
			xSemaphoreGiveFromISR(pn532_p70_intr_semphr, NULL);
		}
	}

	uint32_t __pn532_i2c_send_cmd(uint8_t * _cmd, uint32_t _cmd_len)
	{
		uint32_t ret = 0;
		uint32_t i = 0, j = 0, sum = 0;

		pn532_packet_buff[j++] = (pn532_address << 1) | I2C_MASTER_WRITE;
		
		pn532_packet_buff[j++] = 0;
		pn532_packet_buff[j++] = 0;
		pn532_packet_buff[j++] = 0xFF;
		
		pn532_packet_buff[j++] = _cmd_len + 1;
		pn532_packet_buff[j++] = (~(_cmd_len + 1) + 1);
		
		pn532_packet_buff[j++] = 0xD4;
		sum = 0xD4;
			
		for (i = 0; i < _cmd_len; i++)
		{
			sum += _cmd[i];
			pn532_packet_buff[j++] = _cmd[i];
		}
		
		pn532_packet_buff[j++] = (uint8_t)(~sum + 1);
		pn532_packet_buff[j++] = 0x00;
		
		i2c_cmd_handle_t cmd = i2c_cmd_link_create();

		i2c_master_start(cmd);
		i2c_master_write(cmd, pn532_packet_buff, j, 1);
		i2c_master_stop(cmd);

		esp_err_t esp_err = i2c_master_cmd_begin(i2c_num, cmd, 100);

		if (esp_err == ESP_OK)
		{
			printf("i2c master write success\n");
			ret = 1;
		}
		else
		{
			printf("i2c_send_error: %d\n", esp_err);
		}

		i2c_cmd_link_delete(cmd);
		return ret;
	}

	uint32_t __pn532_i2c_read(uint32_t rx_len, uint32_t wait_max_ms)
	{
		uint32_t ret = 0;

		if (rx_len == 0)
		{
			return 1;
		}

		i2c_cmd_handle_t cmd = i2c_cmd_link_create();

		i2c_master_start(cmd);
		i2c_master_write_byte(cmd, (pn532_address << 1 ) | I2C_MASTER_READ, 1);
		
		if (rx_len > 1)
		{
			i2c_master_read(cmd, pn532_packet_buff, rx_len - 1, 0x00);
		}

		i2c_master_read_byte(cmd, pn532_packet_buff + rx_len - 1, 0x01);
		i2c_master_stop(cmd);

		esp_err_t esp_err = i2c_master_cmd_begin(i2c_num, cmd, wait_max_ms);

		if (esp_err == ESP_OK)
		{
			printf("i2c master read success\n");
			if (pn532_packet_buff[0] == 0x01)
			{
				ret = 1;
			}
		}
		else
		{
			printf("i2c master read error: %d\n", esp_err);
		}

		i2c_cmd_link_delete(cmd);
		return ret;
	}

	void __pn532_irq_handshake_init(void)
	{
        gpio_config_t io_conf;

        io_conf.intr_type 		= GPIO_INTR_NEGEDGE;
        io_conf.mode 			= GPIO_MODE_INPUT;
        io_conf.pin_bit_mask 	= (1 << pn532_p70_irq);
        io_conf.pull_up_en 		= GPIO_PULLUP_ENABLE;
        io_conf.pull_down_en 	= GPIO_PULLDOWN_DISABLE;

        gpio_config(&io_conf);
        gpio_set_intr_type(pn532_p70_irq, GPIO_INTR_NEGEDGE);
        gpio_install_isr_service(0);
        gpio_isr_handler_add(pn532_p70_irq, __pn532_irq_func, NULL);
	}

	void nfc_app(void *pv)
	{
        vTaskDelay(100);

		gpio_config_t pn532_reset_pin = {
											BIT(pn532_rst),
											GPIO_MODE_OUTPUT,
											GPIO_PULLUP_DISABLE,
											GPIO_PULLDOWN_DISABLE,
											GPIO_INTR_DISABLE
										};

		if (gpio_config(&pn532_reset_pin) != ESP_OK)
		{
			printf("failed to configure pn532_reset_pin\n");
		}
		printf("pn532_reset_pin configured\n");
		
		i2c_config_t i2c_master_config;
		
		i2c_master_config.mode = I2C_MODE_MASTER;
		i2c_master_config.sda_io_num = i2c_sda;
		i2c_master_config.sda_pullup_en = GPIO_PULLUP_DISABLE;
		i2c_master_config.scl_io_num = i2c_scl;
		i2c_master_config.scl_pullup_en = GPIO_PULLUP_DISABLE;
		i2c_master_config.master.clk_speed = i2c_clock;
			
		if (i2c_param_config(i2c_num, &i2c_master_config) == ESP_OK)
		{
			printf("i2c%d_param_config success, sda: %d, scl: %d\n", i2c_num,
                            i2c_master_config.sda_io_num, i2c_master_config.scl_io_num);
		
			if (i2c_driver_install(i2c_num, i2c_master_config.mode, 0, 0, 0) == ESP_OK)
			{
				printf("i2c%d_driver_install success\n", i2c_num);
			}
			else
			{
				printf("i2c%d_driver_install failed !!!!!!!\n", i2c_num);
			}
		}
		else
		{
			printf("i2c%d_param_config failed !!!!!!\n", i2c_num);
		}

		__pn532_irq_handshake_init();
		
		__pn532_reset(100 / portTICK_PERIOD_MS);
		vTaskDelay(2000 / portTICK_PERIOD_MS);
		
		/* get firmware version */
		uint8_t cmd[32] = {0};
		
		cmd[0] = 0x02;  // commands to fetch firmware version from the PN532 NFC module
		
		
		while(1)
		{
            __pn532_i2c_send_cmd(cmd, 1);
            __wait_for_p70_irq(200 / portTICK_PERIOD_MS);
            if (__pn532_i2c_read(sizeof(ack_frame), 1000 / portTICK_PERIOD_MS) == 1)
            {
                if (memcmp(ack_frame, pn532_packet_buff, sizeof(ack_frame)) == 0)
                {
                    __wait_for_p70_irq(200 / portTICK_PERIOD_MS);

                    if (__pn532_i2c_read(14, 1000 / portTICK_PERIOD_MS) == 1)
                    {
                        printf("**********************************\n");
                        printf("PN532: IC version: 0x%02X\n", pn532_packet_buff[8]);
                        
                        printf("PN532: Firmware version: %d.%d\n", pn532_packet_buff[9], pn532_packet_buff[10]);
                        
                        if (pn532_packet_buff[11] & (1 << 0))
                        {
                            printf("PN532: Supports: ISO/IEC 14443 TypeA\n");
                        }
                        
                        if (pn532_packet_buff[11] & (1 << 1))
                        {
                            printf("PN532: Supports: ISO/IEC 14443 TypeB\n");
                        }
                        
                        if (pn532_packet_buff[11] & (1 << 2))
                        {
                            printf("PN532: Supports: ISO18092\n");
                        }
                    }
                }
                else
                {
                    printf("ACK failed\n");
                }
            }

			vTaskDelay(5000 / portTICK_PERIOD_MS);
		}
	}

	void app_main()
	{
		nvs_flash_init();
        
		pn532_p70_intr_semphr = xSemaphoreCreateBinary();
        
		xTaskCreate(nfc_app, "NFC", 1024 * 2, NULL, 2, NULL);
        xTaskCreate(__blinky, "Blinky", 1024 * 2, NULL, 1, NULL);
	}
----------------------------------------------------------------------------------------------------------

RESULT:
-----------------------------------------------------------------------------------------------------------
SLA-W:
[image]

SLA-R:
[image]

ESP32 debug output:
[esp32_outputs.txt](https://github.com/espressif/esp-idf/files/2012688/esp32_outputs.txt)

PN532-NFC module I'm using:
[image]

KrrishnaShah, 2018-05-17:
Sir,

The above address is msb 7 bit address. If we consider it for 8bit address It will be 0x48.
Sorry for confusion.

The problem is solved when I decreased the frequency to the 50KHz.

Thanks for your contribution.

closed 2018-05-17 by KrrishnaShah

stickbreaker, 2018-05-17:
@KrrishnaShah you are correct, I stand corrected.  I reviewed the NXP documentation where it stated the slave address was 0x48.  I assumed, because it was NXP(owner or Phillips, who invented the protocol), that they would conform to the the protocol standards of 7bit slave address.
  But, alas, They did not:
[image]
 
As this capture from the datasheet show.

I am sorry to have distracted you.

Chuck.