espressif/esp-idf #1971: task locked on call pppapi_free

0BackDoor1, 2018-05-17:
Hi everybody,

## Environment
- Development Kit: none
- Core (if using chip or module): ESP-WROOM32
- IDF version: 94ec3c8e53e7f48133f17ae6c6e905fa5a856fd2
- Development Env: Make
- Operating System: MacOS
- Power Supply: external 5V

## Problem Description
gsm Task stuck trying to free ppp_pcb

### Expected Behavior
release ppp_pcb to re initialize a brand new ppp connection

### Actual Behavior
gsm task stuck trying to free ppp_pcb using api pppapi_free

### Steps to repropduce
1. perform a ppp connection
2. wait\trigger event ppp connection lost
3. re-init ppp_pcb releasing the previous one

### Code to reproduce this issue
```cpp
static ppp_pcb *ppp = NULL;
static bool ppp_connect()
{
	params_t *config = params_get_struct();
	static struct netif ppp_netif;
	
	if(ppp != NULL)
	{
		ESP_LOGW(TAG, "before pppapi_free");
		pppapi_free(ppp);
		ESP_LOGW(TAG, "after pppapi_free");
	}
	
	// After first successful initialization create PPP control block
	ppp = pppapi_pppos_create(&ppp_netif, ppp_output_callback, ppp_status_cb, NULL);

	if (ppp == NULL){
		return false;
	}
	
	pppapi_set_default(ppp);
	pppapi_set_auth(ppp, PPPAUTHTYPE_PAP, config->gsm_user, config->gsm_pass);
	pppapi_connect(ppp, 0);
	
	return true;
}

```
On console appear the log "before pppapi_free" but never the following one "after pppapi_free".
Can someone extend the pppos_client example with a pppapi_free case please?
> esp-idf/examples/protocols/pppos_client/main/pppos_client_main.c

sarpdaltaban, 2018-05-21:
> @sarpdaltaban <https://github.com/sarpdaltaban> thanks for your reply,
> I gave to it 16k (previously it was 2048), but same result, it never exit
> from pppapi_free or pppapi_close, other tasks still working as usual for
> even hours.
> I checked also free heap size, it is 153384

check which phase ppp is stucked in by printf(ppp->phase)

sarpdaltaban, 2018-05-21:
I can't take response to the pings after PPPoS sets IP to Telit GL865, do
you have any experience on the pinging issue of LWIP PPPoS? I have
activated ICMP but no response..

2018-05-21 13:55 GMT+03:00 Sarp Daltaban <sarpdaltaban@gmail.com>:

liuzfesp, 2018-06-19:
Hi @0BackDoor1, I'm afraid you are right, currently we have no plan to provide support for PPP support in IDF.

Torarqui, 2019-10-17:
I have the same problem. Also, I'm using Wifi in my application, and after disconnecting PPP. Wifi stops working, it tries to connect, then it doesn't get an IP, disconnects and because of that my code returns to wifi scan, but this scan never ends.

Torarqui, 2019-10-24:
I made a workaround in my code by calling ppp functions directly, without the pppapi layer, from what I saw here: http://lists.openvehicles.com/pipermail/ovmsdev/2018-April/004622.html this may be not thread-safe however.

AxelLin, 2020-02-12:
pppapi_* is supposed to be called in non-tcpip_thead context.
Make sure the ppp_connect() is not called in lwip tcpip_thread context.

david-cermak, 2020-12-21:
@0BackDoor1 Any update on this issue? Was the `pppapi_free()` called from the lwip thread, as Axel noted? Could this issue be closed now?

This issue references IDF `v3.0`, have you checked any newer versions? Recent versions may provide better user experience in calling the ppp api from user context.

closed 2021-01-07 by Alvin1Zhang