espressif/esp-idf #1976: Cannot ping the modem..

sarpdaltaban, 2018-05-19:
Thanks to the examples, established the IP and the network mask right now; however, I cannot ping the modem even the LWIP_ICMP module activation is enabled over STM32CubeMX.
Does it need a function to be called to  enable pings?

liuzfesp, 2018-06-04:
@sarpdaltaban could you let me know the detailed steps to reproduce this issue?

sarpdaltaban, 2018-06-04:
You must enable outgoing and incoming checksum for ICMP as well as
activating module on STM32CubeMX

liuzfesp, 2019-03-22:
@sarpdaltaban sorry for late response,  we don't have STM32CubeMX, is it possible for you to reproduce this issue without STM32CubeMX? E.g. with PC etc.

Alvin1Zhang, 2019-05-17:
@sarpdaltaban Thanks for reporting the issue. Is it possible for you to reproduce this issue without STM32CubeMX? E.g. with PC etc.? Thanks.

Alvin1Zhang, 2019-10-29:
@sarpdaltaban Thanks for reporting. For third party issue or "how" questions, please help submit a post at our forum at https://esp32.com. Thanks.

closed 2019-10-29 by Alvin1Zhang