espressif/esp-idf #1977: DMA ADC Fails On Oonnection to AP when in STA (station) mode

Splediferous, 2018-05-19:
Development Kit: Platform IO
    Kit version (for WroverKit/PicoKit/DevKitC): v3.1-dev-661-gf586f5e6
    Core (if using chip or module): ESP32 Lolin
    IDF version v3.1-dev-661-gf586f5e6
    Development Env: VS Code + PlatformIO
    Operating System: Windows
    Power Supply: USB

Platform Espressif 32 (Stage)

Updating espressif32 @ 29b3a81
Updating tool-esptoolpy @ 1.20100.0
Updating framework-arduinoespressif32 @ 6826508
Updating tool-espotapy @ 1.1.0

## Problem Description
When acquiring ADC data through DMA transfers and the device is in station mode (WIFI_MODE_STA). As the device connects to the AP, the ADC DMA buffer ceases to show valid data and continues to read (a probably random) fixed value.

This does not happen in WIFI_MODE_AP or WIFI_MODE_APSTA

### Expected Behavior
The ADC should continue to read valid data.
### Actual Behavior

> C00, F06, 3E6, 7E8, 0, 2D, 604, 22F, DFF, A84, F84, FE0, 8D2, CF2, FE, 4E1, 18E, 0, 93A, 518, F94, D2C, 9F0, FF0, 21F, 5CA, 66, 0, 84C, 430, F38, C62, E7C, FF0, 6C3,
> B00, 0, 300, 34E, 0, B80, 748, FFC, EB0, C00, F00, 3AF, 7A6, 0, E, 670, 27E, E12, A8F, F80, FE0, 8DC, CB1, D5, 49E, 168, 0, 964, 580,
> 0, 9D, 5C0, 20C, DC0, 9FF, FC7, FCF, 900, D27, 160, 500, 11C, 0, 900, 4CE, F8D, D0F, DE2, FD4, 5E9, A45, 0, 218, 3EF, 3F, C38, 7FF, FFD, F1E, B03, EA7, 300, 71A, 0, 0, 700, 302, E8E, B48, F00, FFE, 7E6, C12, 33, 3FE, 22B, 0, A6E, 60F, FE0, DE0, CFF, F86, 4D8, 90A, 0, 104, 509, 14E, D22, 931, FA4, F92,
> 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160,
> 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160,
> 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160, 160,

### Steps to repropduce

1. Start WiFi.
2. Start ADC acquisition.
3. Wait for device to connect to AP.
### Code to reproduce this issue

```cpp
#include <Arduino.h>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_log.h"
#include "driver/i2s.h"
#include "soc/syscon_reg.h"
#include "driver/adc.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "WiFi.h" 
#include "dac_cosine.h"
#define TAG "adc_i2s"
 
// Physically connect pins 25 and 34
 
uint32_t    SAMPLE_RATE = 120E3;  
uint32_t    NUM_SAMPLES = 64;    
uint32_t    SIGNAL_FREQ = 10E3; 

#define WIFICLIENT_SSID ("Fermion")
#define WIFICLIENT_PWD  ("1126567943")

#define WIFISERVER_SSID ("DAQ")
#define WIFISERVER_PWD  ("1126567943")

IPAddress local_IP(192,168,4,22);
IPAddress gateway(192,168,4,22);
IPAddress subnet(255,255,255,0);   
 
static QueueHandle_t i2s_event_queue;
static EventGroupHandle_t wifi_event_group;

void WiFiEvent(WiFiEvent_t event)
{
    switch(event) {
        case SYSTEM_EVENT_STA_GOT_IP:{
          Serial.println();
          Serial.println("CONNECTED");
          Serial.println();
          break;
        }
        default:{
          break;
        }
    } // switch
} // WiFiEvent
 
void setup(){
    Serial.begin(921600);
    WiFi.onEvent(WiFiEvent);
    WiFi.mode(WIFI_MODE_STA);
    WiFi.begin(WIFICLIENT_SSID, WIFICLIENT_PWD);

    i2s_config_t i2s_config ;
    i2s_config.mode =  i2s_mode_t(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN);
    i2s_config.sample_rate = SAMPLE_RATE;                           // 120 KHz
    i2s_config.dma_buf_len = NUM_SAMPLES;                           // 512
    i2s_config.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT;         // Should be mono but doesn't seem to be
    i2s_config.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT;
    i2s_config.use_apll = true,
    i2s_config.fixed_mclk = SAMPLE_RATE;
    i2s_config.communication_format = I2S_COMM_FORMAT_I2S;
    i2s_config.intr_alloc_flags = ESP_INTR_FLAG_LEVEL1;
    i2s_config.dma_buf_count = 2;
    //install and start i2s driver
    i2s_driver_install(I2S_NUM_0, &i2s_config, 1, &i2s_event_queue);
    // Connect ADC to I2S
    i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_6);
   
    // Generate test signal
    int step = nearbyint(0.008192 * SIGNAL_FREQ);       // ~10kHz
    dac_cosine_enable(DAC_CHANNEL_1);                   // enable the generator
    dac_frequency_set(0, step);                         // frequency setting is common to both channels
    dac_scale_set(DAC_CHANNEL_1, 0);                    // No scaling (0~3.7v)
    dac_offset_set(DAC_CHANNEL_1, 0);                   // No Offest
    dac_invert_set(DAC_CHANNEL_1, 2);                   // Sinewave
    dac_output_enable(DAC_CHANNEL_1);                   // Start DAC
 
    i2s_adc_enable(I2S_NUM_0);                          // Start ADC
}
 
void loop(){
    uint16_t i2s_read_buff[NUM_SAMPLES];
        system_event_t evt;
        if (xQueueReceive(i2s_event_queue, &evt, portMAX_DELAY) == pdPASS) {
            if (evt.event_id==2) {
                i2s_read_bytes(I2S_NUM_0, (char*)i2s_read_buff,  NUM_SAMPLES * 2, portMAX_DELAY);
                for (int i=0;i<NUM_SAMPLES;i++){
                Serial.printf("%X, ", i2s_read_buff[i] & 0xFFF);
                }
                Serial.println();
            }
        }
}

phonec, 2018-05-20:
Strangely this issue was solved recently here?
[ADC1 does not work in Multiple-channel scanning by pattern table, when enable WiFi.](https://github.com/espressif/esp-idf/issues/1333)
try the last master,

Splediferous, 2018-05-20:
@phonec 
I don't think it is the same issue. The ADC works fine in the other WiFi modes so it is not failing because the WiFi is enabled. In WIFI_MODE_STA, however, it works fine until there is a successful connection to the AP. It's at that point that it fails.

koobest, 2018-06-01:
Hi, @Splediferous, Is this problem solved?  if not, can you tell me your IDF commit ID?  I want  to reproduce the issue  you described,

 thanks !!

Splediferous, 2018-06-01:
@koobest No. It is not solved.

Splediferous, 2018-06-06:
@koobest.
No. I specifically don't want to use DMA to generate a signal which is what all the examples show. The measurements will always be of an external signal and the dac sinewave generator in the example above is just to emulate the external signals to demonstrate the issue (hence requiring a connection between pins). 
I'm not sure what the commit ID of the IDF was when the issue was raised but for the current version that still exhibits the behaviour it is.

`89859f7f4c07ce3e5d487a4b689dbda78f5d5c3a (HEAD -> master, origin/master, origin/HEAD) Calculate an absolute path for a custom partitions table (#1452)`

negativekelvin, 2018-06-07:
the commit referenced above has not been merged to arduino

mishafarms, 2019-05-19:
Has a solution to this been found?

tony-ward, 2019-07-12:
anyone?

mishafarms, 2019-07-14:
Is anyone from Espressif looking at this? I think this is a real bug that should be addressed?

Michael

koobest, 2019-07-15:
Hi,
I think this is the same issue as https://github.com/espressif/esp-idf/issues/1333, we fixed it on ESP-IDF(update rtc library on the master), but I'm not sure if this fix has been applied to the ardinno project. Another solution to this issue is to `initialize the ADC after WIFI initialization and connection`.

thanks!!

Alvin1Zhang, 2019-07-15:
@Splediferous @mishafarms Thanks for reporting and tracking the issue, checked that the issue is same as [issue 3714](https://github.com/espressif/esp-idf/issues/3714), will close this issue for now, and let us track in [issue 3714](https://github.com/espressif/esp-idf/issues/3714). Thanks.

closed 2019-07-15 by Alvin1Zhang