espressif/esp-idf #1980: truncate function not defined in unistd.h library

davctv, 2018-05-21:
Hi,

I would like to use "truncate" function of unistd.h to reduce the length of a file. However it seems not available/defined. The following link is a thread on esp32.com

https://esp32.com/viewtopic.php?f=13&t=5760&p=25136#p25136

Is it possible to introduce it in newlib library?

Thank you

davctv, 2018-06-11:
Hello @FayeY. Is there any update on this issue?

Thanks

renzbagaporo, 2018-06-13:
Hi @davctv, what filesystem are you working with? We have something (ongoing) with truncating files on fatfs, but spiffs might follow a bit later.

davctv, 2018-06-13:
Hi @renzbagaporo, I'm using FATfs component. Do you have an estimation time for implementing the truncate function?

Thanks

renzbagaporo, 2018-06-14:
It's an open MR in our internal repo and just passed initial review and test. Hopefully it'll get merged by the end of the week @davctv.

davctv, 2018-06-14:
Good. Thank you. I look forward for merging and to test it.

davctv, 2018-06-15:
The function is now included (commit c77f84e788819a5070a9ea63b9a73fb0aea6df15) and it seems to work correctly. Thanks.

closed 2018-06-15 by davctv

dizcza, 2021-12-09:
@davctv thanks for bringing in the `truncate` function. Wouldn't it be nice to have `ftruncate` as well? :) Implementing it by myself is beyond my capabilities.

Rajssss, 2022-01-23:
@dizcza I am looking for the same, did you find anything useful regarding this anywhere?

dizcza, 2022-01-24:
@Rajssss nope.

igrr, 2022-01-24:
@Rajssss @dizcza I have opened a feature request: https://github.com/espressif/esp-idf/issues/8279