espressif/esp-idf #1982: esp32的sta+ap模式下手机连接AP无法上网

gaye746560359, 2018-05-21:
代码如下：
`#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

#define WIFI_SSID "KKKKg"
#define WIFI_PASS "123456zzz"

static const char *TAG = "espressif";

/* FreeRTOS event group to signal when we are connected & ready to make a request */
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
const int CONNECTED_BIT = BIT0;

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch (event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        /* This is a workaround as ESP32 WiFi libs don't currently
           auto-reassociate. */
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

static void initialise_wifi(void)
{
    tcpip_adapter_init();
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    wifi_config_t sta_wifi_config = {
        .sta = {
            .ssid = WIFI_SSID,
            .password = WIFI_PASS,
        },
    };
    wifi_config_t ap_wifi_config = {
    .ap = {
        .ssid = "esp32_ap",
        .password = "xj123456",
        .authmode = WIFI_AUTH_WPA_WPA2_PSK,
        .max_connection = 3,
    },
    };
    ESP_LOGI(TAG, "Setting WiFi configuration SSID %s...", sta_wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_APSTA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &sta_wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_AP, &ap_wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_start() );
}

void app_main()
{
    ESP_ERROR_CHECK(nvs_flash_init());
    initialise_wifi();
}`

串口打印输出：
[image]

手机连接AP的信息
[image]

最终无法打开网页。请问是什么原因。？

yake827, 2018-05-22:
看你的意思是要 ESP32 STA 连上路由器，然后手机连上ESP32 AP ，希望这样手机也可以上网是么？如果是这样的话是不可以的，目前 ESP32 不支持这种桥的方式

gaye746560359, 2018-05-23:
@yake827 那这种sta+ap存在的价值是什么。如果只是AP模式，手机接入后有什么意义。

FayeY, 2018-05-28:
@gaye746560359 你好，是的，ESP32 不支持桥接。
ESP32 AP 提供的是一个局域网的功能，例如，手机可以作为 sta 连接到 ESP32 AP 进行本地控制。
感谢你对 ESP32 的关注！

closed 2018-05-28 by FayeY

1410898565, 2021-10-16:
你开启了esp32的路由转换了吗，看你代码是没开启ip_napt_enable

szkprx2018, 2021-10-19:
调用ip_napt_enable这个函数，入口参数写哪个IP地址