espressif/esp-idf #1986: Unstable WiFi after multiple HTTP call

vtunr, 2018-05-22:
Hi, 

Here's the issue that i have : 

## Environment

- Development Kit:      [ESP32-Dev-Kit|ESP32 Thing]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version: 
    Started on 6acb38af, checkout latest commit (92c469b5) and same behavior
- Development Env:		[Make|Eclipse]
- Operating System:		[Windows]
- Power Supply:         [USB]
- Router: Apple Timecapsule

## Problem Description

When using HTTP GET, after 120s with one GET per second, WiFi is lost and is not recoverable.

### Expected Behavior
Staying connected and reconnect without any issue

### Actual Behavior
Stuck disconnected without anything running anymore.

### Steps to repropduce
Using https://github.com/tuanpmt/esp-request to create GET request and set up a Task that will GET every second. (Same issue with or without SSL)

### Code to reproduce this issue

Here's a few thing that i set up : 

```cpp
/*
 * ESPRESSIF MIT License
 *
 * Copyright (c) 2018 <ESPRESSIF SYSTEMS (SHANGHAI) PTE LTD>
 *
 * Permission is hereby granted for use on ESPRESSIF SYSTEMS products only, in which case,
 * it is free of charge, to any person obtaining a copy of this software and associated
 * documentation files (the "Software"), to deal in the Software without restriction, including
 * without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense,
 * and/or sell copies of the Software, and to permit persons to whom the Software is furnished
 * to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all copies or
 * substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 */

#include <stdio.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#include "esp_log.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_system.h"
#include "nvs_flash.h"

#include "abl_http_req.h"

/* FreeRTOS event group to signal when we are connected & ready to make a request */
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
const int CONNECTED_BIT = BIT0;

static const char *TAG = "APP";

static void request_task(void *pvParameters)
{
	while(1){
		xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);
		abl_http_response_t get_response;
		const char *header_ua[1] = {"User-agent : myUA"};
		ESP_LOGI(TAG, "------- Starting get --------");
		abl_http_get("http://httpbin.org/get", header_ua, &get_response);
		ESP_LOGI(TAG, "-------  Finish get  --------");
		memset(&get_response, 0, sizeof(abl_http_response_t));
		ESP_LOGI(TAG, "WM : %u", uxTaskGetStackHighWaterMark(NULL));
		vTaskDelay(1000/portTICK_PERIOD_MS);
	}
}

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_LOST_IP:
    	ESP_LOGI(TAG, "Lost IP, disconnecting ...");
    	esp_wifi_disconnect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
    	break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        /* This is a workaround as ESP32 WiFi libs don't currently
           auto-reassociate. */
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

static void initialise_wifi(void)
{
    tcpip_adapter_init();
    wifi_event_group = xEventGroupCreate();
    xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "my_ssid",
            .password = "my_passphrase",
        },
    };
    ESP_LOGI(TAG, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_start() );

}

void app_main()
{
    nvs_flash_init();
    initialise_wifi();

    xTaskCreate(&request_task, "request_task", 8192, NULL, 5, NULL);
}

```

## Debug Logs

Here's the ouput i got : 

```
I (116372) APP: ------- Starting get --------
I (123402) wifi: bcn_timout,ap_probe_send_start
I (125902) wifi: ap_probe_send over, resett wifi status to disassoc
I (125902) wifi: state: run -> init (1)
I (125902) wifi: pm stop, total sleep time: 96762297 us / 124460513 us

I (125912) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
E (134372) HTTP_REQ: Error write header
E (134372) HTTP_REQ: Error send request
I (134372) ABL_HTTP: HTTP Code : 0
I (134372) ABL_HTTP: HTTP Response empty
I (134372) APP: -------  Finish get  --------
I (134382) APP: WM : 4504
I (135682) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (135682) wifi: state: init -> auth (b0)
I (135682) wifi: state: auth -> assoc (0)
I (135682) wifi: state: assoc -> run (10)
I (135702) wifi: connected with my_ssid, channel 1
I (135702) wifi: pm start, type: 1

I (136372) event: sta ip: 192.168.178.110, mask: 255.255.255.0, gw: 192.168.178.1
I (136372) APP: ------- Starting get --------
I (142452) wifi: bcn_timout,ap_probe_send_start
I (144952) wifi: ap_probe_send over, resett wifi status to disassoc
I (144952) wifi: state: run -> init (1)
I (144952) wifi: pm stop, total sleep time: 507229 us / 9243672 us

I (144952) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
W (145322) wifi: alloc eb len=76 type=2 fail, heap:584

W (145322) wifi: m f probe req l=0

W (145442) wifi: alloc eb len=76 type=2 fail, heap:584

W (145442) wifi: m f probe req l=0

W (145562) wifi: alloc eb len=76 type=2 fail, heap:584

W (145562) wifi: m f probe req l=0

W (145682) wifi: alloc eb len=76 type=2 fail, heap:584

W (145682) wifi: m f probe req l=0

W (147732) wifi: alloc eb len=76 type=2 fail, heap:584

W (147732) wifi: m f probe req l=0

o:1 0, ap:255 255, sta:1 0, prof:1
I (241112) event: station ip lost
I (241112) APP: Lost IP, disconnecting ...

```

I tried  : 
```cpp
    case SYSTEM_EVENT_STA_LOST_IP:
    	ESP_LOGI(TAG, "Lost IP, disconnecting ...");
    	esp_wifi_disconnect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        esp_wifi_connect();
    	break;
```
But it's not working either.

Let me know if you need more informations.

Thanks,

Tony

stickbreaker, 2018-05-23:
@vtunr 
>W (145322) wifi: alloc eb len=76 type=2 fail, heap:584

looks like you have a memory leak. 
check out

`void heap_caps_print_heap_info(uint32_t caps)`

Chuck.

vtunr, 2018-05-23:
Hi @stickbreaker,

Thanks for you quick response, I found the memory leak in the HTTP library and proposed a fix (https://github.com/tuanpmt/esp-request/pull/20)

I have no more WiFi issues :)

Tony

snahmad, 2018-09-05:
I am getting similar error
wifi: alloc eb len=192 type=2 fail

I am using WIFI with external RAM.

Internal Heap free: 41120
Heap free: 4195984:4233744

Plenty of external ram and some internal RAM.

Does in default settings WIFI only uses internal ram.