espressif/esp-idf #1987: "Allow VFS file descriptors in select()" breaks my project

ataweg, 2018-05-22:
update to https://github.com/espressif/esp-idf/commit/18e83bcd53f2aacd667959af755f8a443fb2a57e
breaks my project. The previous commit https://github.com/espressif/esp-idf/commit/2aa9c15906b259dd188e88c144cb422eaa091b27 works fine.
Using the latest version,  head revsion doesn't help.
I used a variant of libesphttpd, and with the update, sending the HTML response no longer works.
For me, it looks like an incomptiblity has been added that requires adaptations of the project source code, or there is still a bug in your changes. 
Let me know if you need more info.
Viele Gruesse
Axel

projectgus, 2018-05-23:
Hi Axel,

Thanks for reporting this. Is it possible for you to share any of your source code with us, or perhaps a simplified version which reproduces this problem?

Thanks,

Angus

ataweg, 2018-05-23:
please checkout https://github.com/ataweg/esp32-httpd-demo.git

for your network change in sdkconfig
CONFIG_WIFI_SSID="esp-wlan"
CONFIG_WIFI_PASSWORD="1234567890"

I think issue #1989 has the same or a similar cause.
Viele Gruesse
Axel

loboris, 2018-05-24:
I have a similar issue, I couldn't compile the project, got an error
```
<my_path>/esp-idf/components/lwip/include/lwip/lwip/sockets.h:594:19: error: static declaration of 'select' follows non-static declaration
 static inline int select(int maxfdp1,fd_set *readset,fd_set *writeset,fd_set *exceptset,struct timeval *timeout)
                   ^
```
After adding `CFLAGS += -DESP_PLATFORM` to my **component.mk** everything compiles fine.

It also works if `-DESP_PLATFORM` is added to [esp-idf/make//project.mk](https://github.com/espressif/esp-idf/blob/master/make/project.mk#L319)
```
CFLAGS := $(strip \
	-std=gnu99 \
	-DESP_PLATFORM \
	$(OPTIMIZATION_FLAGS) $(DEBUG_FLAGS) \
	$(COMMON_FLAGS) \
	$(COMMON_WARNING_FLAGS) -Wno-old-style-declaration \
	$(CFLAGS) \
	$(EXTRA_CFLAGS))
```

igrr, 2018-05-24:
@loboris `ESP_PLATFORM` is part of CPPFLAGS (i.e. preprocessor flags, so that it applies to both C and C++): https://github.com/espressif/esp-idf/blob/master/make/project.mk#L265. If your component makefile overrides CPPFLAGS, make sure you keep the flags provided by the build system.

loboris, 2018-05-24:
@igrr I'm aware that `ESP_PLATFORM` is part of CPPFLAGS and none of my componnents makefiles does not modify it, but I still have an error without adding `CFLAGS += -DESP_PLATFORM` to my component.mk.
The error started to appear after I've updated to the commit **1f7b41e**.

igrr, 2018-05-24:
Could you please run make V=1 and paste the gcc command line for the file which produces compilation error?

dobairoland, 2018-05-24:
Hi @ataweg.  Thank you for reporting the issue and providing your code. I found the bug and the correction will be committed very soon.

loboris, 2018-05-24:
@igrr 
```
xtensa-esp32-elf-gcc -I-I. -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/curl/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/libssh2/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/zlib -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/espmqtt/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/espmqtt/lib/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/littlefs -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/lib/mp-readline -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/lib/netutils -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/lib/timeutils -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32 -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/build -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/libs -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/bootloader_support/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/driver/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/driver/include/driver -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp_adc_cal/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/nghttp/port/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/soc/esp32/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/soc/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/ethernet/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/expat/include/expat -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/expat/port/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/json/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/json/port/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/log/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/newlib/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/nvs_flash/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/tcpip_adapter/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/lwip/include/lwip -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/lwip/include/lwip/port -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/lwip/include/lwip/posix -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/mbedtls/mbedtls/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/mbedtls/port/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spi_flash/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/wear_levelling/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/wear_levelling/private_include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/vfs/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/newlib/platform_include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/xtensa-debug-module/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/wpa_supplicant/include -I/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/wpa_supplicant/po [... cut ...]
In file included from /home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/lwip/include/lwip/lwip/sockets.h:41:0,
                 from /home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/modsocket.c:53:
/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/newlib/include/sys/select.h:25:5: note: previous declaration of 'select' was here
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *errorfds, struct timeval *timeout);
     ^

```

igrr, 2018-05-24:
I think this is because in your micropython component, the rules for the object files are generated using custom makefile: https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/f724c162a0f4a5be08079cc99bd9a28f6198f5dc/MicroPython_BUILD/components/micropython/py/mkrules.mk#L36

As you see, this rule passes `CFLAGS` to the compiler but does not pass `CPPFLAGS`. So ESP_PLATFORM and IDF_VER are both missing.

loboris, 2018-05-24:
@igrr Thank you very much, that solves the issue. It's embarrassing not to find it myself.

closed 2018-05-25 by igrr