espressif/esp-idf #1989: WiFiClient TCP/IP packet is no longer sent

Electrozest, 2018-05-23:
I have been using esp-idf checked out from 12th April 2018 with last commit c876ab390b95818eae52e7476eb328c224454724

With this esp-idf, WiFi works correctly. I can connect to router, obtain an IP address and then send a packet to a remote server using. I have been running with successful packets sent once per minute over several days with that version of esp-idf using following code:
```
WiFiClient client;
if(client.connect(server_host,server_port)) {
 client.setNoDelay(true);
 client.setTimeout(WIFI_SOCKET_TIMEOUT);
 client.write(packet);
}
```
Updating to esp-idf with updates up to Mon May 21 20:00:49 2018 +0800, then the above code completes without errors and without any delays waiting for the tcpip connection, however the remote PC server receives no packets.

Reverting to esp-idf from 12th April 2018, the code behaves correctly and the remote PC server receives packets from the ESP32. 

Reverting back to Mon May 21 20:00:49 2018 +0800 code again and the packet is not received by the PC server, but the code completes silently with no indications of any error or delay in making a tcpip connection.

me-no-dev, 2018-05-23:
Sure! But is it IDF or Arduino issue? I am a bit confused by what I read above?

@Electrozest are you using Arduino as IDF component? Are you updating IDF or Arduino?

Electrozest, 2018-05-23:
@me-no-dev, @igrr  I understand that WiFiClient.cpp is a library from arduino-esp32, however I am only updating the esp-idf component and seeing the difference in behaviour, so I suspect that the underlying WiFi or tcpip implementation from esp-idf is not sending the packet.

me-no-dev, 2018-05-23:
@Electrozest I just ran a test using the following simple code and all appears correct (using the latest IDF)

```cpp
void testClient(){
    static size_t i = 0;
    static uint8_t buf[1024];
    WiFiClient client;
    if (!client.connect(IPAddress(192,168,0,109), 8080)) {
        Serial.println("WiFiClient connect failed");
        return;
    }
    client.print("GET / HTTP/1.1\r\nHost: 192.168.0.109\r\nConnection: close\r\n\r\n");
    while(client.connected() && !client.available());
    size_t a = client.available();
    while(a){
        if(a > 1024){
            a = 1024;
        }
        client.readBytes(buf, a);
        Serial.write(buf, a);
        a = client.available();
    }
}
```

Electrozest, 2018-05-23:
I checked with WireShark. With latest esp-idf, the tcpip packet is being sent to the correct port on the server, however it has a zero length payload.
I tried sending more packets with simple ASCII text and they are all sent as zero length and with no payload.

me-no-dev, 2018-05-23:
I am looking at the same thing. I also added noDelay and setTimeout. All is good! can you post a minimal full sketch that displays the issue?

Electrozest, 2018-05-23:
Your simple code example is reading on the ESP32 client and then outputting to serial.

My WiFiClient code only sends data. It does not use the client.readBytes() function, only the client.write(bytes,len) function.

me-no-dev, 2018-05-23:
@Electrozest Another thing that you can do to diagnose this issue is tho change [this line](https://github.com/espressif/esp-idf/blob/master/components/lwip/include/lwip/port/lwipopts.h#L676) to `LWIP_DBG_ON` and rebuild.
Still, please post here a minimal sketch that reproduces the issue for you.

Electrozest, 2018-05-23:
Here is my minimal sketch that is appearing to send, but is not being received at tcpip server, and is sending empty packets.
```
void testClient(){
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(WIFI_SSID1, WIFI_PASS1);
    int status;
    while((status = WiFi.status())!=WL_CONNECTED){
      Serial.println("WiFi wait for connect");
      vTaskDelay(1000);
    }
    Serial.println("WiFi connected");
    WiFiClient client;
    if (!client.connect(IPAddress(192,168,1,54), SERVER_PORT)) {
        Serial.println("WiFiClient connect failed");
        vTaskDelay(1000);
        return;
    }
    Serial.println("WiFiClient connected");
    vTaskDelay(1000);
    Serial.println("WiFiClient send packet");
    client.write("test pkt\r\n",10);
    vTaskDelay(1000);
    Serial.println("WiFiClient stop");
    client.stop();
    vTaskDelay(1000);
}
```

me-no-dev, 2018-05-23:
this code works for me, but this is not a full sketch as I asked, but rather a function.
Please post full sketch!

```
Socket created
Socket setup complete
Connection address: ('192.168.0.148', 63739)
Message[192.168.0.148:63739] - test pkt
Disconnected
Connection address: ('192.168.0.148', 54832)
Message[192.168.0.148:54832] - test pkt
Disconnected
Connection address: ('192.168.0.148', 64867)
Message[192.168.0.148:64867] - test pkt
Disconnected
Connection address: ('192.168.0.148', 58049)
Message[192.168.0.148:58049] - test pkt
Disconnected
```

Electrozest, 2018-05-23:
Here is the output with debug enabled: 
```
load:0x40078000,len:11584
entry 0x40078a60
WiFi wait for connect
WiFi wait for connect
WiFi wait for connect
WiFi wait for connect
WiFi wait for connect
WiFi connected
tcp_enqueue_flags: queueing 6509:6510 (0x2)
tcp_output_segment: 6509:6509
tcp_output: sending ACK for -395215568
WiFiClient connected
tcp_output: nothing to send (0x0)
WiFiClient send packet
tcp_output: nothing to send (0x0)WiFiClient stop
tcp_enqueue_flags: queueing 6510:6511 (0x1)
tcp_output_segment: 6510:6510
tcp_output: nothing to send (0x0)
tcp_output: sending ACK for -395215567
```

me-no-dev, 2018-05-23:
my tcp log looks like this:
```
tcp_enqueue_flags: queueing 6509:6510 (0x2)
tcp_output_segment: 6509:6509
tcp_output: sending ACK for 986485620
WiFiClient connected
tcp_output: nothing to send (0x0)
WiFiClient send packet
tcp_write(pcb=0x3ffda96c, data=0x3f4056d0, len=10, apiflags=5)
tcp_write: queueing 6510:6520
tcp_output_segment: 6510:6520
tcp_output: nothing to send (0x0)
tcp_output: nothing to send (0x0)
WiFiClient stop
tcp_enqueue_flags: queueing 6520:6521 (0x1)
tcp_output_segment: 6520:6520
tcp_output: nothing to send (0x0)
```

note `tcp_write(pcb=0x3ffda96c, data=0x3f4056d0, len=10, apiflags=5)`

Electrozest, 2018-05-23:
Following is sketch without the testclient() function which is as above.
```
#include <Arduino.h>
#include <WiFi.h>

void#define WIFI_SSID1 "_____"
#define WIFI_PASS1 "______"

setup(){
}

void loop() {
  while(1){
    testClient();
  }
}
```

me-no-dev, 2018-05-23:
I can not reproduce this issue on my end.

As a side note, you do not need `while(1)` in the loop. the loop already does that. Do not call `Serial.begin` and `WiFi.begin` each loop either.

Electrozest, 2018-05-23:
Agreed. I originally had other code after the while loop, so removed to make test case, but I know loop() is called from an external while loop anyway.

The tcpdump on the server is not showing the packet which is meant to have been sent, but is showing packets with length set to 0 and following flags:

Flags [.], ack 1, win 5744, length 0
Flags [F.], seq 1, ack 1, win 5744, length 0

All packets from the ESP32 IP address are zero payload length.

me-no-dev, 2018-05-23:
those packets are part of the connect procedure and do not carry data, but in your log it's visible that the tcp layer do not see data to send `tcp_output: nothing to send (0x0)`.
Can you start a clean setup of IDF and Arduino and give it another go? We can not come up with a reason for this behaviour, nor reproduce it.
You can also grab the sdkconfig from Arduino's folder and try that one as well

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig

Splediferous, 2018-05-23:
Do you see the same problem if the WiFi mode is WIFI_MODE_APSTA instead of WIFI_MODE_STA?

Alvin1Zhang, 2019-03-15:
@Electrozest Hi, sorry that get back to you so late. Does your issue still exist? Thanks.

jonaphael, 2020-10-28:
Does this issue still exist?