espressif/esp-idf #1993: ESP32 Wifi doesn't autoreconnect if i turn wifi router off over night

mhcko, 2018-05-24:
## Environment

- Development Kit:  Lolin32

- Core: ESP-WROOM32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		Arduino IDE
- Operating System:		Windows
- Power Supply:         external 5V

## Problem Description

ESP32 Wifi doesn't autoreconnect if i turn wifi router off over night. Connect after power off/on

### Code to reproduce this issue

<script src="https://gist.github.com/mhcko/f46d5c0e4aab02c9d10572792d9755a1.js"></script>

XinDeng11, 2018-06-24:
Hi@mhcko, have you invoke any scan operation after router power off?

FayeY, 2018-07-12:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.

closed 2018-07-12 by FayeY