espressif/esp-idf #1994: I2C driver: The I2C_MASTER_TOUT_CNUM_DEFAULT is to short for some slave devices

amuzyka-grinn, 2018-05-24:
- Development Kit:   none
- Kit version none
- Core ESP32
- IDF version 358c822
- Development Env:		Eclipse
- Operating System:		Ubuntu
- Power Supply:         USB or Battery

## Problem Description
On the custom board, I am using the BQ27441 which i2c response time is longer than the default timeout set for master configuration (8 cycles). That results in the communication error and bus resets.
I would prefer to expose timeout parameter together with other i2c configs to handle such custom cases.

## Other items if possible
In [i2c_timeout_conf](https://github.com/amuzyka-grinn/esp-idf/commit/b8b239c35f1bc16f7d56fc1a8cbc09d32dabec34) I added a possible solution which configures the timeout for both slave and master type of configuration.

koobest, 2018-06-01:
Hi, @amuzyka-grinn 
Thank you for your suggestion. We will update the I2C driver soon.

closed 2018-06-20 by FayeY

surekhaibot, 2019-12-14:
Hi,
I am trying to communicating BQ27441 with esp32 in espidf.  i used max value of timeout reg (0xFFFF) then i am getting garbage value. i am not able to read the device type reg of BQ27441 as 0x0421.

Thanks&regards
rekha