espressif/esp-idf #1996: Crash when using bluetooth RFCOMM with high CPU load.

ErwinTheRabbit, 2018-05-24:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP32]
- IDF version: cc3a5761
- Development Env:		[Eclipse]
- Operating System:		[Windows]
- Power Supply:         [Battery]

## Problem Description

At high load when using bluetooth sometimes esp32 crashes.

### Steps to repropduce

1. Make infinite cycle with ariphmetic operations and timer isr with same operations.
2. Connect to esp32 using RFCOMM and send/receive any data.
3. Sometimes esp32 will crash.

## Debug Logs

```
ASSERT_ERR(co_list_is_empty(&ld_fm_env.fifo) \|\| (timestamp != ((struct ld_fm_frm _elt*)ld_fm_env.fifo.last)->timestamp)), in ld_fm.c at line 316
Guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0)
--
```

or:

```
ASSERT_PARAM(0 1), in ld_pscan.c at line 235
--
```

blueMoodBHD, 2018-06-28:
Hi @ErwinTheRabbit pathspec 'cc3a5761' did not match any file(s) known to git. So, could you please check the IDF version again, Thanks.

igrr, 2018-06-28:
@ErwinTheRabbit After WDT the time returned by gettimeofday is undefined (and can be, for instance, equal to the value of previous settimeofday). Given that it is not possible to preserve accurate timekeeping in case of WDT, what is the desired behaviour? (other than not having the WDT in the first place, of course...)

ErwinTheRabbit, 2018-06-28:
Desired behavior - no crash. Resetting registers is just additional information, maybe it help to find something about this crash. If i generate WDT reboot by myself (for example, infinite cycle in timer interrupt), these registers are not reset.

TianHao-Espressif, 2018-07-09:
@ErwinTheRabbit , as @blueMoodBHD said, please provide a commit based on ESP-IDF. Maybe your esp-idf is one which be patched. Or you can provide the commit in esp-idf/component/bt/lib/

blueMoodBHD, 2018-07-09:
Hi @ErwinTheRabbit  Could you please provide some more information? Thanks.

1. About infinite cycle, is there some delay in the loop? Just `while(true){...}` may cause some problems.
2. About the IDF version, is it newest master from git? or others?
3. About the date, when will you call `esp_spp_write()` to sent data? judging congest or not?

ErwinTheRabbit, 2018-07-09:
About my esp-idf, i have newest version with patched i2c to receive variable lenght data.
About infinite cycle, yes, just while(1){...}, but more impotant is a long timer interrupt.
About esp_spp_write. Ofc, i check overflow.

Bug reproduced once every ~6 hours nonstop working.

For now this bug not reproducing after i replaced long timer interrupt and long spi_flash_erase_range(...,0x200000) with a short interrupt and a lot of short spi_flash_erase_sector called at different times.

blueMoodBHD, 2018-07-09:
Hi，I'm glad to hear that you have solved this issue. 
Bluetooth control is sensitive to the time of the interrupt response. Turning off interrupts or interrupt handlers for too long may cause some errors. So, it is necessary to pay attention to the operation of long interruption time.

closed 2018-07-12 by FayeY