espressif/esp-idf #1998: SPP/rfcomm crashes when initiator/master is resets

jakkra, 2018-05-24:
INSTRUCTIONS
============
- [x] I have read the documentation [ESP-IDF Programming Guide](http://esp-idf.readthedocs.io/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      custom esp32 board
- Core (if using chip or module): [ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    358c822d6a29c8f881fd2fbbb100ec9fa9320596
- Development Env:		[Make]
- Operating System:		[Windows]
- Power Supply:         [USB]

## Problem Description
Have one ESP32 running as a SPP acceptor/slave and then connect to it from another device. Reseting the other device i.e the initiator/master causes the ESP32 to crash. This is not a problem in Bluedroid, as I have the exact same issure running Bluekitchen stack. No data needs to be exchanged between the devices, just beeing connected and the pulling the power/hard reset on the initiator/master causes the EPS32 acceptor/slave to crash.

This doesn't seem to happen the other way around for us, i.e only the acceptor crashes when the initiator/master is reset, and not the other way around.

Sadly I have not managed to get this crash with another esp32 running bt_spp_initiator example and then resetting it.
However I have gotten it with our own application as initiator running on an esp32 and also two other Bluetooth Classic devices with SPP capability at our office. 

The crash happends at least 50% of the time when the master/initiator is reset/killed.

### Expected Behavior
ESP32 doesn't crash after the master resets,
### Actual Behavior
ESP32 crashes when SPP master is reset.

### Steps to repropduce
1. Connect to the esp32 running bt_spp_acceptor
2. On the initiator/master do a hard reset/pull the power
3. The esp32 acceptor will now crash

### Code to reproduce this issue
Use: 
esp-idf/examples/bluetooth/bt_spp_acceptor

## Debug Logs and more
Debug logs, coredump, elf file and more found here:
[_SPP_CRASH_.zip](https://github.com/espressif/esp-idf/files/2035218/_SPP_CRASH_.zip)

Please come back to me if anything is unclear.

Thank you

blueMoodBHD, 2018-06-28:
Hi @jakkra could you please tell me what is the device you used, Thanks.

jakkra, 2018-07-13:
@blueMoodBHD Hi, sorry for my slow response, I have been on holiday.

I can reproduce the problem with an esp32 running bt_spp_acceptor example (the device that crashes) and the initiator running our application on an esp32. I have modified our application and stripped it of most functinality. Can I send the binary and instructions to you using email? In that case may I have your email. With this you will be able to reproduce it.

blueMoodBHD, 2018-07-16:
Hi @jakkra , 

Thanks for your help, my email is the Github account.

blueMoodBHD, 2018-07-16:
Hi @jakkra 
I have received your email, Thanks.

blueMoodBHD, 2018-07-17:
Hi, @jakkra 
Please replace `esp-idf/components/bt/lib/libbtdm.a` with the file below, and test again. Thanks.
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/2201723/libbtdm_app.zip)

jakkra, 2018-07-18:
@blueMoodBHD Awesome, that seems to have fixed the problem! You will patch release 3.1 with this?

blueMoodBHD, 2018-07-18:
@jakkra Yes, I will. But it may takes some time.

jakkra, 2018-08-08:
@blueMoodBHD Thanks for working on this, how is the progress going for getting this merged?

blueMoodBHD, 2018-08-08:
@jakkra This bugfix have been merged into master and release v3.1, and it will synchronize to github in a few days.

closed 2018-08-09 by FayeY