espressif/esp-idf #1999: deinit tcpip_adapter_init and esp_event_loop_init to free resources

my-abousamra, 2018-05-24:
## Environment

- Development Kit:    [none]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version 358c822d
- Development Env:		[Eclipse]
- Operating System:		[Ubuntu]
- Power Supply:         [Battery]

## Problem Description

I'm trying to deinit tcp adapter to free some memory. I can't find deinit function for tcpip_adapter_init to delete LwIP core task and de-initialize LwIP-related work as they are no longer needed. Also, I can't find deinit function for  esp_event_loop_init to delete system Event task and de-initialize an application event's callback function as they are no longer needed.

These functions allocate resources and my application doesn't want WiFi all the time so I have to deinit all its related resources and be able to re-init it again later when needed

### Expected Behavior
deinit tcpip_adapter_init and esp_event_loop_init
### Actual Behavior
no functions for that

liuzfesp, 2018-06-04:
@my-abousamra yes, IDF currently doesn't support tcpip adapter and event deinit, we will consider it in the future.

my-abousamra, 2018-06-04:
I hope it would be soon

my-abousamra, 2018-06-04:
I hope this issue to be labelled with "to be implemented"

Michael2196, 2019-06-11:
Hello,
I'm having the same problem with the tcip_adapter_init() function. Did you find any solution ?
Thank you

FoxKeys, 2019-09-17:
2019 year. Still nothing?
Such "deinit" functions is highly desirable

david-cermak, 2021-09-22:
Hi @my-abousamra 

We already have functions to deinitilalize both `esp_event` and `esp_netif` (successor of the tcpip_adapter), so we will close the issue.

As mentioned in https://github.com/espressif/esp-idf/issues/4628 we are still not able to uninitialize the TCP/IP stack itself. In case you need to free resources or reduce power consumption, please check the methods described in the referenced issue, or possibly suspend and delete the lwip thread. We'd like to address this issue correctly in the longrun, but would need to implement these uninits in lwip first and that won't happen in any near future.

closed 2021-09-23 by Alvin1Zhang