espressif/esp-idf #2000: crash issue

malaimoo, 2018-05-25:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM32|ESP32]
- IDF version 92c469b5
- Development Env:		[Make]
- Operating System:		[MacOS]
- Power Supply:         [USB]

## Problem Description

use esp_wifi_80211_tx send packet frequently，send ble noti, and sniffer 80211

### Expected Behavior

### Actual Behavior

crash frequently

### Steps to repropduce

### Code to reproduce this issue

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.) 
```
SSERT_PARAM(61455 1073470136), in ke_mem.c at line 409
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400847f4  PS      : 0x00060034  A0      : 0x8001a04c  A1      : 0x3ffeb660
0x400847f4: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0xdad4f00f  A5      : 0x60008054
A6      : 0x3ffc1028  A7      : 0x60008050  A8      : 0x800847f1  A9      : 0x3ffeb640
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x40084729  LEND    : 0x40084730  LCOUNT  : 0x00000000
0x40084729: r_assert_param at ??:?

0x40084730: r_assert_param at ??:?

Backtrace: 0x400847f4:0x3ffeb660 0x4001a049:0x3ffeb680 0x400870b6:0x3ffeb6b0 0x4014ba29:0x3ffeb6f0 0x4004240b:0x3ffeb710 0x4001a637:0x3ffeb730 0x40019d11:0x3ffeb760 0x40055b4d:0x3ffeb780 0x40146ca7:0x3ffeb7a0 0x40146db3:0x3ffeb7c0
0x400847f4: r_assert_param at ??:?

0x400870b6: r_lld_pdu_tx_prog at ??:?

0x4014ba29: r_lld_pdu_data_send at ??:?

0x40146ca7: r_rw_schedule at ??:?

0x40146db3: btdm_controller_task at ??:?

Core 0 register dump:
PC      : 0x4019b7c6  PS      : 0x00060e34  A0      : 0x80090536  A1      : 0x3ffd7ad0
0x4019b7c6: esp_vApplicationWaitiHook at /Users/wiki/esp/esp-idf/components/esp32/freertos_hooks.c:66

A2      : 0x00000008  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffd80e8
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc5190  A9      : 0x3ffc5174
A10     : 0x00000000  A11     : 0x00060021  A12     : 0x8008f538  A13     : 0x3ffc05a0
A14     : 0x00000003  A15     : 0x00060523  SAR     : 0x00000000  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4019b7c6:0x3ffd7ad0 0x40090533:0x3ffd7af0
0x4019b7c6: esp_vApplicationWaitiHook at /Users/wiki/esp/esp-idf/components/esp32/freertos_hooks.c:66

0x40090533: prvIdleTask at /Users/wiki/esp/esp-idf/components/freertos/tasks.c:4862

```

PhilColbert, 2018-05-26:
Also seeing this error 

ASSERT_PARAM(61455 1073461356), in ke_mem.c at line 409
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400870da  PS      : 0x00060b34  A0      : 0x8001a04c  A1      : 0x3ffbca50  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0xb86cf00f  A5      : 0x60008054  
A6      : 0x3ffc1030  A7      : 0x60008050  A8      : 0x800870d5  A9      : 0x3ffbca30  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4008700d  LEND    : 0x40087014  LCOUNT  : 0x00000000  

Backtrace: 0x400870da:0x3ffbca50 0x4001a049:0x3ffbca70 0x4008999a:0x3ffbcaa0 0x4018e311:0x3ffbcae0 0x4004240b:0x3ffbcb00 0x4001a637:0x3ffbcb20 0x40019d11:0x3ffbcb50 0x40055b4d:0x3ffbcb70 0x40189577:0x3ffbcb90 0x40189683:0x3ffbcbb0

Core 1 register dump:
PC      : 0x401f5962  PS      : 0x00060334  A0      : 0x80090e7e  A1      : 0x3ffdc230  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffdbb78  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc53c8  A9      : 0x3ffc538c  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80091a2f  A13     : 0x3ffde9b0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x401f5962:0x3ffdc230 0x40090e7b:0x3ffdc250

Rebooting...
ets Jun  8 2016 00:22:57

in sending BLE.

pCharacteristic->setValue(substr.c_str());
pCharacteristic->notify();

PhilColbert, 2018-06-02:
Hi malaimoo ,

Did you find anything out about this problem?

It keeps appearing on my system :(

Thanks

malaimoo, 2018-06-05:
@PhilColbert Hi, I can not find ke_mem.c either. and I have not solve this problem..

Yulong-espressif, 2018-06-05:
@PhilColbert @malaimoo We have find the issue now, we will solve it as soon as possible. Thank you very much.

PhilColbert, 2018-06-05:
Just in case this helps 

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40086f05  PS      : 0x00060734  A0      : 0x8001a04c  A1      : 0x3ffb3580
0x40086f05: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0xb86cf00f  A5      : 0x60008054
A6      : 0x3ffc1030  A7      : 0x60008050  A8      : 0x80086f05  A9      : 0x3ffb3560
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40086e3d  LEND    : 0x40086e44  LCOUNT  : 0x00000000
0x40086e3d: r_assert_param at ??:?

0x40086e44: r_assert_param at ??:?

Backtrace: 0x40086f05:0x3ffb3580 0x4001a049:0x3ffb35a0 0x400897ca:0x3ffb35d0 0x4018e2a9:0x3ffb3610 0x4004240b:0x3ffb3630 0x4001a637:0x3ffb3650 0x40019d11:0x3ffb368
0 0x40055b4d:0x3ffb36a0 0x4018950f:0x3ffb36c0 0x4018961b:0x3ffb36e0
0x40086f05: r_assert_param at ??:?

0x400897ca: r_lld_pdu_tx_prog at ??:?

0x4018e2a9: r_lld_pdu_data_send at ??:?

0x4018950f: r_rw_schedule at ??:?

0x4018961b: btdm_controller_task at ??:?

Core 1 register dump:
PC      : 0x400f5b02  PS      : 0x00060434  A0      : 0x800e4cdd  A1      : 0x3ffde260
0x400f5b02: loop() at F:/msys32-30052018/home/main.cpp:2333

A2      : 0x00000001  A3      : 0x00000048  A4      : 0x000001ff  A5      : 0x00000001
A6      : 0x000000d1  A7      : 0x00000079  A8      : 0x800f5aa0  A9      : 0x3ffde240
A10     : 0x00000017  A11     : 0x00000015  A12     : 0x00000000  A13     : 0x0000000a
A14     : 0x3ffdeb38  A15     : 0x3ffde190  SAR     : 0x0000000a  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400f5b02:0x3ffde260 0x400e4cda:0x3ffdebe0
0x400f5b02: loop() at F:/msys32-30052018/home/main.cpp:2333

0x400e4cda: loopTask(void*) at F:/msys32-30052018/home/main.cpp:17 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5880
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:15016
entry 0x40078630

malaimoo, 2018-06-12:
@Yulong-espressif Hi, Anything update about this issue? Looking forward for solving this issue.

PhilColbert, 2018-06-12:
I am still waiting 4 months for another bug to be fixed.... so I think we have a long wait!

jakkra, 2018-06-20:
We also see this issue sending data over GATT with 247 MTU.
This crash happens 100% of the time in our application just after about 5s sending data at max speed to the other device. This needs to be fixed for esp-idf 3.1, otherwise our application is unusable with it. It works fine on 3.0.

Yulong-espressif, 2018-06-21:
@All We have fix this bug and push it to the master now, the commit ID is: 225c4371.
Thank you very much! Any question, don't hesitate to let me know.

malaimoo, 2018-06-22:
@Yulong-espressif I can't find commit 225c4371 at master branch, Is it enough to pull the latest master to solve this?

malaimoo, 2018-06-22:
@Yulong-espressif I test the latest master code , the bug is not solved...

Yulong-espressif, 2018-06-22:
@malaimoo It may be due to the problem that gitlab and github are out of sync, which caused you to find no corresponding commit on github. You will see the corresponding commit today or tomorrow, thanks.

mws-rmain, 2018-06-22:
@Yulong-espressif Same problem here. Your commit is still not visible on GitHub 
:-(
Awaiting sync.

mws-rmain, 2018-06-22:
When I first encountered this issue, I was chasing other problems, and found a quick-fix was to restrict everything to a single core (far from optimal, but it allowed me to progress). 

If anyone wants to try this, run 'make menuconfig', and:
- set bluedroid locked to core 0
- enable FreeRTOS unicore;
- set FreeRTOS locked to core 0

PhilColbert, 2018-06-25:
Just updated, still getting the issue....

ASSERT_PARAM(61455 1073461356), in ke_mem.c at line 409

Any sign of this commit yet ? :)

mws-rmain, 2018-06-25:
@Yulong-espressif could you please check your commit? It's still not visible publicly.
The last commit visible on Master was from 5 days ago, yet there are several commits to 3.1 from 5 hours ago. This makes be believe your commit is held up somehow.

Yulong-espressif, 2018-06-25:
@mws-rmain I will check it again, and this bug also fix to the v3.1 branch. Thank you very much.

mws-rmain, 2018-06-26:
@Yulong-espressif any word on this?
You made your commit 5 DAYS AGO, and still no public sign of it.
Please help us understand the release procedures? 
Is there anything the users on this thread can do to highlight that we are ALL WAITING FOR THIS FIX?

I could understand delays while waiting & working toward a fix when the cause is unknown, or if there were considerations regarding how to implement the fix. It is very frustrating to know a fix is defined & committed, but for some unknown reason, we are denied access to it.

Yulong-espressif, 2018-06-26:
@mws-rmain I have find the commit already in the new master now. Please see the capture.
[image]

mws-rmain, 2018-06-26:
Thanks very much @Yulong-espressif.
I guess part of the issue with checking was I was expecting to see your name against the commit (not 'jack0c'), and the commit you mentioned above (225c437) not 'b7d02f7'.
I'm building & testing now.
Thanks for your efforts.
Best Regards,
Robb.

Yulong-espressif, 2018-06-26:
@malaimoo In our git management system, only a few people have permission to merge branches to the Master branch, so you can't see my name. Thank you very much. Any question, don't hesitate to let me know.

mws-rmain, 2018-06-26:
@Yulong-espressif seems like this commit has resolved this problem for me.
Thanks again.

PhilColbert, 2018-06-27:
Fixed here, along with some other BLE related problems I was getting :)

Stayed up for 10 hours no issues.

Thanks

malaimoo, 2018-06-27:
The problem have been solved , Thanks.

closed 2018-06-27 by malaimoo

BogdanDIA, 2024-07-24:
Hi, I see the issue in IDF5.2.2, see below:

I (4550) main_task: Returned from app_main()
ASSERT_PARAM(61455 1073442656), in ke_mem.c at line 409
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400909e0: f01d020c 00004136 f01d0000
0x400909e0: btdm_sleep_check_duration at /home/user/projects/esp/esp-idf/components/bt/controller/esp32/bt.c:914

Core  1 register dump:
PC      : 0x400909e7  PS      : 0x00060434  A0      : 0x8008928a  A1      : 0x3ffce240  
0x400909e7: r_assert at /home/user/projects/esp/esp-idf/components/bt/controller/esp32/bt.c:1926

A2      : 0x00000000  A3      : 0x0000f00f  A4      : 0x3ffb6f60  A5      : 0x3ff9cac6  
A6      : 0x00000199  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffce1b0  
A10     : 0x00000000  A11     : 0x3ffce1d3  A12     : 0x3ffce17f  A13     : 0x00000039  
A14     : 0x00000000  A15     : 0x3ffce184  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
0x4000c349: memcpy in ROM
0x4000c36b: memcpy in ROM