espressif/esp-idf #2001: SPIFFS mount fails when flash encryption is enabled

Buffalchill, 2018-05-25:
I tried the SPIFFS example while flash encryption was enabled. 
the .bin file was encrypted with a pre-generated-key from espsecure.py 
and uploaded with the esptool.py

The result was: mount fail. Isn't it possible to use Spiffs if flash encryption is enabled ?

[image]

me-no-dev, 2018-05-25:
SPIFFS is not supported when flash encryption is enabled. You can use FATFS instead

Buffalchill, 2018-05-25:
but FATFS is only for SD card right ?

me-no-dev, 2018-05-25:
FATFS is supported on the flash as well. https://github.com/espressif/esp-idf/tree/master/examples/storage/wear_levelling it's the wear leveling lib that makes it possible to be used when flash is encrypted.

Buffalchill, 2018-05-25:
Thank you ! Is this also available in the Arduino-Esp32 ? (I didn't find a sketch or a example, yet)

Buffalchill, 2018-05-25:
I tried the FATFS wear leveling example from the esp-idf but it doesn't seem to work with flash encryption.
I also tried the arduino-esp32 EEPROM example which doesn't work too.
It seems like there is no way to store data, without loosing it when esp32 shuts down  🌊 

[image]
encryption

Buffalchill, 2018-06-18:
Spiffs and Preferences work with encryption enanbled. Just didn't have the right partition table on my board

closed 2018-06-18 by Buffalchill

bastianhjaeger, 2020-06-15:
> Spiffs and Preferences work with encryption enanbled. Just didn't have the right partition table on my board
I do not see how SPIFFS can work with encryption (but I am trying to) if I see:

[image]
https://github.com/espressif/esp-idf/blob/master/components/spiffs/esp_spiffs.c#L166

@Buffalchill Could you elaborat this, please?