espressif/esp-idf #2003: Invalid magic number in feature/cmake branch

Gouthamm4G, 2018-05-25:
## Environment

- Development Kit :  TTGO LoRa 32
- Core : [ESP32]
- IDF version : 829e6c44 (feature/cmake branch)
- Development Env :		[CMake]
- Operating System :	[Windows]
- Power Supply :         [USB]

## Problem Description

Invalid magic number is used even when I don't change it.

In `main.c`,
```
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_LOGD("Main", "Magic Number is 0x%x", cfg.magic);
ESP_ERROR_CHECK(esp_wifi_init(&cfg));
```
In `monitor`,
```
D (819) Main: Magic Number is 0x1f2f3f4f
E (819) wifi: invalid magic number 3ffbb5f0, call WIFI_INIT_CONFIG_DEFAULT to init config
```
I compiled the same code without any changes including sdkconfig, in make environment in the latest master branch and no problems.

### Steps to repropduce

1. Set up Python 2.7, Cmake 3.11, ninja from installer, esp-idf (feature/cmake branch)
2. Run `python <path-to-idf.py> all flash monitor -p COM5 -b 115200`

projectgus, 2018-05-28:
Hi @Gouthamm4G ,

Thanks for reporting this.

If you build using the old build system (ie make, no cmake) but still in the CMake branch, do you still see this problem?

Can you please double check the git submodules are up to date in the checked out feature/cmake branch (git submodule update --all)?

The root cause of this error is probably either a stack overflow or a mismatch between the wifi binary libraries (which are in a submodule) and the wifi header. The master branch added some fields to the wifi_config_t structure which are not yet in the CMake branch, which will cause the "magic" field read by the libraries to read past the end of the structure (intentional, to avoid more subtle problems caused by version mismatches).

Gouthamm4G, 2018-06-05:
Sorry for the late reply. I was in the middle of a data migration and I did something wrong.
Anyway now my cmake build system throws a new error on even an example project. And I just saw that cmake related docs have disappeared from the docs server.
I don't understand what this means. I give up after three days of searching and editing files. I think maybe I just got the wrong commit after I reinitialized the repo due to a permission problem. Anyway I get this
```
Running cmake in directory D:\blink\cmake
Executing "cmake -G Ninja --warn-uninitialized D:\blink"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.17.0.windows.1")
-- Component names:
-- Component paths:
Traceback (most recent call last):
  File "D:/esp32/esp-idf[cmake]/tools/kconfig_new/confgen.py", line 203, in <module>
    main()
  File "D:/esp32/esp-idf[cmake]/tools/kconfig_new/confgen.py", line 80, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "D:\esp32\esp-idf[cmake]\tools\kconfig_new\kconfiglib.py", line 653, in __init__
    self.top_node)  # prev_node
  File "D:\esp32\esp-idf[cmake]\tools\kconfig_new\kconfiglib.py", line 1596, in _parse_block
    node)         # prev_node
  File "D:\esp32\esp-idf[cmake]\tools\kconfig_new\kconfiglib.py", line 1543, in _parse_block
    self._enter_file(sourced_file)
  File "D:\esp32\esp-idf[cmake]\tools\kconfig_new\kconfiglib.py", line 1127, in _enter_file
    .format(self._filename, self._linenr, e.message))
IOError: D:/esp32/esp-idf[cmake]/Kconfig:178: Could not open '' (EINVAL: invalid mode ('r') or filename). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances). Also note that e.g. $FOO in a 'source' statement does not refer to the environment variable FOO, but
rather to the Kconfig Symbol FOO (which would commonly have 'option env="FOO"' in its definition).
CMake Error at D:/esp32/esp-idf[cmake]/tools/cmake/kconfig.cmake:136 (message):
  Failed to run confgen.py
  (C:\Python27\python.exe;D:/esp32/esp-idf[cmake]/tools/kconfig_new/confgen.py;--kconfig;D:/esp32/esp-idf[cmake]/Kconfig;--config;D:/blink/sdkconfig;--defaults;D:/blink/sdkconfig.defaults;--create-config-if-missing;--env;COMPONENT_KCONFIGS=;--env;COMPONENT_KCONFIGS_PROJBUILD=
  D:/blink/main/Kconfig.projbuild;--env;IDF_CMAKE=y).  Error 1
```
After this I created a new clone of the latest in the branch and installed updated versions of mconf and tried again only to be hit with the same error. I checked that I have all the environmental variable set up.

projectgus, 2018-06-12:
Hi @Gouthamm4G,

Sorry for the slow reply from my side. The error is because of this, at the end of the confgen.py command line:
```
COMPONENT_KCONFIGS=;--env;COMPONENT_KCONFIGS_PROJBUILD=
```

Which has a root cause that can be seen here:
```
-- Component names:
-- Component paths:
```

Somehow the build system has decided there are no components to build. I'll add an error check that the build can't contain zero components, so at least the error is more predictable.

I have no explanation of why this happened, the only unusual thing I can see is `[` & `]` in IDF_PATH but this shouldn't be any problem. Can you please try running:

In IDF_PATH directory:
```
git pull --recurse-submodules
```

then, in blink project directory:
```
idf.py fullclean reconfigure
```
... and post the complete output from this last command?

Thanks for your patience working through this.

Gouthamm4G, 2018-06-15:
Sorry, got too excited and lost sight of the real problem

projectgus, 2018-06-15:
> Sorry, got too excited and lost sight of the real problem

Do you mean you're still seeing "invalid magic number"? Can you please check your submodules are up to date ("git submodule update"). The cmake build log will also contain some warnings if they are not up to date.

Angus

Gouthamm4G, 2018-06-15:
I don't see the problem anymore. But how do I do this
> If you build using the old build system (ie make, no cmake) but still in the CMake branch, do you still see this problem?
now?

Did you guys update the wifi subsystem library?

projectgus, 2018-06-15:
> I don't see the problem anymore. 

Sorry, I'm really confused. Is there still a problem? If so, which problem?

> If you build using the old build system (ie make, no cmake) but still in the CMake branch, do you still see this problem?

If you have the cmake branch, the old GNU make based build system (used in master) is still there alongside CMake. So you can build as if you were on master, ie:

```
rm -rf build
make -j4
```

(Deleting build directory when switching between CMake and the GNU Make system is not necessary but it is recommended.)

projectgus, 2018-06-15:
>Did you guys update the wifi subsystem library?

It's updated regularly. The output of `git status` and/or `git submodule update` will tell you if you have the correct version (latter command will also check out the correct version, if it's not already checked out).

Gouthamm4G, 2018-06-15:
Sorry If I was not clear. I have no errors whatsoever anymore including the "Invalid magic number". This issue can be closed now. I closed the issue once the build was successful and then I reopened it after seeing the issue title one more time.

The previous message was because I wanted to know the root cause of the problem. Nothing more. So should I close the issue? (Should've asked it before, would have made me think twice)

Thank you for your patience