espressif/esp-idf #2004: ESP32 crashes if  UART pins are changed in ble_spp_client example

jscastonguay, 2018-05-25:
# Environment

- Development Kit: ESP32-DevKitC
- Kit version: DevKitC v4
- IDF version: 358c822 
- Development Env: Make
- Operating System: Windows
- Power Supply: external 3.3V

# Problem Description

If the UART pins are changed (whatever UART0, UART1 or UART2 is used) on ble_spp_client, the ESP32 crashes with the following dump:

> Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
> Core 0 register dump:
> PC  	: 0x40122524  PS  	: 0x00060934  A0  	: 0x800466c6  A1  	: 0x3ffc0540
> A2  	: 0x00000000  A3  	: 0x40122524  A4  	: 0x00000000  A5  	: 0x00000010
> A6  	: 0x3ffb93cc  A7  	: 0x3ffb8360  A8  	: 0x80019fb8  A9  	: 0x000044ac
> A10 	: 0xf2f2f2f2  A11 	: 0x00000000  A12 	: 0x800905fb  A13 	: 0x00000022
> A14 	: 0x000023f0  A15 	: 0x3ffb6710  SAR 	: 0x00000018  EXCCAUSE: 0x00000007
> EXCVADDR: 0x00000000  LBEG	: 0x4000c2e0  LEND	: 0x4000c2f6  LCOUNT  : 0x00000000
> Core 0 was running in ISR context:
> EPC1	: 0x4008b327  EPC2	: 0x00000000  EPC3	: 0x00000000  EPC4	: 0x40122524
>  
> Backtrace: 0x40122524:0x3ffc0540 0x400466c3:0x3ffc0570 0x40085da6:0x3ffc0590 0x40086906:0x3ffc05b0 0x40087846:0x3ffc05d0 0x4008875b:0x3ffc05f0 0x40082659:0x3ffc0610 0x40062230:0x00000000

# Steps to reproduce

- Take the ble_spp_client.
- In spp_client_demo.c, change the following line:

`uart_set_pin(UART_NUM_0, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);`

for

`uart_set_pin(UART_NUM_0, GPIO_NUM_4, GPIO_NUM_5, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);`

- Run the example along with ble_spp_server example and the client will crash.
- Can try with UART_NUM_2, GPIO_NUM_16 and GPIO_NUM_17 or whatever, still crashing.

Thanks for your help.

koobest, 2018-06-03:
Hi, @jscastonguay,
    Thanks for your feedback. we have changed the uart pin in `spp_client_demo.c`, the client works well.  Can you provide your test code? 
Thanks !!

negativekelvin, 2018-06-03:
>by the way, GPIO16 and GPIO 17 is used by psram, we can not use these two pins.
Only with wrover, not wroom

jscastonguay, 2018-06-04:
Hi,

For a unknown reason, the crashes disappeared. We just missed what happened. Sorry about that. Now, it is working.

Thanks for your help.

JS

closed 2018-06-04 by jscastonguay