espressif/esp-idf #2006: build failed by file system is not *case-sensitive* but it is

iGyver, 2018-05-26:
I'm at the end of the toolchain setup from the sratch for MacOS. Actually there is just an issue with the last step the build.

* The file system is Case Sensitiv.
* The build procedure doesn't accept it.

Unfortunately I'm stuck now.

[image]

projectgus, 2018-05-28:
Hi Kay,

The "ctng" filesystem in the screenshot is mounted at `/Volumes/ctng` but the path in the error is `/Users/xxx/esp/crosstool-NG` which is part of your main filesystem.

Is there a symlink from one to the other? If so, a workaround might be to change `cd` into the "real" mount point directory and run ct-ng from there.

Angus

FayeY, 2018-06-04:
Hi @iGyver , did the above suggestion help you resolve the issue?

closed 2018-06-12 by FayeY

MGonzalez506, 2018-09-16:
I have the same error, I think I did @projectgus suggestion but nothing happened. (Using High Sierra 10.13.6)

[image]