espressif/esp-idf #2007: Bluetooth Classic HID client demo

Lunatix89, 2018-05-27:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v2
- Core (if using chip or module): ESP-WROOM32
- IDF version: Unknown (arduino-espressif32 @ 9570b77)
- Development Env:		VSCode with PlatformIO
- Operating System:		Windows
- Power Supply:         USB

## Problem Description
I'm trying to connect to a Xbox One S gamepad using bluetooth. Since the gamepad uses classic bluetooth I can not use the (very good) BLE library available for arduino, neither use the BLE examples as a reference.

I used the discovery example as a reference and can now discover the HID service offered by the gamepad - but I could not figure out how to actually connect to the discovered gamepad because I could not find an example for this case.

Is the functionality for BTC HID already implemented and/or are there any examples on how to communicate with such a device? 

Thanks,
Denis

blueMoodBHD, 2018-06-04:
Hi @Lunatix89  Sorry to tell you that,  we haven’t implemented this feature yet.

Lunatix89, 2018-06-05:
Okay, but that's good to know.
Any chance that this will be implemented in the future?

blueMoodBHD, 2018-06-05:
At present, we do not have this plan yet. 
I will inform you if we have plans to begin. 
Thanks.

blueMoodBHD, 2018-06-11:
Is there any other problems?

closed 2018-06-13 by FayeY

ricardoquesada, 2019-01-01:
@Lunatix89 did you find out how to do it? I'm trying to do more or less the same thing.

@blueMoodBHD 
I might be able to find out how to do the HID thing myself, but I would need a sample that shows to to interact with l2cap. Any link? Thanks!

MrSwiss, 2019-03-30:
@ricardoquesada 
Do you have any news, I would like to control my esp32 with xbox controller.

ricardoquesada, 2019-04-18:
> @ricardoquesada
> Do you have any news, I would like to control my esp32 with xbox controller.

yes. I'm not using bluedroid from esp-idf. Instead I'm using btstack: https://github.com/bluekitchen/btstack/
which supports BT Classic + HID. It is a commercial project, but free for open source.

And I implemented support for gamepads (Xbox One included) in this project:
https://retro.moe/unijoysticle2/
(which is open source, feel free to reuse the code, but you have to comply with BlueKitchen license)

MrSwiss, 2019-04-18:
Thank @ricardoquesada for following up. I currently use my labtop open javascript html5 webpage to send my controller keystroke to ESP32 web server. It for my personal use. Thanks, your method is better than mine. I will try.