espressif/esp-idf #2010: "localhost:3333: The system tried to join a drive to a directory on a joined drive." GDB Debugging how to resolve this

jegajith, 2018-05-29:
Hi I am getting this following error when I am trying to debug the ESP32 with a FTDI 2232HL Adapter 

"localhost:3333: The system tried to join a drive to a directory on a joined drive."
How to resolve this issue 

[image]

igrr, 2018-06-05:
Please make sure that openocd is running when GDB is started, and that local port 3333 is not blocked by a firewall. This error indicates that GDB can not open port 3333.

jegajith, 2018-06-06:
Thank  you Ivan , for your reply. Its not able to detect the driver,  I have changed the Windows Driver using the below Software. 

[image]

Now its working ! , But i observed one thing  the GDB debugging is inconsistent, often it will go offline , Every time i need to restart the GDB.  

Do you recommend any other JTAG Device ?

igrr, 2018-06-06:
We are working on improvements to JTAG debugging reliability. Please follow updates to OpenOCD releases, and file issues in OpenOCD repository if you see incorrect behavior.

closed 2018-06-06 by igrr

jegajith, 2018-06-07:
Thanks  for your support .

Waldkircher, 2018-08-22:
Hello, 
I am using the ESP-WROVER-KIT together with eclipse and I have configured the toolchain and eclipse as described in the manual from espressif on Windows 10. 

https://dl.espressif.com/doc/esp-idf/latest/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse

But if I am starting the debugger I was always getting the following error message (seen on screenshot below)... Do you have any idea what`s wrong ???

[image]

Open OCD seems to be OK..
[image]

Configuration of Eclipse is OK..
[image]

USB Driver is OK...
[image]

vgonet, 2019-01-25:
Hello, 
I've exactly the same issue. Have you an idea how to solve it ?
Best regards.