espressif/esp-idf #2013: UART for console output setting impacts on bootloading and wifi drivers ?

GPantoine, 2018-05-30:
This is my first time, please leave constructive comments !

## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Core (if using chip or module): ESP32-Wrover
- IDF version : commit  0d65f3b7

- Development Env:		[Make|Eclipse]
- Operating System:		[MacOS]
- Power Supply:         USB

## Problem Description

My application has a custom console task component that uses UART0 to reveive commands and send data and a wifi task. Until I add a Get_time function for into the wifi task, which is the same as the one in the IDF example for sntp (examples/protocols/sntp), all ran fine. After I added the function to app, the ESP32 keeps on resetting. If I keep declaration and defenition, but comment every call of the function in code, the application runs fine. What is weird is that Get_time cannot be called unless I type the command in the console.  

Where it gets even weirder, is when I change menuconfig UART for console output to default (uart0, tx = gpio1, rx = gpio) : Everything runs fine!!!   (component config -> ESP32-specific)

The reason I set UART for console output to none is because I have a custom console and needed to redirect ESP_LOGx to my console task "send" function for debugging purposes. Since I call esp_set_vprintf after boot, it doesn't matter in the application for now, but when product will be closer to final version, I won't redirect ESP_LOGx to my console task, and therefore will again put UART for console output to none...

## Debug Logs

See log file attached, contains outputs for both settings. As explained above, when setting is set to none, keeps on resetting and since there is no output for ESP_LOGx, we can't see how far it goes into bootloader. Please note that the first Reset outputs a different "garbage" message than the second and subsequent resets.

[Monitor_output.txt](https://github.com/espressif/esp-idf/files/2055239/Monitor_output.txt)

Thank for your feedbacks :D

koobest, 2018-06-06:
Hi, @GPantoine ,
    Did you find the cause of this problem? 
    From your description, it seems that  your WiFi task stack is overflowing, but I'm not sure. Does your WiFI task handle a lot of things? and can you kindly provide a code that can reproduce this issue?

thanks !!

GPantoine, 2018-06-08:
Hi @koobest,

No the issue hasn't been resolved (to my knowledge). Unfortunatly, I'm unable the post code today, this will have to go after the weekend :/ . Have you experienced this bug ?

koobest, 2018-06-20:
Hi, @GPantoine 
Is there any progress on this issue? I run `examples/protocols/sntp` to do the test but can't reproduce the problem.

GPantoine, 2018-06-20:
Hi @koobest,
Sorry for the delay. The Issue have not been resolved.

Here are some code which I been able to reproduce the issue :
[Uploading uart_bug.zip…]()

with that code, if  Uart for console output is set to uart0 (default), no problem.
If uart for console output is set to none, can't even enter bootloader.

Thank you !

koobest, 2018-06-25:
Hi,@GPantoine, 
The follwing is my output log when I run the code that you provided.(uart for console output set to none)
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4992
ho 0 tail 12 room 4
load:0x40078000,len:7796
load:0x40080000,len:7304
entry 0x40080390
LOG:Hello from console
```

FayeY, 2018-07-17:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.

closed 2018-07-17 by FayeY