espressif/esp-idf #2014: ESP-MESH Problem!

akshar001, 2018-05-31:
Hello,
I am using internal_reciever example for two days,
And there's one problem i am facing is it sometimes stucks at 

(1806) mesh: 2626<MESH_NEW_ROOT_VOTE_DONE>MESH_VOTE_STOP_FROM_HEALING, s_vote_scan_times:0

sometimes it connects to the wifi router but many times it is not?!
I have read the documentation several times but MESH_VOTE_STOP_FROM_HEALING is not mentioned anywhere...
here's the log..

I (30) boot: ESP-IDF v3.1-dev-1193-g64b56bef-dirty 2nd stage bootloader
I (30) boot: compile time 11:20:47
I (32) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18ea4 (102052) map
I (131) esp_image: segment 1: paddr=0x00028ecc vaddr=0x3ffb0000 size=0x038f0 ( 14576) load
I (137) esp_image: segment 2: paddr=0x0002c7c4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /Users/akshar/Desktop/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (138) esp_image: segment 3: paddr=0x0002cbcc vaddr=0x40080400 size=0x03444 ( 13380) load
I (153) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x7553c (480572) map
0x400d0018: _stext at ??:?

I (324) esp_image: segment 5: paddr=0x000a555c vaddr=0x40083844 size=0x0c4ac ( 50348) load
0x40083844: spi_flash_read at /Users/akshar/Desktop/esp/esp-idf/components/spi_flash/flash_ops.c:484

I (345) esp_image: segment 6: paddr=0x000b1a10 vaddr=0x400c0000 size=0x00000 (     0) load
I (355) boot: Loaded app from partition at offset 0x10000
I (355) boot: Disabling RNG early entropy source...
I (357) cpu_start: Pro cpu up.
I (361) cpu_start: Starting app cpu, entry point is 0x40080fa8
0x40080fa8: call_start_cpu1 at /Users/akshar/Desktop/esp/esp-idf/components/esp32/cpu_start.c:225

I (0) cpu_start: App cpu up.
I (371) heap_init: Initializing. RAM available for dynamic allocation:
I (378) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (384) heap_init: At 3FFBB0A0 len 00024F60 (147 KiB): DRAM
I (390) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (397) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (403) heap_init: At 4008FCF0 len 00010310 (64 KiB): IRAM
I (409) cpu_start: Pro cpu start user code
I (92) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (176) wifi: wifi driver task: 3ffc294c, prio:23, stack:4096, core=0
I (176) wifi: wifi firmware version: 4ae09c9
I (176) wifi: config NVS flash: enabled
I (176) wifi: config nano formating: disabled
I (186) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (196) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (216) wifi: Init dynamic tx buffer num: 32
I (216) wifi: Init data frame dynamic rx buffer num: 32
I (216) wifi: Init management frame dynamic rx buffer num: 32
I (216) wifi: Init static rx buffer size: 1600
I (226) wifi: Init static rx buffer num: 10
I (226) wifi: Init dynamic rx buffer num: 32
I (296) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
I (296) wifi: mode : sta (30:ae:a4:45:b1:ac) + softAP (30:ae:a4:45:b1:ad)
I (1296) wifi: Set ps type: 0

I (1296) mesh_main: <MESH_EVENT_STARTED>heap:190828
I (1296) mesh_main: mesh starts successfully, heap:190828, root not fixed

I (1656) mesh: [SCAN][ch:6]AP:2, otherID:0, idle:0, mAP:1, root:0, top mAP:0[Layer:0/0.ing]router found
I (1656) mesh: 916, [SCAN]init rc[30:ae:a4:45:b1:ad,-51], mine:0, voter:0
I (1666) mesh: 952, vote myself, router rssi:-51 > voted rc_rssi:-120
I (1666) mesh: [SCAN:1/10]rc[128][30:ae:a4:45:b1:ad,-51], self[30:ae:a4:45:b1:ac,-51,reason:0,votes:1,idle][mine:1,voter:1(1.00)percent:1.00][128,1,30:ae:a4:45:b1:ad]

I (2046) mesh: [VOTE][ch:6]AP:3, otherID:0, idle:0, mAP:1, root:0, [Layer:1]router found
I (2046) mesh: 2626<MESH_NEW_ROOT_VOTE_DONE>MESH_VOTE_STOP_FROM_HEALING, s_vote_scan_times:0

yudongchen95, 2018-06-04:
BUG

closed 2018-06-20 by igrr

akshar001, 2018-06-26:
Yes bug has been fixed...

rockineagle, 2018-11-09:
How ?