espressif/esp-idf #2015: addrinfo error: DNS lookup failed err=201 res=0x0

KryvosheyaIvan, 2018-05-31:
Hello!
I am using example http_request and everything was ok!

But recently I had to choose different Wi-Fi router and I faced the following issue:

`        int err = getaddrinfo(WEB_SERVER, "WEB_PORT", &hints, &res);		

	printf("\n Error getaddrinfo = %d !\n", err);

        if(err != 0 || res == NULL) {
            ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
            vTaskDelay(1000 / portTICK_PERIOD_MS);
            continue;
        }`

It prints out:
`E (71016) example: DNS lookup failed err=201 res=0x0`

And I have no idea what is it a problem and how to solve it!
Also i have noy founded 201 error code in the Internet as well as in the getaddrinfo declaration.
Can anyone help me?

negativekelvin, 2018-05-31:
https://gcc.gnu.org/onlinedocs/gcc-4.8.5/cpp/Stringification.html

KryvosheyaIvan, 2018-05-31:
@negativekelvin, please explain how it can help me?
I mean how it relates to my problem?

igrr, 2018-05-31:
You are passing literally "WEBPORT" for the port, there is no macro expansion happening.

FayeY, 2018-06-04:
Hi, did the above suggestion help you resolve the issue?

KryvosheyaIvan, 2018-06-04:
@FayeY , yes! Thank you!

closed 2018-06-04 by FayeY