espressif/esp-idf #2019: setting up tool chain on eclipse using windows

ahmedawais, 2018-05-31:
i followed all the instruction as per https://esp-idf.readthedocs.io/en/latest/get-started/eclipse-setup-windows.html#project-properties

but in eclipse it shows error although i set up the environment and path according to instruction
Description	Resource	Path	Location	Type
Program "g++" not found in PATH	firstesp		Project Properties, C++ Preprocessor Include.../Providers, [CDT Cross GCC Built-in Compiler Settings] options	C/C++ Scanner Discovery Problem

Description	Resource	Path	Location	Type
Program "gcc" not found in PATH	firstesp		Project Properties, C++ Preprocessor Include.../Providers, [CDT Cross GCC Built-in Compiler Settings] options	C/C++ Scanner Discovery Problem

Please help me

PhilColbert, 2018-06-02:
Are you sure you have changed the compiler to Cygwin GCC and added the correct path ?

FayeY, 2018-06-04:
Hi @ahmedawais , did the above suggestion help you resolve the issue?

projectgus, 2018-06-13:
HI @ahmedawais ,

It seem unusual that the error mentions "CDT Cross GCC" if compiler type is set to Cygwin.

Can you please attach your `.project` and `.cproject` files to this issue?

Angus

projectgus, 2018-06-26:
@ahmedawais 
Another note on this:

> Program "g++" not found in PATH
> Program "gcc" not found in PATH

These errors suggest you need to check the toolchain setting under Project Properties -> "Preprocessor Include Paths" -> "Providers" -> "CDT GCC Built-In Compiler Settings Cygwin" and change the "Command to Get Compiler Specs" to `xtensa-esp32-elf-gcc ${FLAGS} -E -P -v -dD "${INPUTS}"` (see Windows Eclipse Setup Guide for the same steps).

Does this help fix the problem?

ahmedawais, 2018-06-28:
no still i am having the same problem

dsptech, 2018-07-11:
Hi @ahmedawais,
I experienced a similar issue some time ago. If the issue is still on, you could create a batch file "checkEnvironment.bat" (see the bottom of this message) and save it in your project directory.

After that, change the command inside CDT GCC Built-In Compiler Settings Cygwin by writing:
' checkEnvironment.bat xtensa-esp32-elf-gcc ${FLAGS} -E -P -v -dD "${INPUTS}" ' 

After the final click on "Apply and close" the batch will be called from eclipse (any error from the console must be ignored) 
The log of the batch file (placed in the project directory) could help you to check if the shell is receiving the correct environment (paths, commans, etc...).
Note: you could find a double log because the batch will be called two times (one call for c scan and one call for c++ scan)

regards.

--- checkEnvironment.bat ----------

@echo off
set _logFile=checkEnvironment.log
echo ---------------------------------------------------------------------------- >>%_logFile%
echo checkEnvironment STARTED: "%DATE%" "%TIME%" >>%_logFile%
echo . >>%_logFile%
echo RECEIVED COMMAND LINE (9PARs): %0 %1 %2 %3 %4 %5 %6 %7 %8 %9 >>%_logFile%
echo . >>%_logFile%
echo CURRENT DIR : %cd% >>%_logFile%
echo . >>%_logFile%
echo CURRENT PATH: %PATH% >>%_logFile%
echo . >>%_logFile%
echo CURRENT TEMP: %TEMP% >>%_logFile%

--- end of  checkEnvironment.bat ----------

projectgus, 2019-03-01:
Closing due to a lack of updates and no project files.

Also, the Eclipse setup guide has changed a bit. If following the new guide still does not work then please open a new issue with the details.

closed 2019-03-01 by projectgus