espressif/esp-idf #2020: wifi connection sometimes fails silently

JcBernack, 2018-05-31:
I just noticed that _sometimes_ the device is unable to connect to wifi without any error whatsoever. `esp_wifi_connect()` returns `ESP_OK`, but no further event is generated and no connection is established.

Relevant log messages:
```
I (274) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 1
I (284) wifi: mode : sta (b4:e6:2d:b0:76:a9)
I (284) network: wifi event loop id: 2
I (284) network: station start
I (294) network: connecting to: mywifi
I (424) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (1404) wifi: state: init -> auth (b0)
I (2404) wifi: state: auth -> init (2)
I (2404) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
```

The `network` messages are from me. As you can see I get the event for `SYSTEM_EVENT_STA_START`, in the event handler I call `ESP_ERROR_CHECK(esp_wifi_connect());` and that's it. There is no event after that.

My program currently just connects to wifi, does very little and then goes to sleep for two seconds. I can watch this loop run a few times but sooner or later it will hang at the above situation. Currently I use `xEventGroupWaitBits` with an infinite timeout to wait for a connection event bit I set/clear from the wifi event handler. So this situation causes my program to hang foreover.

As a workaround I can use a finite timeout and restart wifi until it works, but this should not be necessary or did I miss something?

This happens on the current esp-idf master (commit 64b56bef).

liuzfesp, 2018-06-05:
@JcBernack does the SYSTEM_EVENT_STA_DISCONNECTED event happen? How do you handle the SYSTEM_EVENT_STA_DISCONNECTED?

JcBernack, 2018-06-09:
No, there is no event. If I would get a disconnected event I would try to reconnect:

```
case SYSTEM_EVENT_STA_DISCONNECTED: {
    ESP_LOGI(TAG, "disconnected");
    xEventGroupClearBits(system_event_group, EVENT_BIT_NETWORK_CONNECTED);
    if (reconnect) {
        // this is a workaround as ESP32 WiFi libs don't currently auto-reassociate
        ESP_LOGI(TAG, "reconnecting");
        ESP_ERROR_CHECK(esp_wifi_connect());
    } else {
        ESP_LOGD(TAG, "stopping");
        ESP_ERROR_CHECK(esp_wifi_stop());
    }
    break;
}
```

vonnieda, 2018-06-10:
Sounds like what I saw in https://github.com/espressif/esp-idf/issues/1797. Espressif said they fixed it, but I have not yet had a chance to test.

xentec, 2018-06-15:
Can confirm issue on ESP32D0WDQ6 rev. 1 on master.
After flashing WiFi works as expected. A reset makes it fail to connect without any catchable error (only state change logged).
```
I (0) cpu_start: App cpu up.
I (448) heap_init: Initializing. RAM available for dynamic allocation:
I (455) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (461) heap_init: At 3FFD1F40 len 0000E0C0 (56 KiB): DRAM
I (467) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (474) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (480) heap_init: At 40095040 len 0000AFC0 (43 KiB): IRAM
I (486) cpu_start: Pro cpu start user code
I (169) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (190) wifi: wifi driver task: 3ffdb0ec, prio:23, stack:4096, core=0
I (190) wifi: wifi firmware version: 974aa77
I (190) wifi: config NVS flash: enabled
I (190) wifi: config nano formating: disabled
I (200) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (210) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (230) wifi: Init dynamic tx buffer num: 32
I (230) wifi: Init data frame dynamic rx buffer num: 32
I (230) wifi: Init management frame dynamic rx buffer num: 32
I (230) wifi: Init static rx buffer size: 1600
I (240) wifi: Init static rx buffer num: 10
I (240) wifi: Init dynamic rx buffer num: 32
I (320) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
I (320) wifi: mode : sta (24:0a:c4:13:0e:98)
D (320) wifi-app: event 2
I (320) bt: init controller
I (320) BTDM_INIT: BT controller compile version [bc70352]

I (330) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (580) bt: init api
I (660) bt: The number handle: 6
D (660) wifi-app: wifi data found on flash
I (800) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1800) wifi: state: init -> auth (b0)
I (1810) wifi: state: auth -> auth (8a0)
I (2800) wifi: state: auth -> init (2)
I (2800) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
```
However resetting it again produces normal behavior. The bug occurs after every second reset.

negativekelvin, 2018-06-19:
Have also seen this problem and it was improved after recent commits but just had a station drop off the network after 6 days (before it was 1-2 days) and required power cycle.

XinDeng11, 2018-06-24:
@JcBernack @xentec @negativekelvin Have you invoke any scan operation before connected?

JcBernack, 2018-06-25:
@XinDeng11 Hm, on first boot-up I probably did a scan yes. I will try to reproduce the issue and check if scanning has an effect.

xentec, 2018-06-25:
@XinDeng11 no explicit scan, just an `esp_wifi_connect` with a valid `wifi_config_t`.

XinDeng11, 2018-06-26:
@JcBernack  Because we found there is one case would case this issue, one STA disconnected, it will try to reconnect if you invoke connect in disconnect event, however, if there is a user scan come, it will report scan_done event, which would stop the reconnect flow, and if there is no other connect cmd come, the reconnect will stop forever.

negativekelvin, 2018-07-03:
Have not seen this problem again yet, station has been up for 14days

negativekelvin, 2018-07-09:
Finally dropped again , no connection, dhcp lease expired. Don't have any logs though.

XinDeng11, 2018-08-03:
Hi, @negativekelvin, recently, we fix two bugs couldn't reconnect. However, you don't have log. I couldn't judge whether it can help you, but you can use the latest do test, but it still need few days merge

negativekelvin, 2018-08-04:
@XinDeng11 it looks like those fixes are related to coexist but I am not using Bluetooth.  Is there any effect without Bluetooth?

XinDeng11, 2018-08-05:
@negativekelvin  No, the change hasn't merge yet,  it is only about wifi, please wait a few days

XinDeng11, 2018-08-07:
@negativekelvin The change has merged, you can use the latest IDF do test again, thanks