espressif/esp-idf #2021: esp_bluedroid_init and esp_bluedroid_deinit documentation is ambiguous

mreutman, 2018-06-01:
## Environment
- Development Kit: none
- IDF version:  6ceedabb
- Development Env: Make
- Operating System:	 Ubuntu

## Problem Description
The documentation in the **esp_bt_main.h** header file regarding when to call these functions is not very clear and appears to be contradicted based upon how the information is interpreted. For example, the documentation for **esp_bluedroid_init** states "**Init and alloc the resource for Bluetooth, must be prior to every bluetooth stuff**". If one is to take this in a completely absolute manner such that no other bluetooth calls are made before calling, then this appears to be contradicted in the various bluetooth examples provided where calls to functions such as **esp_bt_controller_init** and **esp_bt_controller_enable** are called prior. Without any clarification, it is possible to class those functions as _blue tooth stuff_.

### Expected Behavior
Well defined calling order in the documentation to know when and how to call functions to initialize and deinitialize bluetooth capabilities. 

### Actual Behavior
Ambiguous documentation.

## Other items if possible
I know it's easy to overlook documentation, but it should be viewed as essential for the more complex components of the ESP32 system such as Bluetooth. The examples are a real help figuring this stuff out, but it would be preferable to not develop based on inference alone.

One idea if Espressif resources are limited such that fixing this would be far out in the future, if someone with knowledge of the Bluetooth side of things could point me to the authoritative and relevant Bluetooth info, I could try and document this myself and submit a pull request.

Thanks in advance!

Weijian-Espressif, 2018-06-06:
@mreutman Thanks for your advice, we will modify this document as soon as possible. We very much welcome you to submit a pull request to us. Thanks.

esp-cjh, 2023-11-20:
This issue will be closed and that they can reopen anytime.

closed 2023-11-21 by Indastri