espressif/esp-idf #2022: How do I SPI commucation

genoinosonda, 2018-06-01:
Hello guys,

I'm starting my first projects with ESP-IDF, and I'm finding it very practical and effective.

I am currently using the HELTEC ESP32, it has an integrated chip (sx1276) that is connected by SPI bus.

Reading the documentation I found that the ESP-IDF only works on only two SPI buses, being: HSPI / VSPI, but the mapping below does not follow the established standard.

The sx1276 is mapping to the following pins:

  #define pin_SCK 5 // - SX1278's pin sck
  #define pino_MISO 19 // - SX1278's pin miso
  #define pino_MOSI 27 // - SX1278's pin mosi
  #define pin_SS 18 // - SX1278's pin slave select
  #define pin_RST 14 // - SX1278's pin reset
  #define pin_DI0 26 // - SX1278's IRQ pin

  My main question is, is it possible to create SPI communication using ESP-IDF in this case?

  Thank you.

me-no-dev, 2018-06-01:
Something like this will work: (note it does not init the RST pin or setup the IRQ, but that is separate code)
```cpp
#define pin_SCK 5 // - SX1278's pin sck
#define pin_MISO 19 // - SX1278's pin miso
#define pin_MOSI 27 // - SX1278's pin mosi
#define pin_SS 18 // - SX1278's pin slave select
#define pin_RST 14 // - SX1278's pin reset
#define pin_DI0 26 // - SX1278's IRQ pin

static spi_device_handle_t spi = NULL;

esp_err_t lora_spi_init()
{
    esp_err_t ret;
    spi_bus_config_t buscfg={
        .miso_io_num= pin_MISO,
        .mosi_io_num=pin_MOSI,
        .sclk_io_num=pin_SCK,
        .quadwp_io_num=-1,
        .quadhd_io_num=-1,
        .max_transfer_sz=32*8 //in bits (this is for 32 bytes. adjust as needed)
    };
    spi_device_interface_config_t devcfg={
        .clock_speed_hz=1*1000*1000,//this is for 1MHz. adjust as needed
        .mode=0,
        .spics_io_num= pin_SS,
        .queue_size=3,//how many transactions will be queued at once
    };
    ret=spi_bus_initialize(HSPI_HOST, &buscfg, 1);
    ESP_ERROR_CHECK(ret);
    ret=spi_bus_add_device(HSPI_HOST, &devcfg, &spi);
    ESP_ERROR_CHECK(ret);
    
    return ret;
}
```

genoinosonda, 2018-06-01:
Thank you my friend.
The code worked perfectly.
Hugs

closed 2018-06-01 by genoinosonda