espressif/esp-idf #2026: Assert Seen

hvishu, 2018-06-04:
One of our customers is seeing below reset issue and log says an assert is done from below.  We are using 3.0 release from ESP.  What is the meaning of this assert?
    uint32_t getEntryAddress(size_t entry) const
    {
        **assert(entry < ENTRY_COUNT);**

Also, is there any redirection can be done for asserts used in ESP components.  i.e. Can application control what to be done when an assert happens?

[image]

igrr, 2018-06-05:
Hi @hvishu, please use [IDF monitor tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html) (`make monitor`) — it will decode the backtrace automatically. Please post the decoded backtrace here.
Also please use `esptool.py` to download the problematic NVS partition:
```
$IDF_PATH/components/esptool_py/esptool/esptool.py --port COMx read_flash 0x9000 0x6000 nvs_dump.bin
```
(replace COMx with your port number). Please attach resulting nvs_dump.bin file here as well.

Regarding assertions, usually a failed assertion indicates an unexpected condition for which there is no recovery path in the code. You can disable assertions (in menuconfig, under Compiler options), but this is not advisable because the program will likely crash soon after the failed assertion anyway, but debugging output would be much less useful.

FayeY, 2018-06-12:
Hi @hvishu , did the above suggestion help you resolve the issue?

hvishu, 2018-06-12:
We increased the NVS partition size as it was at the border.  Right now seeing this issue, but I am not 100% sure lesser NVS partition was the root cause of the issue.

closed 2018-11-16 by igrr