espressif/esp-idf #2028: Esp_mesh_send doesn't sends message to external address

ugallu, 2018-06-04:
I'm not sure it is an issue, or its working properly and I'm doing something wrong.
Esp_mesh_send doesn't sends message to external address.

## Environment

- Development Kit:       none
- Core (if using chip or module): ESP-WROOM32
- IDF version: 64b56be
- Development Env:		VSCode
- Operating System:		Ubuntu 16.04
- Power Supply:         USB

Note: I'm using only one ESP.

## Problem Description

Esp_mesh_send doesn't sends message to external address.

I'm using a modified version of the idf mesh example ([https://github.com/espressif/esp-idf/tree/master/examples/mesh/internal_transceiver](https://github.com/espressif/esp-idf/tree/master/examples/mesh/internal_transceiver)) to send a messsage to my server, without success.
I checked the API's esp_mesh_send part: 
[http://esp-idf.readthedocs.io/en/latest/api-reference/mesh/esp_mesh.html#_CPPv213esp_mesh_sendPK11mesh_addr_tPK11mesh_data_tiA_K10mesh_opt_ti](http://esp-idf.readthedocs.io/en/latest/api-reference/mesh/esp_mesh.html#_CPPv213esp_mesh_sendPK11mesh_addr_tPK11mesh_data_tiA_K10mesh_opt_ti)
It says "if the packet is outgoing to external IP network such as an IP server address, translate IPv4:PORT known as “to”."
I tried to use MESH_PROTO_BIN, MESH_PROTO_JSON, MESH_PROTO_HTTP data types without any success.

### Expected Behavior

It should send some message to an external server.

### Actual Behavior

It doesn't sends anything.
I can ping the ESP32 and my port is open.

### Steps to repropduce

* set SSID and PASSWORD for your wifi in menuconfig
* connect to the same network with your pc
* get IP with ifconfig
* start netcat in your PC `nc -l -k -p 8089`
* Clone my repo: [https://github.com/ugallu/esp-mesh-socket](https://github.com/ugallu/esp-mesh-socket)
* Modify the IP number and port number at line 112,113.
* Make and Flash the program to the esp
* Check monitor and netcat. It should try to send a message when it got an IP.

### Code to reproduce this issue

 [https://github.com/ugallu/esp-mesh-socket](https://github.com/ugallu/esp-mesh-socket)

dongdong004, 2018-06-05:
If you want to send a message to external address, socket shoud be created by yourself. And using esp_mesh_recv_toDS() to receive packets destined to external IP network.
* Here's the brief of `esp_mesh_recv_toDS()`
```
  * @brief     receive a packet targeted to external IP network
  *            root uses this API to receive packets destined to external IP network
  *            root forwards the received packets to the final destination via socket.
  *            if no socket connection is ready to send out the received packets and this esp_mesh_recv_toDS()
  *            hasn't been called by applications, packets from the whole mesh network will be pending in toDS queue.
  *            Use esp_mesh_get_rx_pending() to check the number of packets available in the queue waiting
  *            to be received by applications in case of running out of memory in root.
  *            Use esp_mesh_set_xon_qsize() could configure the RX queue size, default:72. If this size is too large,
  *            and esp_mesh_recv_toDS() isn't called in time, there is a risk that a great deal of memory is occupied
  *            by the pending packets. If this size is too small, it will impact the efficiency on upstream. How to
  *            decide this value depends on the specific application scenarios.
```
* More examples may help you in  https://github.com/espressif/esp-mdf

negativekelvin, 2018-06-06:
Will there be official border router implementation?

closed 2018-06-13 by FayeY

dongdong004, 2018-06-13:
@negativekelvin 

```
 "that's not included in mesh"
 "normal router is able to establish a connection between mesh networks."

JerrysGarage, 2018-06-14:
@ugallu did you get this working? I am finding esp_mesh_send is hanging when trying to send data to an external IP.

ugallu, 2018-06-15:
I made a workaround. I sent data from nodes to root via 
`esp_err_t err = esp_mesh_send(NULL, &data, 0, NULL, 0);`
and on the root, I got the message with esp_mesh_recv.
Then I opened a standard socket to my server (or simply to netcat) and sent it with the socket's write method (this maybe can help, I called write(socket, message, messageLength), instead of read: [https://github.com/espressif/ESP8266_RTOS_SDK/issues/46](https://github.com/espressif/ESP8266_RTOS_SDK/issues/46))
Please send a message if you get further.

dongdong004, 2018-06-15:
@JerrysGarage @ugallu 

You could refer to the examples in https://github.com/espressif/esp-mdf
The lastest esp-idf master is not fully tested on ESP-MESH.
You'd better use esp-mdf submodule before v3.1 release.

ugallu, 2018-06-15:
I think you use the receiving API a bit off.
The first parameter needs a pointer, and the api will write the address of the sender to the allocated variable. Then you can process this address.

I created a branch named send2Root in my repo for you with a working and tested (I have only one esp32 at home, so I sent from root to root, but I think it should work) example:
[https://github.com/ugallu/esp-mesh-socket/blob/send2Root/main/mesh_main.c](https://github.com/ugallu/esp-mesh-socket/blob/send2Root/main/mesh_main.c)

_ps: I would use as few dynamic memory allocation as possible.
And be sure you use the correct esp-idf, I currently use this:
https://github.com/espressif/esp-idf/tree/a2556229aa6f55b16b171e3325ee9ab1943e8552
 with upgraded submodules (git submodule update --init --recursive)_

dongdong004, 2018-06-16:
@JerrysGarage 
* root receive to DS： https://github.com/espressif/esp-mdf/blob/master/components/protocol_stacks/mdf_wifi_mesh/mdf_wifi_mesh.c#L376
* node send data： https://github.com/espressif/esp-mdf/blob/master/components/protocol_stacks/mdf_wifi_mesh/mdf_wifi_mesh.c#L210

JerrysGarage, 2018-06-18:
Thank you both! What is the best way for a node to get the address dest_addr for Root?

ugallu, 2018-06-18:
Its NULL, if you need it just for sending message.
From the[ esp_mesh_send API](http://esp-idf.readthedocs.io/en/latest/api-reference/mesh/esp_mesh.html#_CPPv213esp_mesh_sendPK11mesh_addr_tPK11mesh_data_tiA_K10mesh_opt_ti): `to: the address of the final destination of the packet (1)if the packet is to root, just set “to” to NULL and set flag to zero`

JerrysGarage, 2018-06-18:
NULL works for me when using esp_mesh_send however when using the mdf_wifi_mesh_send function, the first line is MDF_PARAM_CHECK(dest_addr); which prevents me from passing a NULL.

JerrysGarage, 2018-06-23:
@dongdong004 @ugallu Got it working, Thanks for all the help!

danylook, 2018-08-22:
hi @JerrysGarage 
could you explain me how you did it? that what im looking for my project.
Regards
Daniel

vcmorini, 2020-04-29:
> @JerrysGarage
> 
> * root receive to DS： https://github.com/espressif/esp-mdf/blob/master/components/protocol_stacks/mdf_wifi_mesh/mdf_wifi_mesh.c#L376
> * node send data： https://github.com/espressif/esp-mdf/blob/master/components/protocol_stacks/mdf_wifi_mesh/mdf_wifi_mesh.c#L210
> * Use the esp-idf @ugallu said.

Hi @dongdong004 , could you please update the "node send data" example? (:

dongdong004, 2020-05-06:
> Hi @dongdong004 , could you please update the "node send data" example? (:

Hi @vcmorini,
- esp-mdf:
  - https://github.com/espressif/esp-mdf
- esp_mesh_send:
  - https://github.com/espressif/esp-mdf/blob/master/components/mwifi/mwifi.c#L631
- esp_mesh_recv_toDS
  - https://github.com/espressif/esp-mdf/blob/master/components/mwifi/mwifi.c#L1275