espressif/esp-idf #2030: When num_pin set to GPIO25, gpio bootloader_common_check_long_hold_gpio will not work proerly

mashaoze, 2018-06-05:
When num_pin set to GPIO25, gpio bootloader_common_check_long_hold_gpio will not work proerly. this function always return GPIO_LONG_HOLD, event if the pin is not pull down. Because GPIO25 is also used by RTC and DAC/ADC, need set peripheral register in addition.

bart112233, 2018-06-07:
Pins 24,25,32-39 is not enabled on reset, so need to enable it. Something like:
```
**#include "soc/io_mux_reg.h"**
...
esp_comm_gpio_hold_t bootloader_common_check_long_hold_gpio_new1(uint32_t num_pin, uint32_t delay_sec)
{
	**const uint32_t disabled_io2[]={IO_MUX_GPIO32_REG, IO_MUX_GPIO33_REG, IO_MUX_GPIO34_REG, IO_MUX_GPIO35_REG, IO_MUX_GPIO36_REG, IO_MUX_GPIO37_REG, IO_MUX_GPIO38_REG, IO_MUX_GPIO39_REG};
	const uint32_t disabled_io1[]={IO_MUX_GPIO25_REG, IO_MUX_GPIO26_REG};
	switch (num_pin) {
		case 25 ... 26:
			PIN_INPUT_ENABLE(disabled_io1[num_pin-25]);
			break;
		case 32 ... 39:
			PIN_INPUT_ENABLE(disabled_io2[num_pin-32]);
			break;
		default:
			break;
	}**
    gpio_pad_select_gpio(num_pin);
    gpio_pad_pullup(num_pin);
    uint32_t tm_start = esp_log_early_timestamp();
    if (GPIO_INPUT_GET(num_pin) == 1) {
        return GPIO_NOT_HOLD;
    }
    do {
        if (GPIO_INPUT_GET(num_pin) != 0) {
            return GPIO_SHORT_HOLD;
        }
    } while (delay_sec > ((esp_log_early_timestamp() - tm_start) / 1000L));
    return GPIO_LONG_HOLD;
}
```

Tanmayagarwal14, 2018-06-08:
Hello 
I want to use the gpio pins to connect push button to my NodeMCU esp-32s so I was using gpio pin 0,2,8,7.
Firstly I was reading there state and the changing it using the push button that i connected to them. 
The thing I observed was that  I was not able to toggle the Gpio 8.
Can anyone help me what is the problem.

I was using Arduino Ide for programming .Other three pins where working fine.

xiongyumail, 2018-06-08:
Hi,@Tanmayagarwal14 
I'm sorry that GPIO8(SD_DATA_1) is currently occupied by FLASH's DI pin. You can use other free pin.   
For details, please see [esp32_hardware_design_guidelines_en.pdf](https://www.espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf)  and [esp32_datasheet_en.pdf](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf)

Tanmayagarwal14, 2018-06-08:
Ya I know that. But I was able to use Gpio7 which is also connected to the Flash pin. That's why was having the doubt . Can you please clear my doubt.
Thanks

xiongyumail, 2018-06-08:
Hi,@Tanmayagarwal14
It's a good question, because GPIO7 (SD_DATA_0) will change only when ESP32 writes data to FLASH. The code is stored in FLASH, so ESP32 will continue to get data from FLASH. But you need to understand that using GPIO7 and GPIO8 as GPIO is illegal, unless you don't use FLASH, ha ha!

closed 2018-06-26 by igrr

alsaleem00, 2021-02-06:
@igrr, was this issue fixed in idf 4.0.1 or I have to do it like above?