espressif/esp-idf #2031: NVS found nvs_flash_init() nvs_flash_deinit() MEMORY LEAKS

lmayainsite, 2018-06-05:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Core (if using chip or module): [ESP32-Wrover-I|]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //b2ff235
- Development Env:		Make]
- Operating System:		[Ubuntu]
- Power Supply:         [external 5V]

## Problem Description

We have encountered a memory leak reported by the HEAP TRACING mechanism, the problem occurs with the following code

```
TEST_CASE(" ** simple test","[product]"){
    nvs_flash_init(); 
    nvs_flash_deinit();
    TEST_ASSERT_EQUAL_INT(true,0==0);
}

```

### Expected Behavior

No memory leaks detected

### Actual Behavior
124 bytes 'leaked' in trace (3 allocations)
total allocations 12 total frees 11

igrr, 2018-06-06:
This seems to be a lazy-initialized mutex which is used to protect the list of initialized partitions. Since it is allowed to call `nvs_flash_deinit` in one task and `nvs_flash_init_partition` in another task, we can't simply delete the mutex in `nvs_flash_deinit`. This is not a recurring leak though. If you call `nvs_flash_init` again, same mutex will be used.

lmayainsite, 2018-06-06:
Thanks @igrr

closed 2018-06-12 by FayeY