espressif/esp-idf #2036: BLE secure pairing: Bonding does not work when permission is ENC_MITM on characteristics.

boriszweimueller, 2018-06-07:
## Environment

- Development Kit:      [ESP32-Wrover-Kit and others]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): b2ff235b 
- Development Env:		[Make]
- Operating System:		[Ubuntu and MacOS]
- Power Supply:         [external 5V]

## Problem Description

Example Code: examples/bluetooth/gatt_security_server

Client Device: Various iPhones (6,6+,7,8,X) and Android. Same behaviour with ALL of these devices.

If I change the code to use passkey pairing with MITM:

```cpp
esp_ble_auth_req_t auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND;   
esp_ble_io_cap_t iocap = **ESP_IO_CAP_OUT**;
```
Everything works as expected. After Rebooting the ESP, my client device is still bonded and can connect/read/write without another pairing. 

HOWEVER:

If I additionally change the security permissions of the characteristics to 'xxx_ENC_MITM' e.g.

```cpp
    // Heart Rate Control Point Characteristic Value
    [HRS_IDX_HR_CTNL_PT_VAL]             =
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&heart_rate_ctrl_point, **ESP_GATT_PERM_WRITE_ENC_MITM** | **ESP_GATT_PERM_READ_ENC_MITM**,
      sizeof(uint8_t), sizeof(heart_ctrl_point), (uint8_t *)heart_ctrl_point}},
```

On the first connect, passkey pairing is triggered as expected, and the characteristic can be accessed.

**But then, after rebooting the ESP, the client device CAN NOT reconnect/repair with the ESP.**

The lowlevel error ist:
```
I (16062) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
E (18472) BT_GATT: GATT_INSUF_AUTHENTICATION
```

This worked perfectly with release V2 and release V3 RC1. 

I checked the connection with OSX' packet-logger and it seems the connection does really use MITM and is encrypted. (This is due to the fact, that security is switched on in the GATT CONNECT handler in the example)

Does anybody know/can confirm if this is a bug? 

Thank you very much!
Boris

eminarcissus, 2018-06-13:
I’m facing the same problem, looking forward to dig it a little deeper to see what’s the problem here.
Also the connection bond itself is not correctly settled somehow, every time I will have to rebound to the device after it disconnects

boriszweimueller, 2018-06-28:
It does still not work with latest IDF. I already tried different versions (3.0.1, 3.1rc, master)...

chegewara, 2018-06-28:
I have to disagree. I have to mention im not using plain esp-idf but im using Neil Kolban's library.
Ive got just one single issue with bonding but like i said it was just one time. 

My procedure:
- server does not force to connect with secure connection (ESP_BLE_SEC_ENCRYPT_MITM),
- all other settings like in op topic,
- when android device is connecting with nRF connect, nothing unusual is happening,
- when i try to read characteristic first time bonding with passkey is triggered:
`E (213466) BT_GATT: GATT_INSUF_ENCRYPTION`
`D (213611) BLEUtils: Received a GAP event: ESP_GAP_BLE_SEC_REQ_EVT`
- after providing pass key devices are bonded and characteristic is read,
- disconnect device,
- recnnect adroid and esp32,
- read characteristic without passkey because devices are bonded.

@boriszweimueller  Let me know if im doing something wrong to perform test.

chegewara, 2018-06-28:
Just in case, here is part of logs when devices getting bonded. Just ignore that part is pointing class and look at part of message from bluedroid (like this: ESP_GAP_BLE_PASSKEY_NOTIF_EVT):
```
I (11818) BLEDevice: ESP_GAP_BLE_PASSKEY_NOTIF_EVT
I (11823) BLEDevice: passKey = 114885
I (11827) SampleServer: On passkey Notify number:114885
D (11833) BLEServer: BLEServer ... handling GAP event!
D (11960) BLEUtils: Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
D (11960) BLEUtils: [status: 0, bd_addr: 76:e0:bb:76:56:75, min_int: 0, max_int: 0, latency: 0, conn_int: 6, timeout: 500]
D (11967) BLEServer: BLEServer ... handling GAP event!
W (19873) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
D (19907) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (19907) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (19907) BLEDevice: ESP_GAP_BLE_KEY_EVT
I (19912) BLEDevice: key type = ESP_LE_KEY_LENC
D (19917) BLEServer: BLEServer ... handling GAP event!
D (19922) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (19928) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (19937) BLEDevice: ESP_GAP_BLE_KEY_EVT
I (19938) BLEDevice: key type = ESP_LE_KEY_PENC
D (19943) BLEServer: BLEServer ... handling GAP event!
D (19949) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (19954) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (19960) BLEDevice: ESP_GAP_BLE_KEY_EVT
I (19965) BLEDevice: key type = ESP_LE_KEY_LID
D (19970) BLEServer: BLEServer ... handling GAP event!
D (19975) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (19981) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (19986) BLEDevice: ESP_GAP_BLE_KEY_EVT
I (19991) BLEDevice: key type = ESP_LE_KEY_LCSRK
D (19996) BLEServer: BLEServer ... handling GAP event!
D (20003) nvs: nvs_open_from_partition bt_config.conf 1
D (20007) nvs: nvs_set_blob bt_cfg_key0 526
D (20015) nvs: nvs_close 4
D (20016) nvs: nvs_open_from_partition bt_config.conf 1
D (20019) nvs: nvs_set_blob bt_cfg_key0 526
D (20027) nvs: nvs_close 5
D (20027) nvs: nvs_open_from_partition bt_config.conf 1
D (20032) nvs: nvs_set_blob bt_cfg_key0 526
D (20039) nvs: nvs_close 6
D (20040) nvs: nvs_open_from_partition bt_config.conf 1
D (20044) nvs: nvs_set_blob bt_cfg_key0 526
D (20051) nvs: nvs_close 7
D (20052) nvs: nvs_open_from_partition bt_config.conf 1
D (20056) nvs: nvs_set_blob bt_cfg_key0 526
D (20101) nvs: nvs_close 8
D (20101) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (20102) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (20104) BLEDevice: ESP_GAP_BLE_KEY_EVT
I (20109) BLEDevice: key type = ESP_LE_KEY_PID
D (20114) BLEServer: BLEServer ... handling GAP event!
D (20119) BLEUtils: Received a GAP event: ESP_GAP_BLE_AUTH_CMPL_EVT
D (20126) BLEUtils: [bd_addr: 76:e0:bb:76:56:75, key_present: 0, key: ***, key_type: 0, success: 1, fail_reason: 0, addr_type: ***, dev_type: ESP_BT_DEVICE_TYPE_BLE]
I (20140) BLEDevice: ESP_GAP_BLE_AUTH_CMPL_EVT

```

chegewara, 2018-06-28:
@boriszweimueller I just found something in my logs:
`E (11480) BT_GATT: GATT_INSUF_AUTHENTICATION: MITM Required`
Its log before bonding and is different than yours.

boriszweimueller, 2018-06-28:
Hi.

I don't know Neil's library in detail (I know his book though!), but:
- Can you reboot your esp/arduino and does it still work then? (not just connecting/disconnecting)
- Are you sure the library uses the permission ESP_GATT_PERM_READ_ENC_MITM internally and not just
ESP_GATT_PERM_READ_ENCRYPTED ? (this is a major difference!!)
- Could you please enable lowlevel BT logging for GATT? Then I can see more details.

chegewara, 2018-06-28:
I can provide code so you can test it yourself.

boriszweimueller, 2018-06-28:
@chegewara 
E (11480) BT_GATT: GATT_INSUF_AUTHENTICATION: MITM Required

This is good. Same here - but it does not work after rebooting..

boriszweimueller, 2018-06-28:
code would be great. I don't care if you use Neil's lib - I can pick the relevant parts out of it.

chegewara, 2018-06-28:
Ok, i didnt try with reboot. After reboot i have the same issue on read:
```
E (13514) BT_GATT: GATT_INSUF_AUTHENTICATION: MITM Required
E (13611) BT_GATT: GATT_INSUF_AUTHENTICATION: MITM Required
E (13709) BT_GATT: GATT_INSUF_AUTHENTICATION: MITM Required
```
Yes, i can confirm its something wrong (i would say it can be my code if not your issue).

chegewara, 2018-06-28:
I performed one more test. This time with:
`esp_ble_set_encryption(param->connect.remote_bda, ESP_BLE_SEC_ENCRYPT_MITM);`
and i got exact same result.

chegewara, 2018-06-28:
I performed one more test. This time with:
`esp_ble_set_encryption(param->connect.remote_bda, ESP_BLE_SEC_ENCRYPT_MITM);`
and i got exact same result. But in this case esp32 recognizes bonded android devices and does not request pairing/bonding, only there is issue with insuficient authentication:
```
D (6779) BLEUtils: Received a GAP event: ESP_GAP_BLE_AUTH_CMPL_EVT
D (6780) BLEUtils: [bd_addr: 50:4a:4d:45:e8:59, key_present: 0, key: ***, key_type: 0, success: 1, fail_reason: 0, addr_type: ***, dev_type: ESP_BT_DEVICE_TYPE_BLE]
```

boriszweimueller, 2018-06-28:
yes. the call to `esp_ble_set_encryption` does not change anything. It simply tells the stack what it's possibilities are. 
Actually I don't understand why this should be necessary. Normally the characteristic permissions define the security constraints which have to be met to access them.

chegewara, 2018-06-28:
Maybe its small offtopic, just want to explain some thing:
Some devices requires to connect with authentication (passkey, yes/no confirm) and it that case you can but you dont have to protect characteristics with this flag `ESP_GATT_PERM_WRITE_ENC_MITM`, and some devices allow to connect but requires authentication to read and/or write characteristic.
You can allow to read without authentication but request authentication when you try to write. In that case you wont use `esp_ble_set_encryption`. 

There is one more thing ive learned very recently. BLE peripherals that could work with iOS devices must to implement security in very special way. You cant use `esp_ble_set_encryption`, because device should connect without authentication and characterictics needs to be secured the way you are trying to do (just MITM is not required). At least this solution helped to solve issues with esp32 BLE HID keyboard.

boriszweimueller, 2018-06-28:
Oh yes, of course - my comment was bad (i had `esp_ble_gap_set_security_param` in mind).

With `esp_ble_set_encryption` you switch security on. If I remember correctly, the standard defines, that the central should switch security on, not the peripheral. And i think iPhone & Co does this correctly.

My experience is: iOS & Co is very stable (together with esp32) once you have managed to make it work. With android we encounter many issues (spontaneous disconnections all the time).

linsid, 2018-06-29:
Hi, I am having the same problem.

I did a test - paired, bonded, disconnected. Connect, pair - log the central info (read works), reset, connect, pair - log the central info (insufficient authentication). In both cases all the information exchanged during the pairing was exactly the same. The keys are maintained during the power off.

There is another case, if we set ESP_LE_AUTH_REQ_SC_MITM (no bonding), but we keep characteristic permissions ESP_GATT_PERM_READ_ENC_MITM. It always gives the insufficient authentication error. Comparing the Central logs of Espressif and Cypress peripherals pairing I see that in the Espressif is missing this message:

    [16:09:38:444] : 'Misc Event Notification' received
    [16:09:38:444] : Event Code: 0x002C
    [16:09:38:444] : Event Data: [0A:2D:F4:65:E3:BD:7B:49:1E:B4:C0:95:95:13:46:73]

I assume it has something to do with encryption or authentication key?
However, if we enable bonding, this message will be exchanged. Maybe that is why it works only if the bonding is enabled?

Moreover, even thou I set pairing to ESP_LE_AUTH_REQ_SC_MITM and characteristic permissions to ESP_GATT_PERM_READ_ENCRYPTED and ESP_GATT_PERM_WRITE_ENCRYPTED, with this pairing I can read, but if I try to write I am getting insufficient authentication error.

boriszweimueller, 2018-06-29:
Hi. Thanks for the info. There really seems to be some issues with BLE security.

Athough I can not reproduce your behaviour with the not bonded devices.

Specifying `ESP_LE_AUTH_REQ_SC_MITM` in the security setup and `ESP_GATT_PERM_READ_ENC_MITM` for the characteristics perfectly works. 

Because we have no bonding, the devices need to pair (exchange keys) on every new connection - that's the expected behaviour.

I tested with IDV v3.0.1, with iPhones and Android Phones. Maybe it's different with Cypress pheripherals.

The problem arises as soon as I enable bonding. The keys are stored in NVS and are still available after a reboot - but then I'm getting `GATT_INSUF_AUTHENTICATION`.

boriszweimueller, 2018-06-29:
@linsid An additional important fact:

> There is another case, if we set ESP_LE_AUTH_REQ_SC_MITM (no bonding), but we keep characteristic permissions ESP_GATT_PERM_READ_ENC_MITM. It always gives the insufficient authentication error. 

Maybe you have to adjust your IOCaps. Because for MITM protection to work, you have your devices to do a secure pairing - e.g. passkey or numeric comparison. So - you have to specify at least `ESP_IO_CAP_OUT `. If you for example use `ESP_IO_CAP_NONE` your devices end up doing just-works pairing which is not enough for MITM protection.

linsid, 2018-07-02:
Thanks for the help, but I am already using:
`esp_ble_auth_req_t BLE_SECURITY_AUTH_REQ_MODE = ESP_LE_AUTH_REQ_SC_MITM;`
`esp_ble_io_cap_t BLE_SECURITY_IO_CAP = ESP_IO_CAP_OUT;`
And:
`esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &BLE_SECURITY_AUTH_REQ_MODE, sizeof(BLE_SECURITY_AUTH_REQ_MODE));`
`esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &BLE_SECURITY_IO_CAP, sizeof(BLE_SECURITY_IO_CAP));`
It should be ok for the MITM, but it does not work. Obviously, if it works for you, I am doing something wrong.

linsid, 2018-07-02:
I got more details about my problem. After enabling the stack log `#define CONFIG_BT_STACK_NO_LOG 0` I got that error:
`E (39253) BT: Non bonding: No keys will be exchanged`
`E (39254) BT: lmp_version_below LMP version 6 < 8`

Mr Google says:

LMP | Bluetooth Version
-- | --
0 | Bluetooth 1.0b
1 | Bluetooth 1.1
2 | Bluetooth 1.2
3 | Bluetooth 2.0 + EDR
4 | Bluetooth 2.1 + EDR
5 | Bluetooth 3.0 + HS
6 | Bluetooth 4.0
7 | Bluetooth 4.1
8 | Bluetooth 4.2
9 | Bluetooth 5

That means Espressif, which is v4.2, will not talk to v4.0?

chegewara, 2018-07-02:
I dont know if that is fixed already, but this message was also when device is BLE 5.0
`E (39254) BT: lmp_version_below LMP version 6 < 8`

You can google about SMP in ble 4.0 devices and here:
https://github.com/espressif/esp-idf/issues
ive got a lot discussions about security and bonding with android smartphones BLE 4.0. Always answer was: SMP was introduced in BLE 4.2

linsid, 2018-07-02:
Hmm. I can find definition of SMP in specification of V4.0 Vol 3 Part H 3.
Anyway, I tested with 4.1 and 5 and you are right - it allways gives that message.

chegewara, 2018-07-02:
This message also does not mean error if you are not using bonding:
`E (39253) BT: Non bonding: No keys will be exchanged`
and from your post https://github.com/espressif/esp-idf/issues/2036#issuecomment-401697131 we can see you dont use bonding.

linsid, 2018-07-03:
After a bit more of investigations I found that:
If I am using `ESP_LE_AUTH_REQ_SC_MITM`, after pairing `sec_flags=0x1680`.
If I am using `ESP_LE_AUTH_REQ_SC_MITM_BOND`, after pairing `sec_flags=0x3680`
If I am using `ESP_LE_AUTH_REQ_SC_MITM_BOND`, after bonding and processor reset `sec_flags=0x1680`

So the difference is this flag:
`#define BTM_SEC_LE_AUTHENTICATED   0x0200   /* LE link is encrypted after pairing with MITM */`
`#define BTM_SEC_LE_LINK_KEY_AUTHED 0x2000   /* pairing is done with MITM */`

Inside `BTM_GetSecurityFlagsByTransport` function the flag is shifted `(p_dev_rec->sec_flags >> 8)` and later on, when I am trying to read the characteristic `gatts_check_attr_readability` function checks the security flag:

`if ((perm & GATT_READ_MITM_REQUIRED ) && !(sec_flag & GATT_SEC_FLAG_LKEY_AUTHED)) {`

Which means:
`#define GATT_SEC_FLAG_LKEY_AUTHED       BTM_SEC_FLAG_LKEY_AUTHED`
`#define BTM_SEC_FLAG_LKEY_AUTHED    0x20`

To me, reading the comments, the meaning of `BTM_SEC_LE_AUTHENTICATED` and `BTM_SEC_LE_LINK_KEY_AUTHED` sounds very similar. But it makes a big difference in operation. 

To get `BTM_SEC_LE_AUTHENTICATED` the function `btm_sec_encrypt_change` is called. 
To get `BTM_SEC_LE_LINK_KEY_AUTHED` the function `btm_sec_save_le_key` is called.

boriszweimueller, 2018-07-03:
Thanks for this investigation!
Actually I thing during bonding, the keys should be persisted in the NVS (btm_sec_save_le_key), and later on - during reading - it should be read back.
As everything worked with IDF version 3.0RC, maybe we could try to compare where the 3.0 RC and the current version behave differently.

chegewara, 2018-07-03:
@Yulong-espressif @blueMoodBHD Could you help here?

Weijian-Espressif, 2018-07-10:
Hi all:
      I am sorry, it is really a bug, I'm fixing the bug in master and v3.1. By the way, it works well in master and v3.1 for Android phones. Now iPhones have a bug, I am fixing it, thanks.

Weijian-Espressif, 2018-08-06:
Hi all:
   we have fixed the bug, the commit id is 163dbc32 , please update to the latest IDF and retry, thanks.

boriszweimueller, 2018-08-06:
@Weijian-Espressif: Thank you very much for the fix. 

I did a short test with the latest IDF with MITM and bonding and it works. I tested with an iPhone 6, a Huawei P20 and an OSX BLE client.

I will later do some more testing with more devices. I think we can close the issue for now.

Thank you!

closed 2018-08-06 by boriszweimueller