espressif/esp-idf #2038: esp32 ota failed,  ota_0 size set to 2M  in partition table,

NOA-Vernon, 2018-06-08:
Hello, 

I want to use the OTA function to my project, I set the ota_0 size to 2M in partition table as my bin file is greater than 1M,  however, I can download the bin file and write to flash successfully, but it doesn't work after the update the firmware. Can anyone point out the problems? 
thanks in advance.

Following is my partition table and esp32 print log from uart.

# Name,   Type, SubType, Offset,   Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
factory,  0,    0,       0x10000,  2M
ota_0,    0,    ota_0,   ,         2M
ota_1,    0,    ota_1,   ,         2M

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5652
load:0x40078000,len:0
load:0x40078000,len:14868
entry 0x40078628
[0;32mI (28) boot: ESP-IDF v3.1-dev-961-ga255622 2nd stage bootloader[0m
[0;32mI (28) boot: compile time 11:07:34[0m
[0;32mI (28) boot: Enabling RNG early entropy source...[0m
[0;32mI (34) boot: SPI Speed      : 40MHz[0m
[0;32mI (38) boot: SPI Mode       : DIO[0m
[0;32mI (42) boot: SPI Flash Size : 16MB[0m
[0;32mI (46) boot: Partition Table:[0m
[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (72) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (79) boot:  3 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (87) boot:  4 ota_0            OTA app          00 10 00210000 00200000[0m
[0;32mI (94) boot:  5 ota_1            OTA app          00 11 00410000 00200000[0m
[0;32mI (102) boot: End of partition table[0m
[0;32mI (106) boot: Defaulting to factory image[0m
[0;32mI (111) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x11ca4 ( 72868) map[0m
[0;32mI (145) esp_image: segment 1: paddr=0x00021ccc vaddr=0x3ffb0000 size=0x0376c ( 14188) load[0m
[0;32mI (151) esp_image: segment 2: paddr=0x00025440 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (153) esp_image: segment 3: paddr=0x00025848 vaddr=0x40080400 size=0x0a7c8 ( 42952) load[0m
[0;32mI (179) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x5cb60 (379744) map[0m
[0;32mI (312) esp_image: segment 5: paddr=0x0008cb80 vaddr=0x4008abc8 size=0x04c3c ( 19516) load[0m
[0;32mI (321) esp_image: segment 6: paddr=0x000917c4 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (330) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (330) boot: Disabling RNG early entropy source...[0m
[0;32mI (333) cpu_start: Pro cpu up.[0m
[0;32mI (336) cpu_start: Starting app cpu, entry point is 0x40080f84[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (347) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (354) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (360) heap_init: At 3FFB9A80 len 00026580 (153 KiB): DRAM[0m
[0;32mI (366) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (372) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (379) heap_init: At 4008F804 len 000107FC (65 KiB): IRAM[0m
[0;32mI (385) cpu_start: Pro cpu start user code[0m
[0;32mI (67) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
I (191) wifi: wifi firmware version: c25fd80
I (191) wifi: config NVS flash: enabled
I (191) wifi: config nano formating: disabled
[0;32mI (191) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (201) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (231) wifi: Init dynamic tx buffer num: 32
[... 687 lines cut ...]
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

chegewara, 2018-06-08:
I hope you have esp32 with at least 8MB flash, because you have setup 3 partitions 2MB each + few smaller special partitions.

NOA-Vernon, 2018-06-08:
Hi chegewara,
thank you for your reply, from the booting log,  the flash is 16MB. Aren't this log the real flash size? how to get the real size of the esp32

st:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5652
load:0x40078000,len:0
load:0x40078000,len:14868
entry 0x40078628
�[0;32mI (28) boot: ESP-IDF v3.1-dev-961-ga255622 2nd stage bootloader�[0m
�[0;32mI (28) boot: compile time 11:07:34�[0m
�[0;32mI (28) boot: Enabling RNG early entropy source...�[0m
�[0;32mI (34) boot: SPI Speed : 40MHz�[0m
�[0;32mI (38) boot: SPI Mode : DIO�[0m
�[0;32mI (42) boot: SPI Flash Size : 16MB�[0m
�[0;32mI (46) boot: Partition Table:�[0m
�[0;32mI (50) boot: ## Label Usage Type ST Offset Length�[0m
�[0;32mI (57) boot: 0 nvs WiFi data 01 02 00009000 00004000�[0m
�[0;32mI (65) boot: 1 otadata OTA data 01 00 0000d000 00002000�[0m
�[0;32mI (72) boot: 2 phy_init RF data 01 01 0000f000 00001000�[0m
�[0;32mI (79) boot: 3 factory factory app 00 00 00010000 00200000�[0m
�[0;32mI (87) boot: 4 ota_0 OTA app 00 10 00210000 00200000�[0m
�[0;32mI (94) boot: 5 ota_1 OTA app 00 11 00410000 00200000�[0m
�[0;32mI (102) boot: End of partition table�[0m

chegewara, 2018-06-08:
Thats fine, i just missed it.

negativekelvin, 2018-06-08:
You can confirm flash chip size by doing 
`esptool.py flash_id`

NOA-Vernon, 2018-06-08:
@negativekelvin Thank you, I have confirmed through esptool.py flash_id, the flash size is 4MB only.

FayeY, 2018-06-13:
Any update? Is this issue solved now?